UniversalRobots/Universal_Robots_ROS_Driver #100: unable to control the Robot after the Velocity controller update

Karthik-Ranga, 2020-02-05:
# Summary
Velocity controller update

*Introduction to the issue*

Trouble connecting to the robot after the updating the velocity controller.

# Versions
 - ROS Driver version: Melodic
 - Affected Robot Software Version(s): 
 - Affected Robot Hardware Version(s): UR3e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): ExternalControl 

# Impact
The robot does not respond to the motions through python or rviz after the velocity controller update.

# Issue details

Throws an error 
`Failed to load joint_group_vel_controller`

[image]

## Project status at point of discovered
After the Velocity controller update. Prior to that, the robot connected fine and executed python scripts. 
Couple of hours ago, after the velocity controller update, the robot connects, but shows the below error and the robot doesn't move.

[image]

gavanderhoorn, 2020-02-05:
Just making sure: you have rebuilt your workspace after pulling in the updates, correct?

Because the first screenshot (please just copy-paste terminal text next time) suggests the `ros_control` config is trying to load a controller for which the hw-interface has no support. But that support was only added by the same PR that adds the configuration files.

So without rebuilding your workspace, you'll have the configuration files, but no corresponding infrastructure support in the actual driver binary.

Karthik-Ranga, 2020-02-05:
First, I apologize for the screenshot. I am working on 2 different computers. so I couldn't copy paste. I will make sure I follow the next time. 
Yes I forgot to rebuild the workspace. Silly me! 
I guess that will solve my problem. I will test it now.

Karthik-Ranga, 2020-02-06:
Yes it worked.

closed 2020-02-06 by Karthik-Ranga