UniversalRobots/Universal_Robots_ROS_Driver #102: Topic with the ur_driver/joint_speed in ur_modern_driver

MinhDangTG, 2020-02-10:
# Summary

Are there any same topics with the ur_driver/joint_speed in the ur_modern_driver package

# Versions
 - ROS Driver version: 16.04
 - Affected Robot Software Version(s): 4.1
 - Affected Robot Hardware Version(s): UR3 CB3
 - Robot Serial Number:
 - UR+ product(s) installed: No
 - URCaps Software version(s):

Hello everyone, 

I was using the ur_modern_driver package and then I figured out this package had a better checking system with all the stops (safeguard stop, protective stop...) with the new version of the UR Software also when the ur_modern_driver began to be a bit old. 

With the ur_modern_driver package, I can publish the joint velocities with accelerations to the ur_driver/joint_speed at the rate of 125Hz. I did the installation of the 
Universal_Robots_ROS_Driver package so do anyone know which ros_topic has the same usage/benefits as the ur_driver/joint_speed in the ur_modern_driver package? 

Thank you

fmauch, 2020-02-10:
You can use the `joint_group_vel_controller` instead of the default `scaled_pos_traj_controller`. See the [controller_manager](http://wiki.ros.org/controller_manager) documentation on how to switch between different controllers.

closed 2020-02-10 by fmauch

tkelestemur, 2020-02-20:
I was able to send velocity command by switching the controllers:
```
rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_vel_controller']
stop_controllers: ['scaled_pos_traj_controller']
strictness: 2
start_asap: false
timeout: 0.0"
```
But if I don't stop the `scaled_pos_traj_controller` I cannot start the `joint_group_vel_controller`. Is there way to use both of the at the same time?

gavanderhoorn, 2020-02-20:
> But if I don't stop the `scaled_pos_traj_controller` I cannot start the `joint_group_vel_controller`. Is there way to use both of the at the same time?

No, that's not how `ros_control` works.

Can you clarify why you'd want to have both controllers active at the same time?

You can actually ask `ros_control` to *switch* the controllers.

tkelestemur, 2020-02-20:
In my task, the UR goes to a neutral joint configuration using MoveIt and then it starts executing a linear motion in task space using the cartesian velocity controller that I wrote using the `joint_group_vel_controller/command'. I guess I can switch between controllers during the execution.

gavanderhoorn, 2020-02-20:
Yes, that would be the envisioned flow of events.

See [Switching between controllers with ros_control / controller_manager](https://answers.ros.org/question/259022) for a related Q&A on ROS Answers.

MinhDangTG, 2020-05-13:
Hello fmauch,

what is the script command that the joint_group_vel_controller is using? I used speedj with the ur_modern_driver. 

I should explain more. I am using a force-torque sensor at the tcp of the ur3 and an admittance control. I was working alright with ur_modern_driver since I could control the acceleration while using velocity control. When I switched to ur_robot_driver, the movement is jerky and shaking a lot since I can not define the acceleration at all (liked the speedj command). Should I continue using the /joint_group_vel_controller or switch to other controllers?

fmauch, 2020-05-13:
The speedj acceleration used with the velocity controllers is currently hard-coded [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/ros_control.urscript#L90). While I did have good results with this, that might not be the best choice for every application. Could you try whether changing this can solve your use-case? We could then make this a parameter.

You can modify this by changing this line locally on your PC. The robot controller will request this script each time the "External Control" program node gets started.

MinhDangTG, 2020-05-15:
Do you know how I can make the acceleration rate as a parameter/subscriber? I tried to open a socket similar to qd by using URServer class but I have no luck to bind the port together. 

I can add another value to write function in URServer class but it will change the driver by a lot (future update will overdrive it as well).

fmauch, 2020-05-18:
Sorry, I don't think, I quite understand what you are trying to achieve.Is this related to this issue? Otherwise, please open a new one.

sunnymints, 2021-05-18:
> The speedj acceleration used with the velocity controllers is currently hard-coded [...]

  Hi fmauch, thank your professional work. I have a question  want to ask you. 
  I hope ues joint_group_vel_controller instead of /joint_speed in ur_modern_driver to  do visual servoing, but  joint_group_vel_controller can not get the smooth vel. I think the reason is that the acceleration is fixed when  joint_group_vel_controller  calls ros_control.urscript,before. But when I changed the accelertion value from 40 to 1, the effect is still not change(the speed change is not smooth).
 I want to use joint_group_vel_controller  to get the effect of /joint_speed in ur_modern_driver what should I do？