UniversalRobots/Universal_Robots_ROS_Driver #107: Tool communication failing

emielke12, 2020-02-12:
# Summary

I'm trying to run the driver on a UR10e. I've installed the URCaps for the driver and tool communication as documented. I can run the driver just fine and publish/subscribe to topics. The issue arises when I try to set the `use_tool_communication` parameter to true. I've set the device location to `/tmp/ttyUR` but when I launch the file I get the following error:

```
Traceback (most recent call last):
.
.
.
 line 35, in main
    subprocess.call(cmd)
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

```
Is there something I am missing with the tool setup?

# Versions
 - ROS Driver version: 1.0.1
 - Affected Robot Software Version(s): 5.1.0.40195
 - Affected Robot Hardware Version(s): UR10e
 - Robot Serial Number: 20185000482
 - UR+ product(s) installed: Robotiq Vacuum Gripper
 - URCaps Software version(s): 1.0 

# Impact

Trying to get prove feasibility of product for Industrial Application. Fairly Urgent

gavanderhoorn, 2020-02-12:
Can you make sure you have `socat` installed?

emielke12, 2020-02-12:
I did not have `socat` installed. Installed and it pushes past that error. Now seeing 
```
E unlink("/tmp/ttyUR"): Operation not permitted
```
And the node terminates shortly thereafter

gavanderhoorn, 2020-02-12:
> I did not have `socat` installed.

Did you follow the build instructions in the readme? The [Building](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/6f0f2c236fa76ed9ab93fd2961ad3f76dea3ab37#building) section includes an invocation of `rosdep install`. That should have installed `socat` for you.

emielke12, 2020-02-12:
> Did you follow the build instructions in the readme? The Building section includes an invocation of rosdep install. That should have installed socat for you.

I did follow the section, it failed on another ros package I had in the same workspace.

> And please don't paraphrase/single out errors. It's rather difficult to figure out what is going on if you don't include the context.

Noted. The missing error piece was filepath to the `tool_communication` script.

Is the `E unlink("/tmp/ttyUR"): Operation not permitted` expected?

gavanderhoorn, 2020-02-12:
> I did follow the section, it failed on another ros package I had in the same workspace.

But were you able to rectify that and run `rosdep` successfully?

> The missing error piece was filepath to the `tool_communication` script.

I don't understand what you mean by this.

Could you please copy-paste the complete output of the driver startup. From the `roslaunch` command line to (and including) where things start to go wrong.

emielke12, 2020-02-12:
I did run rosdep successfully.

```
roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=192.168.5.80 use_tool_communication:=true
... logging to /home/emielke/.ros/log/fba4d106-4dd5-11ea-838b-38dead941d03/roslaunch-lich-8713.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://lich:45819/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_traj_controller/action_monitor_rate: 10
 * /pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/goal_time: 0.6
 * /pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_traj_controller/state_publish_rate: 500
 * /pos_traj_controller/stop_trajectory_duration: 0.5
 * /pos_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /scaled_pos_traj_controller/action_monitor_rate: 10
 * /scaled_pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_traj_controller/state_publish_rate: 500
 * /scaled_pos_traj_controller/stop_trajectory_duration: 0.5
 * /scaled_pos_traj_controller/type: position_controll...
 * /scaled_vel_traj_controller/action_monitor_rate: 10
 * /scaled_vel_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_vel_traj_controller/constraints/elbow_joint/trajectory: 0.1
 * /scaled_vel_traj_controller/constraints/goal_time: 0.6
[... 232 lines cut ...]
[controller_stopper-7] killing on exit
[ros_control_stopped_spawner-6] killing on exit
[ros_control_controller_spawner-5] killing on exit
[ur_hardware_interface-3] killing on exit
[robot_state_publisher-2] killing on exit
[INFO] [1581539213.640382]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1581539213.642759]: Shutting down spawner. Stopping and unloading controllers...
Interrupt signal (2) received.
[INFO] [1581539213.644208]: Stopping all controllers...
[INFO] [1581539213.646263]: Stopping all controllers...
[ERROR] [1581539213.655303360]: Could not stop controller 'pos_traj_controller' since it is not running
[INFO] [1581539213.655987]: Unloading all loaded controllers...
[INFO] [1581539213.658893]: Trying to unload joint_group_vel_controller
[ERROR] [1581539213.661757510]: Could not stop controller 'scaled_pos_traj_controller' since it is not running
[INFO] [1581539213.662183]: Unloading all loaded controllers...
[INFO] [1581539213.664627]: Trying to unload force_torque_sensor_controller
terminate called without an active exception
[WARN] [1581539213.800721]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[WARN] [1581539213.800721]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

emielke12, 2020-02-12:
```roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=192.168.5.80 use_tool_communication:=true
... logging to /home/emielke/.ros/log/07259ff2-4dda-11ea-838b-38dead941d03/roslaunch-lich-15737.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://lich:46797/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_traj_controller/action_monitor_rate: 10
 * /pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/goal_time: 0.6
 * /pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_traj_controller/state_publish_rate: 500
 * /pos_traj_controller/stop_trajectory_duration: 0.5
 * /pos_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /scaled_pos_traj_controller/action_monitor_rate: 10
 * /scaled_pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_traj_controller/state_publish_rate: 500
 * /scaled_pos_traj_controller/stop_trajectory_duration: 0.5
 * /scaled_pos_traj_controller/type: position_controll...
 * /scaled_vel_traj_controller/action_monitor_rate: 10
 * /scaled_vel_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_vel_traj_controller/constraints/elbow_joint/trajectory: 0.1
 * /scaled_vel_traj_controller/constraints/goal_time: 0.6
 * /scaled_vel_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_vel_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /scaled_vel_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
[... 221 lines cut ...]
^C[ur_hardware_interface/ur_robot_state_helper-8] killing on exit
[controller_stopper-7] killing on exit
[ros_control_stopped_spawner-6] killing on exit
[ros_control_controller_spawner-5] killing on exit
[ur_hardware_interface-3] killing on exit
[robot_state_publisher-2] killing on exit
[INFO] [1581541793.397297]: Shutting down spawner. Stopping and unloading controllers...
Interrupt signal (2) received.
[INFO] [1581541793.400669]: Stopping all controllers...
[ERROR] [1581541793.411924484]: Could not stop controller 'pos_traj_controller' since it is not running
[INFO] [1581541793.412647]: Unloading all loaded controllers...
[ INFO] [1581541793.416525286]: Connection to robot dropped, waiting for new connection.
[INFO] [1581541793.417604]: Trying to unload joint_group_vel_controller
[ INFO] [1581541793.424660500]: Robot ready to receive control commands.
terminate called without an active exception
[WARN] [1581541793.535362]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1581541793.899369]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1581541793.902265]: Stopping all controllers...
[WARN] [1581541793.908737]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

emielke12, 2020-02-12:
Figured it out. `/tmp/ttyUR` was a directory already. Removed directory and it pushes past error.

closed 2020-02-12 by emielke12

gavanderhoorn, 2020-02-13:
> I did run rosdep successfully.

And `socat` was still not present on your system? That would be something to figure out, as we rely on `rosdep` to install it for us (via [this line](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/6f0f2c236fa76ed9ab93fd2961ad3f76dea3ab37/ur_robot_driver/package.xml#L46)).

> Figured it out. `/tmp/ttyUR` was a directory already. Removed directory and it pushes past error.

Thanks for reporting back.

We should add a check to [ur_robot_driver/scripts/tool_communication](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/6f0f2c236fa76ed9ab93fd2961ad3f76dea3ab37/ur_robot_driver/scripts/tool_communication) for this.

I'll create an issue.

Would you happen to remember which user created that directory (symlink most likely actually)?

gavanderhoorn, 2020-02-13:
Tracking the `unlink` problem in #108.

emielke12, 2020-02-13:
> Did you run rosdep successfully after you'd installed socat manually, or before?

After, but I think the issue was other packages in my workspace failing when I called `rosdep`. When I called `rosdep` on the the `Universal_Robots_ROS_Driver` and `universal_robot` packages individually it succeeded, but not on my whole workspace (with other custom ROS packages)

emielke12, 2020-02-13:
> Would you happen to remember which user created that directory (symlink most likely actually)?

I don't recall, sorry!