UniversalRobots/Universal_Robots_ROS_Driver #108: tool_communication: check path for 'tool_device_name' is free/available/owned by user if it exists

gavanderhoorn, 2020-02-13:
As reported in #107, if the path configured for `tool_device_name` already exists, there is a chance `socat` will be unable to use it.

This could be due to permissions on the link, the file already existing (with access permissions which don't allow sufficient access) or the path existing but being a directory (a real one).

The `tool_communication` could check for this and error out with an appropriate message.

Right now the only output would be something similar to:

```
socat[15820] E unlink("/tmp/ttyUR"): Operation not permitted
```

fmauch, 2020-02-17:
Thanks for tracking that down :-)

gavanderhoorn, 2020-02-17:
It was @emielke12 who figured it out really.