UniversalRobots/Universal_Robots_ROS_Driver #112: external control URCap from ur_robot_driver cannot establish connection with host pc

ipa-rwu, 2020-02-19:
*Introduction to the issue*

# Versions
 - ROS Driver version: 
 - Affected Robot Software Version(s): 5.2.1
 - Affected Robot Hardware Version(s): ur5e
 - Robot Serial Number: ur5e
 - UR+ product(s) installed:
 - URCaps Software version(s): `externalcontrol-1.0.1.urcap` 

# Issue details
I am trying to run the external control node on the UR5e robot, but I get the following message from robot side:
``The connection to the remote PC could not be established``
For network from robot side, I use static address and  set IP for robot is `192.168.56.2`. For remote PC, I set up IP is `192.168.56.1`.
I can ping robot from my remote PC.

fmauch, 2020-02-19:
Did you setup the `Exernal Control` URCap with the correct IP as described in the [setup guide](../blob/master/ur_robot_driver/doc/install_urcap_e_series.md)? It should contain the IP of your remote PC (`192.168.56.1`).

Then, did you start the driver on the ROS side before starting the external control on the robot ? I just had a look at the quick start documentation and it seems to be not entirely clear there, that the two components have to be started in this order.

mdxtinkernick, 2020-02-28:
I am getting the same message. The IP address is set correctly, I am getting an error on the ROS side
Accepting socket connection failed. (errno:22)
when I launch the program on the polyscope

on the polyscope I get
Compile error: name 'null' is not defined

If I click ok and try to run it again that error does not appear, then I get
The connection to the remote PC could not be established

Each time I try, the same error appears on the ROS side

gavanderhoorn, 2020-02-28:
@fmauch: should `addr_len` have been initialised here before calling `accept(..)`?

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/e3554dafbb026d6cd70ce8dd8f8fac15990368fe/ur_robot_driver/src/comm/server.cpp#L91

mdxtinkernick, 2020-03-02:
Just for info, I have updated the software on one of the arms up to 3.12.1 and still get the same result

mdxtinkernick, 2020-03-02:
just tried building on none pi computer and it is working fine, so must be an issue for me with the pi build/infrastructure.

gavanderhoorn, 2020-03-04:
> so must be an issue for me with the pi build/infrastructure.

Not necessarily.

Seeing as a "none pi computer" is most likely a regular `amd64` device, and a Pi is actually an ARM, there may be endiannes and/or CPU architecture differences which we're not properly handling.

urielmtz, 2020-03-05:
Now, I have initialised **addr_len**  to the size of **addr** and so far the communication has not stopped. Then, it seems that this has fixed my issue, but I will continue doing more tests.

Many thanks!

fmauch, 2020-03-27:
@ipa-rwu @mdxtinkernick Could you please test, whether #148 fixes the issue for your cases, as well?

@urielmtz Thanks for testing this out!

ipa-rwu, 2020-03-27:
@fmauch I am sorry for response later. The driver already worked for me. If you update again, I will try when I can enter lab.

fmauch, 2020-08-14:
Closing this, as #148 got merged already.

closed 2020-08-14 by fmauch