UniversalRobots/Universal_Robots_ROS_Driver #113: RT-kernel + Nvidia Geforce rtx incompatible

carcormir, 2020-02-19:
Hi!

During the installation process of the real-time kernel I did not have any issues. However as soon as I rebooted my computer the graphic card was not recognized anymore, for that I had to switch kernel again but it is not rt.

We do need the GPU (Nvidia GeForce RTX 2080 Ti)  to run some deep learning image analysis stuff but we would also need to have a real time kernel (given the requirements specified for the UR10e ROS driver).

I have been doing some research but all I could not find a way to solve my issue. Maybe someone here has some similar experience?

Thanks in advance for the help.

Carlos

fmauch, 2020-02-19:
Hi Carlos, this might not be the right place to ask this. As far as I know nobody of our dev team has been working with a similar setup, yet. We usually have a separate machine controlling the robot and then have a multi-PC ROS setup for the whole application.

Your main problem ist to get a realtime-enabled kernel running together with an NVidia graphics card (or maybe this is just an issue with that model, but I doubt it). It might help to recompile the nvidia kernel module for the new kernel, which is typically done using [dkms](https://help.ubuntu.com/community/DKMS). However, from a quick review it seems that a preempt_rt kernel and the nvidia module don't play along very well.

This is as much support as I can give you at this point. As being said initially, this might not be the best place to ask this, after all.

gavanderhoorn, 2020-02-19:
The nvidia binary blob driver (ie: `nvidia-current` et al. on Debian/Ubuntu) is indeed not compatible with PREEMPT patched kernels.

This is a known issue.

I would indeed recommend to use a separate machine, as @fmauch suggests.

fmauch, 2020-02-19:
@gavanderhoorn Do you have a resource for this? I'd like to add this to the FAQ section.

gavanderhoorn, 2020-02-19:
I wouldn't be able to link you to any specific statement somewhere, no.

It's just something "the internet" knows.

I'll try to remember this later today and see whether I can find something.

It's not specific to PREEMPT_RT btw. Nvidia's driver is just really picky about what it is and isn't compatible with.

janisa9, 2020-02-19:
Regarding the same situation, I made multi-PC ROS setup where one ROS PC (rt kernel) is controlling the robot and other one processing CV alghoritms. But I ran into "hickups" -'catch up' behaviour. 

Are there any prerequisites for multi-PC ROS setup and running Universal_Robots_ROS_Driver?

fmauch, 2020-02-19:
@janisa9 please don't hijack issues by moving them to another direction. Feel free to open a new issue describing your problems thoroughly.

janisa9, 2020-02-19:
> @janisa9 please don't hijack issues by moving them to another direction. Feel free to open a new issue describing your problems thoroughly.

Sorry, please feel free to delete this, i will open a new issue

carcormir, 2020-02-19:
Thanks @fmauch and @gavanderhoorn for your replies. Even though it is not the best solution to use a not PREEMPT_RT kernel I will give it a try since I am not sure I can get another PC.
However I will close this issue since you provided me with a solution.
Thanks again :)

closed 2020-02-19 by carcormir

fmauch, 2020-02-19:
@carcormir Maybe a quick feedback would be great: Would the answer provided in #114 have helped you initially?

carcormir, 2020-02-19:
@fmauch I think so yes. Mainly because it has the Nvidia keyword which would probably help trying to find related issues.