UniversalRobots/Universal_Robots_ROS_Driver #116: Multi-ROS PC setup and driver behaviour

janisa9, 2020-02-19:
# Versions
 - ROS Driver version:
 - Affected Robot Software Version(s): 3.9.1
 - Affected Robot Hardware Version(s): ur5
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): externalcontrol-1.0.1.urcap
 - ROS version - melodic

# Issue details
I have multi-ROS pc setup, where one ROS PC (rt kernel) is controlling the robot and other one processing CV alghoritms (using intel real sense d415 camera). Communication between robot control node and CV node is done via Service.

While using multi-ROS pc setup I ran into "hickups" -'catch up' behaviour. While using only one ROS PC (rt kernel) setup (same robot movements) I didn't observe such behaviour.

I am wondering if multi-ROS pc setup causes this behaviour, or it is only related to ROS PC with rt kernel ( where driver is installed)? Are there any prerequisites for multi-PC ROS setup and running Universal_Robots_ROS_Driver?

fmauch, 2020-02-19:
What is your current network setup? Is the robot connected to the same network as the two PCs? Is any large amount of data transferred over the network (e.g. image streams from the camera)?

What does the panel say about "extrapolation count"? You can see this inside the "Variables" tab in the program view or simply in the run view.

Please note that in the [setup guide](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md) we suggest to use a direct connection between the robot and the control PC as a shared network adds non-deterministic problems to the communication.

I assume that the network is flooded with too many packages being sent such that delivery time for single packages get quite high. You can only circumvent this by either getting some fancy hardware that can prioritize packages between the control-PC and the robot or by physically separating the two network connections, in which case you will need a control-PC with two network ports, though.

gavanderhoorn, 2020-02-19:
Additionally it would be good to know a bit more about this:

> Communication between robot control node and CV node is done via Service.

What exactly is part of the service request? What are its semantics? How often is it called?

Also: how is motion controlled exactly by the "robot control node"? Are you using the scaled `JointTrajectoryController`, or one of the position or velocity joint group controllers?

There are no services that communicate any motion related information in `ur_hardware_interface`, so I would concur with @fmauch and suggest providing a high-level overview of the node graph (or a boxes-and-arrows diagram) would be useful.

janisa9, 2020-04-09:
Sorry for me delay in response.

I have upgraded the control PC with additional network card for direct connection with robot controller, but I can still occasionally observe the problem.

Bit more in details for CV soft and service requests. The pc that is responsible for computer vision soft is connected with robot control PC on different network than robot control pc and robot controller. The service request for object position is done before the execution of robot trajectory, there are no active service requests while robot is executing trajectory, so I highly doubt this could be the problem. 

Extrapolate_count = 0
Extrapolate__max_count = 1

I am using scaled_pos_traj_controller.

Please see attached node graph if this helps.
[image]

fmauch, 2020-04-14:
If the `extrapolate_max_count` value is that low, the problem can't be due to missing network packages. Is speed scaling involved into the trajectory execution? Do the hickups occur deterministically (at the same time in the same trajectory if played multiple times)?

You can check speed scaling by plotting the topic `/speed_scaling_factor`.

fmauch, 2020-05-12:
Any updates on this?

janisa9, 2020-05-12:
The hickups occur randomly while executing the same trajectory.

Actually the issue is not connected to the multi robot setup as I thought earlier, as hickups also occurs without multi-ros PC setup. As the hickups occur from time to time I suppose they were not present when I made initial tests without multi-ROS PC setup.

[image]
[image]
[image]
[image]

fmauch, 2020-05-12:
Thank you for the plots. Could you maybe do one more that includes the `/speed_scaling_factor` topic? And the `extrapolation_max_count` is really that low after such a hickup occurred?

Could you also please post the full log output of the driver startup?

janisa9, 2020-05-12:
I haven't seen anything else on `extrapolation_max_count` than 1

Here it is with `/speed_scaling_factor ` included
[image]

`
SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['shoulder_pan_jo...
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /mongo_wrapper_ros_roboarm_22716_3737315636332119792/database_path: /home/edi/ROS_wor...
 * /mongo_wrapper_ros_roboarm_22716_3737315636332119792/overwrite: False
 * /pos_traj_controller/action_monitor_rate: 10
 * /pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/goal_time: 0.6
 * /pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_traj_controller/state_publish_rate: 125
 * /pos_traj_controller/stop_trajectory_duration: 0.5
 * /pos_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_description_kinematics/manipulator/kinematics_solver: kdl_kinematics_pl...
 * /robot_description_kinematics/manipulator/kinematics_solver_attempts: 3
 * /robot_description_kinematics/manipulator/kinematics_solver_search_resolution: 0.005
 * /robot_description_kinematics/manipulator/kinematics_solver_timeout: 0.005
 * /robot_description_planning/joint_limits/elbow_joint/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/elbow_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/elbow_joint/max_acceleration: 3.15
 * /robot_description_planning/joint_limits/elbow_joint/max_velocity: 3.15
 * /robot_description_planning/joint_limits/shoulder_lift_joint/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/shoulder_lift_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/shoulder_lift_joint/max_acceleration: 3.15
 * /robot_description_planning/joint_limits/shoulder_lift_joint/max_velocity: 3.15
 * /robot_description_planning/joint_limits/shoulder_pan_joint/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/shoulder_pan_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/shoulder_pan_joint/max_acceleration: 3.15
 * /robot_description_planning/joint_limits/shoulder_pan_joint/max_velocity: 3.15
 * /robot_description_planning/joint_limits/wrist_1_joint/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/wrist_1_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/wrist_1_joint/max_acceleration: 3.2
 * /robot_description_planning/joint_limits/wrist_1_joint/max_velocity: 3.2
 * /robot_description_planning/joint_limits/wrist_2_joint/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/wrist_2_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/wrist_2_joint/max_acceleration: 3.2
 * /robot_description_planning/joint_limits/wrist_2_joint/max_velocity: 3.2
 * /robot_description_planning/joint_limits/wrist_3_joint/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/wrist_3_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/wrist_3_joint/max_acceleration: 3.2
 * /robot_description_planning/joint_limits/wrist_3_joint/max_velocity: 3.2
[... 258 lines cut ...]
[INFO] [1589293897.097484]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1589293897.099600]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1589293897.100971]: Loading controller: scaled_pos_traj_controller
[INFO] [1589293897.101340]: Loading controller: pos_traj_controller
[INFO] [1589293897.133095]: Loading controller: speed_scaling_state_controller
[INFO] [1589293897.156975]: Loading controller: joint_group_vel_controller
[ INFO] [1589293897.158578399]: Robot's safety mode is now NORMAL
[ INFO] [1589293897.159430836]: Robot mode is now RUNNING
[INFO] [1589293897.164968]: Loading controller: force_torque_sensor_controller
[ERROR] [1589293897.165070044]: Could not load controller 'joint_group_vel_controller' because controller type 'velocity_controllers/JointGroupVelocityController' does not exist.
[ERROR] [1589293897.165171758]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[... cut ...]
[ERROR] [1589293898.169298]: Failed to load joint_group_vel_controller
[INFO] [1589293898.170079]: Controller Spawner: Loaded controllers: pos_traj_controller
[ERROR] [1589293898.171244]: Failed to load force_torque_sensor_controller
[INFO] [1589293898.172020]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller
[INFO] [1589293898.181039]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller
`

fmauch, 2020-05-12:
Thank for coming back on this To summarize the situation from your data:
 * No speed scaling involved
 * No significant package loss happening between PC and robot
 * Realtime kernel seems to be running fine
 * There are other nodes running on the same machine (-> We could have a load problem, to be checked)

Except for the load problem I am a little out of guess here... To debug this further we would have to look at the position commands being sent to the robot and their timing. This is something that requires code modification though. I'll dig around for my debug branches, could you have a look at your machine's load in the meantime?

fmauch, 2020-05-13:
I dug out my debugging branch and rebased it ontop of the current master: https://github.com/fmauch/Universal_Robots_ROS_Driver/tree/debug_timing

Note that this debugging level might impact the driver's performance, though it shouldn't be too bad.

With this you have a couple of additional topics:
 * `/command_sent` commands being sent to the network socket by the hardware interface
 * `/target_q` / `/target_qd` target values (positions / velocities) being sent by the running program (on the robot) to the lower-level controller. This includes any interpolation / smoothing being done by `servoj`
 * `/actual_q` / `/actual_qd` the actual values being reported by the robot hardware (from the same cylcle as the respective `target_x` values.

It would be interesting at which level the hickups start. We can see that they are not in the controller's desired state, but show up in the joint states (which should be equal to `/target_q`).

Could you please test again including plotting all these values? Please restrict it to one axis as the graph will become rather unreadable otherwise...

In the plot, please activate "Show plot markers" in the rqt_plot options. This way it is easier to have a look at the exact timing instead of looking for sharp turns in the curves.

Could you also send me your test trajectory please (What is sent to `/scaled_pos_traj_controller/follow_joint_trajectory/goal`), so I can have a look at it myself, if I find the time.

gavanderhoorn, 2020-05-13:
It would also be good to check what is running on the robot in parallel to the external control program.

I know it "should not matter", but I've seen UR controllers almost overloaded with concurrent threads and programs running as daemons (urcaps) leading to all sorts of strange behaviour.

janisa9, 2020-05-13:
See machine  usage below. Neither the robot controller or robot control PC is overloaded, thus when the hickups happen there is no significant change in load or anything else. I will continue with debugging branch.

**UR_robot top**
```
top - 12:14:05 up  3:15,  1 user,  load average: 0.23, 0.21, 0.23
Tasks: 127 total,   1 running, 126 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.7 us,  0.7 sy,  1.5 ni, 97.1 id,  0.0 wa,  0.0 hi,  0.1 si,  0.0 st
KiB Mem:   1897028 total,   701280 used,  1195748 free,     7180 buffers
KiB Swap:        0 total,        0 used,        0 free,   142044 cached

  PID USER      PR  NI  VIRT  RES  SHR S  %CPU %MEM    TIME+  COMMAND           
 2993 root      20   0  441m 191m  16m S   8.3 10.3   7:41.40 URControl         
 2618 root      30  10  906m 342m  24m S   7.0 18.5   9:17.02 java              
 2551 root      19  -1 32904  12m 5832 S   0.7  0.7   0:46.37 Xorg              
    3 root      -2   0     0    0    0 S   0.3  0.0   0:39.62 ksoftirqd/0       
   20 root      -2   0     0    0    0 S   0.3  0.0   0:38.18 ksoftirqd/1       
   26 root      -2   0     0    0    0 S   0.3  0.0   0:02.33 rcuc/2            
   27 root      -2   0     0    0    0 S   0.3  0.0   0:38.66 ksoftirqd/2       
   34 root      -2   0     0    0    0 S   0.3  0.0   0:36.40 ksoftirqd/3       
 1249 root     -76   0     0    0    0 S   0.3  0.0   0:17.04 urThread          
 1947 root     -51   0     0    0    0 S   0.3  0.0   0:00.27 irq/113-eth1-Tx   
 1948 root     -51   0     0    0    0 S   0.3  0.0   0:00.41 irq/114-eth1-Tx   
    1 root      20   0  2584  756  656 S   0.0  0.0   0:00.98 init              
    2 root      20   0     0    0    0 S   0.0  0.0   0:00.00 kthreadd          
    5 root       0 -20     0    0    0 S   0.0  0.0   0:00.00 kworker/0:0H      
    6 root      20   0     0    0    0 S   0.0  0.0   0:00.15 kworker/u8:0      
    7 root      rt   0     0    0    0 S   0.0  0.0   0:00.00 posixcputmr/0     
    8 root      20   0     0    0    0 S   0.0  0.0   0:00.00 kclksetdelayd     
```
**Control PC top**
```
top - 12:19:02 up 1 day,  2:05,  1 user,  load average: 9.60, 6.95, 5.19
Tasks: 412 total,   3 running, 327 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.0 us,  5.5 sy, 14.4 ni, 77.5 id,  0.0 wa,  0.0 hi,  2.6 si,  0.0 st
KiB Mem : 16259848 total,  9851660 free,  2672212 used,  3735976 buff/cache
KiB Swap:  2097148 total,  2097148 free,        0 used. 12931436 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND     
30504 edi       39  19 1755944 195948  83944 R 105.3  1.2   0:26.02 rqt_plot    
27875 edi       rt  19 1530076  25348  18044 S  24.1  0.2   4:30.23 ur_robot_d+ 
 1452 edi       39  19  594964 117452  94296 S  20.1  0.7   9:00.29 Xorg        
27906 edi       39  19 3279336 353292 141012 S  18.2  2.2   3:14.28 rviz        
28843 edi       39  19  659844  55768  43232 S  17.8  0.3   0:25.09 gnome-syst+ 
 1587 edi       39  19 4554560 391104 126852 R   7.9  2.4   4:31.10 gnome-shell 
30436 edi       39  19 1277344  69704  34288 S   3.6  0.4   0:02.87 pick_and_p+ 
30437 edi       39  19 1524632 136396  41912 S   3.6  0.8   0:05.31 move_group  
27874 edi       39  19  530400  15536  13840 S   1.7  0.1   0:19.06 robot_stat+ 
   89 root      -2   0       0      0      0 S   1.3  0.0   1:15.90 ktimersoft+ 
 1922 edi       39  19  815112  49040  28908 S   1.3  0.3   0:27.07 gnome-term+ 
23989 root     -51   0       0      0      0 S   1.3  0.0   0:14.26 irq/143-en+ 
   10 root      -2   0       0      0      0 S   1.0  0.0   1:14.36 ktimersoft+ 
   26 root      -2   0       0      0      0 S   1.0  0.0   1:12.89 ktimersoft+ 
   35 root      -2   0       0      0      0 S   1.0  0.0   1:12.77 ktimersoft+ 
   53 root      -2   0       0      0      0 S   1.0  0.0   1:13.60 ktimersoft+ 
   63 root      20   0       0      0      0 S   1.0  0.0   0:47.39 ksoftirqd/5 

```

janisa9, 2020-05-13:
Please see attached plots. Only the last contains also /joint_states/position
The overview (1st and 2nd are connected) rest of plots I just zommed on random hickups.
[image]

Zoomed in hickup
[image]

[image]

[image]

janisa9, 2020-05-13:
I attached the trajectory below, thus earlier there were to separate trajectories executed (can be seen in plots), but I don't this matter as hickups are random anyway.

[trajectory_pp.txt](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/4622388/trajectory_pp.txt)

janisa9, 2020-06-01:
Any clues/updates, what could cause this?

janisa9, 2020-06-29:
@fmauch are there any updates on this issue?

fmauch, 2020-06-29:
Thanks for coming back on me. I though I had answered this already, as I already looked at the graphs... Sorry.

Looking at the graphs I'm afraid, there's not much I can help you with from a ROS driver's perspective. The joint position commands don't show any large hickups and in the end it's joint commands that we are commanding here. The `/scaled_pos_traj_controller/state/desired/velocities` value is never commanded, it is only part of the trajectory controller's internal spline representation of the trajectory.

For some reason, the controller's `target_qd` value jumps a bit at some point, which is a result of the `servoj` command running on the robot that tries to follow the position commands we are sending from ROS. You could try playing around with the [servoj parameters](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#servoj_gain-required) but you should read the URScript manual first to understand what they mean. E.g. increasing the lookahead time might get rid of those bumps, while the robot will probably increase the time delay between ROS commands and execution, resulting in an all-time larger error of your trajectory execution. So, in this case you might have to increase the path tolerance parameters of the ROS controller, as well...

You could also try to switch to a velocity-based controller which will send velocity commands rather than position commands to the robot. That generally has a lower response time, but the PID values of the controller probably will have to be tweaked (see #124 and #119).

That being said, I tried to reproduce the error with your posted trajectory on a simulated CB3-robot and a real e-Series robot. While the e-Series robot didn't show any of that behavior, with the simulation I could get minor imperfections, but nothing like you showed.

janisa9, 2020-12-10:
Thank you @fmauch, sorry for taking so long time to come back to you.

Since the summer in free moments we have occasionally tried to figure out the problem but without any luck. We rechecked that the joint position commands that are being sent out are right (without any hiccups). 

We played around with the servoj parameters with no luck, the hiccups were present in every case, but occurring very occasionally - sometimes more, sometimes less. Thus there is a note in URScript that you can see in this image, which probably explains why there are no hiccups in UR5e as the control frequency is higher:

[image]

I have updated everything to the latest versions including UR software, external control version and UR ROS driver - no changes.

We also tried to a velocity-based controller, where hiccups were not present, but PID values need to be tweaked, that is also a very time-consuming task. Have anyone successfully tweaked PID parameters?

Now it seems the only place where hiccups could arise is somewhere inside controller SW in conjunction with the ROS driver.

More options that we are considering to try are: reinstalling the real-time kernel, but this doesn't seem to be a problem. As the last thing to do, we are considering installing ROBOT IMAGE from scratch.

If you have any more ideas since summer, we would like to hear them.

janisa9, 2021-01-26:
Hi again,

We did a fresh install on different hardware with real-time kernel, and the jerky motions on real robot is present there all the time, even more distinct (which also is very weird). While testing in URSim everything works fine (setup made as close as possible to setup with real robot, the URSim was running on different PC and connected with dedicated network card). We even tried different kernel versions with no success.

This is a plot of real robot with the freshly installed control PC.
[image]
 
@fmauch @gavanderhoorn  do you have any ideas? We have tried almost everything, and the only conclusion is that something is wrong within UR5 controller itself. Also to mention we have UR5e and with this robot we haven't seen such behavior. Same setup, same packages (installed as describer in install descriptions), with UR5 we get behaviour like in the image, with UR5e everything works correctly.

fmauch, 2021-01-27:
1. URSim will always give you different output when it comes to low-level control as (as far as I know) it just does an output=input method on the lower level without fully simulating the robot's dynamics.
2. Before, you were always talking about `servoj` commands, in the latest plots you look at the velocity. When using position command mode, (using `servoj` on URScript level) the velocities from `/scaled_pos_traj_controller/state/desired/velocities` will not be used anywhere, as only the **position** setpoint will be fed into the `servoj` command. This also explains the different behavior on a UR5 with a 125Hz control loop in contrast to the UR5e with a 500Hz control loop.
3. There's nothing really "wrong" with the controller as this is basically a drawback from the control method used. The `servoj` command doesn't know about future setpoints (or only as far as the `lookahead_time` parameter goes). The robot controller wil try its best to generate a smooth motion to the next known setpoint, but as they always change and the robot should not come to a full stop at the next waypoint, the generated trajectories (at a very low level) will always get aborted and merged into a new one. So, in order to get smoother trajectories you can increase the *lookahead_time* parameter which will however lead to a greater delay between the desired and the actual positions (Basically the time difference in your latest plot). This will lead to a higher error on a ROS controller level, so you might have to increase the error tolerances there.
4. This is why there exist the velocity-based controllers that don't generate those short trajectory snippets, but use setpoints for the desired velocity instead. The main problem with velocity-based controllers is that you'll have a closed loop with the ROS side which requires well setup PID values which we currently don't have, yet. (see #119, #124). Any input on this is welcome.
5. We are currently working on getting trajectory interpolation and execution completely on the robot controller while still keeping the power of ROS control inside the driver. This should lead to more precise and smooth trajectories in cases where a dynamic path update from the ROS side is not needed.

janisa9, 2021-01-27:
@fmauch thank you for your answer.

1. That explains it, yes, but if there were some problems with RT kernel or setup, I suppose that could be also seen in UrSim.
2. I was not talking explicitly about servoj commands, only after your suggestions to play with parameters such as `lookahead_time`, which we tried but there was no difference in behaviour. And the velocity plots is the best way to show the behaviour and check it as there were no deviations in position plots, as it can be seen in earlier figures.
3. We tried to increase `lookahead_time`, but there was no difference. 
4. Yes, we tried also velocity-based controller, but that raises the issue with PID values.
5. Are there any forecast when this could be ready?

fmauch, 2021-01-27:
> 2. I was not talking explicitly about servoj commands, only after your suggestions to play with parameters such as `lookahead_time`, which we tried but there was no difference in behaviour. And the [...]

Position control will yield to position accuracy, but not necessarily to velocity smoothness, which is what I was trying to say in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/116#issuecomment-651061274

> 5. Are there any forecast when this could be ready?

Soonish. As far as I know, beta phase should be starting soon, maybe @stefanscherzinger can say more about this / include you into the beta test.

janisa9, 2021-01-29:
@fmauch @stefanscherzinger  Yes, I would like to join the beta testing, as for now, we don't really need a dynamic path update.

closed 2023-06-20 by github-actions[bot]