UniversalRobots/Universal_Robots_ROS_Driver #117: Multiple controllers in one launch file?

satvu, 2020-02-20:
# Summary

I want to be able to control a ur5e using both the scaled_pos controller and scaled_vel controller, switching between the two when needed. Is there a way to do that?

# Versions
 - ROS Driver version: Kinetic
 - Affected Robot Software Version(s): 5.3.0
 - Affected Robot Hardware Version(s):
 - Robot Serial Number: ur5e 
 - UR+ product(s) installed: 
 - URCaps Software version(s): externalcontrol-1.0.1.urcap

# Issue details
Here is the error I'm getting when I try to include both controllers in the launch file:
`[ WARN] [1582237702.259225516]: Resource conflict on [elbow_joint].  Controllers = [scaled_pos_traj_controller, scaled_vel_traj_controller, ]
[ WARN] [1582237702.259254150]: Resource conflict on [shoulder_lift_joint].  Controllers = [scaled_pos_traj_controller, scaled_vel_traj_controller, ]
[ WARN] [1582237702.259273084]: Resource conflict on [shoulder_pan_joint].  Controllers = [scaled_pos_traj_controller, scaled_vel_traj_controller, ]
[ WARN] [1582237702.259288595]: Resource conflict on [wrist_1_joint].  Controllers = [scaled_pos_traj_controller, scaled_vel_traj_controller, ]
[ WARN] [1582237702.259301967]: Resource conflict on [wrist_2_joint].  Controllers = [scaled_pos_traj_controller, scaled_vel_traj_controller, ]
[ WARN] [1582237702.259316350]: Resource conflict on [wrist_3_joint].  Controllers = [scaled_pos_traj_controller, scaled_vel_traj_controller, ]
[ERROR] [1582237301.432534812]: Could not switch controllers, due to resource conflict[ERROR] [1582237301.432704]: Failed to start controllers: scaled_vel_traj_controller, joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller`

## Steps to Reproduce
Edited this line in the ur5e launch file regarding controllers: 
  `<arg name="controllers" default="scaled_pos_traj_controller scaled_vel_traj_controller joint_state_controller speed_scaling_state_controller force_torque_sensor_controller" doc="Controllers that are activated by default."/>`

fmauch, 2020-02-21:
Hi, please have a look at the [controller manager documentation](http://wiki.ros.org/controller_manager). You cannot start both controllers simultaneously, as they claim the same resources. However, as they are defined in the `controllers.yaml` file they can be switched using the `controller_manager` interfaces as documented in the linked documentation.

There is no need to change anything in the launchfiles, if you don't desire to change the default controller. By default, the controllers in the argument `controllers` will get started, while the ones in `stopped_controllers` will be loaded, but not started. To switch between the `scaled_pos_traj_controller` and the `scaled_vel_traj_controller`, please use the following service calls:

```
$ rosservice call /controller_manager/load_controller "name: 'scaled_vel_traj_controller'"
$ rosservice call /controller_manager/switch_controller "start_controllers:                                                                                                                                                                                                  
- 'scaled_vel_traj_controller'                                      
stop_controllers:                                                   
- 'scaled_pos_traj_controller'                                      
strictness: 2"
```
If you would like to skip the loading at the beginning, you can add the velocity controllers to the `stopped_controllers` list.

I hope, this answers your question.

gavanderhoorn, 2020-02-21:
Related discussion from yesterday: #102.

closed 2020-02-24 by satvu