UniversalRobots/Universal_Robots_ROS_Driver #118: Script command by action API

khssnv, 2020-02-25:
# Enhancement Proposal
Implement an action server API to launch and track arbitrary URScript execution.

# Use case
- track script execution status
- return an error on a script command illegal (if there there is a checker available)
- cancel URScript execution

# Notes
This could be implemented using dirver API already exists (dashboard interface + script command topic).

What do you think about it? Does it make sense for anyone but me?

gavanderhoorn, 2020-02-25:
This would be nice (in fact I'm implementing this for another robot brand right now).

However, afaik it's impossible to track execution status for a particular program with the current interfaces we have. It's possible to see *some program* is executing, but that would also say "executing" when you're using the UI (as all UI actions are essentially URScriptlets).

Does the dashboard server allow introspection to the level required for this?

fmauch, 2020-02-26:
> Does the dashboard server allow introspection to the level required for this?

Not that I know. As you said, tracking script code isn't available in any of the interfaces.

khssnv, 2020-02-26:
Ok, to track what program is actually running and handle action interrupted by another source of command there should be a feedback. There are script TCP/IP communication functions (`socket_open`, `socket_send_byte`, `socket_close`), solution could have a protocol on top of it like it mentioned in #40. Is it a right approach?

fmauch, 2020-02-26:
It is an approach that helps you in some situations. It gets problematic if your script code doesn't run correctly so you never reach that part. Or if your script code is syntactically wrong.

From a driver's perspective we try to keep things as close to the robot's actual interfaces, as possible. However, adding this as a separate wrapper node might be a valid option.

If you are willing to give it a shot, feel free to open up a PR then we can discuss about it in detail...

khssnv, 2020-03-23:
> This would be nice (in fact I'm implementing this for another robot brand right now).
@gavanderhoorn, could you share it? Could be handy to estimate how much time it will take for me.

gavanderhoorn, 2020-03-23:
No, I'm afraid I can't.