UniversalRobots/Universal_Robots_ROS_Driver #119: Optimizing gains (PID values)

satvu, 2020-02-25:
# Summary
Is there a way to optimize gains (PID values) for ur5e on the controller using the driver? I'm new to ROS and UR, so sorry if this is common knowledge. I'd like get the right settings so that I can use the velocity controller optimally. 

# Versions
- ROS Driver version: Kinetic
- Affected Robot Software Version(s): 5.3.0
- Affected Robot Hardware Version(s):
- Robot Serial Number: ur5e
- UR+ product(s) installed:
- URCaps Software version(s): externalcontrol-1.0.1.urcap

# Impact
Having difficulty using the velocity controller and trying to resolve that by optimizing the gains. I played around with one of the joints' respective values and it the velocity controller seemed to work better for that joint.

fmauch, 2020-02-26:
Controller gains are setup in [controller config file](../blob/master/ur_robot_driver/config/ur5e_controllers.yaml) 

Any contribution with optimized values is more than welcome.

gavanderhoorn, 2020-02-26:
And a note: most/some `ros_control` controllers support [dynamic_reconfigure](http://wiki.ros.org/dynamic_reconfigure), which makes tuning things like PID values a lot more efficient.

gavanderhoorn, 2020-03-02:
Did you find something which works for you?

satvu, 2020-03-02:
> Did you find something which works for you?

Scope of our project no longer needs it, but I can reopen this in case this is an issue that other people are interested in or have solutions to.

stefanscherzinger, 2021-08-09:
Feel free to re-open if this becomes an issue again or if you would like to share PID gains for other users.

closed 2021-08-09 by stefanscherzinger