UniversalRobots/Universal_Robots_ROS_Driver #121: Cartesian waypoints through ROS driver

carcormir, 2020-02-26:
Hi,

I was wondering if there was any controller implemented that I could provide cartesian waypoints instead of joint positions. I know that I could use moveit! ComputeCartesianPath or inverse kinematics plugin feature but I was wondering if there was a way to avoid moveit.

I would like to use the _movel_ and _movej_ options, providing cartesian points through the ROS driver the same way that it could be done in the TP.

Thanks in advance for the help!

Carlos

fmauch, 2020-02-26:
This is currently being worked on, but it will require a bit more time, as we want to do it with a proper ROS interface which doesn't exist, yet...

carcormir, 2020-02-26:
Thanks for the reply. Anywhere I could track the development of this feature?

fmauch, 2020-02-26:
We are currently still in research phase. Once we will start active development this will be here.

tkelestemur, 2020-02-26:
You can also use a kinematics library to implement a jacobian-based task space controller. I have a -simple velocity controller for the end-effector that uses KDL. This function should help: https://github.com/tkelestemur/finger_loc_ros/blob/master/scripts/finger_loc.py#L154

carcormir, 2020-02-27:
Thanks for the replies. I consider this issue to be solved. For now.

closed 2020-02-27 by carcormir

cambel, 2020-03-12:
Hi @carcormir, you can check my [repository](https://github.com/cambel/ur3). I have implemented a controller for UR3/e (but it is relatively simple to adapt to other UR versions). It allows control of the robot's end-effector through cartesian waypoints, by solving the IK. It works on top of this `ROS driver` or the `ur_modern_driver`.
It has some other additional useful tools, such as teleoperating through a keyboard or navigational mouse using streams of position commands.