UniversalRobots/Universal_Robots_ROS_Driver #122: How to move the robot using the UR_ROS_Driver

aloukas, 2020-02-26:
I have set up everything regarding the ur_ros_driver but I cannot figure out which topic actually moves the robot, for example in servo mode, which is what I need for my application.

I can move the robot using the gui (through this command: < rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller > ) but after looking at the topic list, I figured that the  "/scaled_pos_traj_controller/command" topic might be the one I should use. After setting the
positions, velocities and joint names in the message, when I try to publish that message I get this error: "[ WARN] [1582705047.536214380]: Dropping all 1 trajectory point(s), as they occur before the current time.
Last point is 0.000000s in the past."

I should also mention that after trying to install a realtime kernel(4.14 r66), I get this error when I launch the robot: " [FATAL] [1582704334.135282509]: Failed to start controller in realtime loop. This should never happen. "

Are those two errors connected?

Ps: I also want to read the forces/torques but the force/torque controller does not seem to be launched. When I launch the robot I get a message that all the default controllers are loaded/started, including the force/torque sensor controller. But when I list the active controllers using < rosrun controller_manager controller_manager list > , the force/torque controller is labeled as 'stopped'. I then tried <rosrun controller_manager controller_manager start force_torque_sensor_controller > but it was still labeled as 'stopped' when I ran it again. 

Any ideas on that one?

fmauch, 2020-02-26:
First of all, those errors are not connected.

To move the robot to a target configuration without any feedback or a chance to abort the motion, you picked the right topic. However, you have to specify a target time for your target configuration, as otherwise the robot doesn't know how far it should move.

We suggest however, to use the action goal topic on `/scaled_pos_traj_controller/goal` which is of same type, but part of the [Action interface](http://wiki.ros.org/actionlib). See the documentation on the [follow_joint_trajectory interface](http://wiki.ros.org/pr2_controllers/Tutorials/Moving%20the%20arm%20using%20the%20Joint%20Trajectory%20Action) for details about `FollowJointTrajectory`

aloukas, 2020-02-26:
I got it running through the /command topic. Thank you for the help. But I cannot seem to get it to move through the /goal topic. I assigned positions, joint names and set Header.Stamp.Sec=4. Am I missing another sort of time?

Both topics correspond to moveJ right?
What's the equivalent of servoJ or servoL?

gavanderhoorn, 2020-02-27:
If you haven't done so, you may want to review [actionlib](http://wiki.ros.org/actionlib) and perhaps also the client tutorials. The `follow_joint_trajectory` page linked by @fmauch also relates to this.

You submit [control_msgs/FollowJointTrajectory](http://docs.ros.org/api/control_msgs/html/action/FollowJointTrajectory.html) goals (note: *not* messages) to the action server run by the driver. You don't publish individual `JointTrajectoryPoint` messages to the `goal` topic.

At a high-level:

 - create a ROS node
 - create a `control_msgs/FollowJointTrajectory` action client
 - use some method to generate a [JointTrajectory](http://docs.ros.org/api/trajectory_msgs/html/msg/JointTrajectory.html)
 - store the `JointTrajectory` inside the `control_msgs/FollowJointTrajectory` goal you created earlier
 - submit the goal to the server (using the created client)
 - wait for completion (or not, that's up to you)

So for individual points (fi for more reactive control):

 - use the `command` topic of the `JointTrajectoryController`, *or*
 - switch to a `JointGroupVelocityController` or `JointGroupPositionController` and publish to their `command` topic

For trajectories:

 - create an action client and submit `control_msgs/FollowJointTrajectory` action goals

note that the `JointTrajectoryController` also supports [trajectory replacement](http://wiki.ros.org/joint_trajectory_controller#Trajectory_replacement).

aloukas, 2020-03-03:
gavanderhoorn thank you so much for the info. After following your instructions, it worked. I wanted to ask another thing, not sure if I should open a new issue for this:

I also want to read the forces/torques but the force/torque controller does not seem to be launched. When I launch the robot I get a message that all the default controllers are loaded/started, including the force/torque sensor controller. But when I list the active controllers using < rosrun controller_manager controller_manager list > , the force/torque controller is labeled as 'stopped'. I then tried but it was still labeled as 'stopped' when I ran it again.

fmauch, 2020-03-27:
> gavanderhoorn thank you so much for the info. After following your instructions, it worked. I wanted to ask another thing, not sure if I should open a new issue for this:
>
> I also want to read the [...]

Yes, this should be discussed in a new issue. Let me just quickly state here, that it should be running. I'll close this issue, as the original question got resolved.

closed 2020-03-27 by fmauch

mat-l, 2020-04-22:
@gavanderhoorn as mentioned in a later issue/commend:

How does a message look like to send the robot in movel or movej to a point at with the previously mentioned "/scaled_pos_traj_controller/follow_joint_trajectory"?

Is there some example code how a client looks like? Or maybe @aloukas could share something?

Best regards

fmauch, 2020-04-22:
@mat-l pleas do not dig out old issues, we can keep this inside #162

mat-l, 2020-04-22:
> @mat-l pleas do not dig out old issues, we can keep this inside #162

All right, sorry. Understood, that I should text over here.