UniversalRobots/Universal_Robots_ROS_Driver #124: Tuning Parameters for Velocity Controller

jbeck28, 2020-02-27:
I've been using the `compliance_controller`, which for reference can be located here:
[compliance_controller](https://github.com/UTNuclearRobotics/ros_controllers/tree/melodic-compliance-updates)

This is a velocity controller, and hence I've had to do a bit of tuning of the PID parameters that I've found deliver acceptable results. The original issue which led to this is here: [#437](https://github.com/ros-controls/ros_controllers/pull/437)

Although I'm using the `compliance_controller`, this inherits from the `joint_trajectory_controller`, and so should work for that controller as well (a more typical controller). 

Here is a copy of the relevant part of my `ur5e_controllers.yaml`, but keep in mind that I'm not expert, and someone with more experience could probably do a better job finding optimal parameters. Further, the performance of these is dependent on the load on the arm, and my arm is carrying about 1.5kg or so. 

```
compliance_controller:
     type: velocity_controllers/ComplianceController
     joints:
       - shoulder_pan_joint
       - shoulder_lift_joint
       - elbow_joint
       - wrist_1_joint
       - wrist_2_joint
       - wrist_3_joint
     compliance_command_timeout: 1.5
     constraints:
        goal_time: 4
        stopped_velocity_tolerance: 0.01
        shoulder_pan_joint: {trajectory: 0.1, goal: 0.001}
        shoulder_lift_joint: {trajectory: 0.1, goal: 0.001}
        elbow_joint: {trajectory: 0.1, goal: 0.001}
        wrist_1_joint: {trajectory: 0.1, goal: 0.001}
        wrist_2_joint: {trajectory: 0.1, goal: 0.001}
        wrist_3_joint: {trajectory: 0.1, goal: 0.001}
     stop_trajectory_duration: 0.5
     state_publish_rate:  125
     action_monitor_rate: 10
     gains:
        #!!These values have not been optimized!!
        shoulder_pan_joint:  {p: 10.0,  i: 0.05, d: 0.1, i_clamp: 1}
        shoulder_lift_joint: {p: 10.0,  i: 0.05, d: 0.1, i_clamp: 1}
        elbow_joint:         {p: 10.0,  i: 0.05, d: 0.1, i_clamp: 1}
        wrist_1_joint:       {p: 10.0,  i: 0.05, d: 0.1, i_clamp: 1}
        wrist_2_joint:       {p: 10.0,  i: 0.05, d: 0.1, i_clamp: 1}
        wrist_3_joint:       {p: 10.0,  i: 0.05, d: 0.1, i_clamp: 1}

     # Use a feedforward term to reduce the size of PID gains
     velocity_ff:
      shoulder_pan_joint: 1.5
      shoulder_lift_joint: 1.5
      elbow_joint: 1.5
      wrist_1_joint: 1.5
      wrist_2_joint: 1.5
      wrist_3_joint: 1.5
```

gavanderhoorn, 2020-02-27:
I asked @jbeck28 to post this here.

Connects to the discussion in #119.

fmauch, 2020-03-05:
@jbeck28 

Without having a deeper look into the compliance_controller implementation: Can you estimate how well those optimizations can be transferred to a pure velocity controller?

I guess especially the weight issue is strongly connected to the compliance implementation?

jbeck28, 2020-03-09:
From my understanding the compliance_controller, it inherits from the joint_trajectory_controller, and therefore when compliance is disabled, it should behave exactly the same. 

Weight does make a big difference for compliance, but I did all this tuning without any compliance. I just figured weight might influence the choice of PID values, even for a more typical controller.

closed 2026-02-02 by urfeex