UniversalRobots/Universal_Robots_ROS_Driver #127: Hand guiding and reoording of path using UR ros driver

aloukas, 2020-03-03:
Is there a way to activate hand guiding using this driver and also record the path of  joints,TCP,forces/torques while teaching? Is it done through a specific topic or service?

I would also like to read the TCP positions. I can read the position of the joints through this topic: /scaled/trajectory_controller/state

DavidYaonanZhu, 2020-03-11:
Maybe implement a TCP damping control.

fmauch, 2020-03-27:
* We deliberately decided against activating hand guiding through this driver as it feels like a good idea to have to press a physical "acknowledge" button while doing so (as the hardware button on the TP or the button on the TP GUI).
 * To read the TCP positions you can make a normal TF lookup as in every other ROS system. Also as with every other ROS system joint states (position, velocity) is published on the `/joint_states` topic.
 * Force-torque information is also being published by default on the topic `/wrench`

closed 2021-08-09 by stefanscherzinger