UniversalRobots/Universal_Robots_ROS_Driver #128: Migration from URX robot driver to UR ROS driver

GhaithIdoudi, 2020-03-03:
# Summary

I was using URX robot driver on UR10 and I want to use the new UR ROS driver. So I want to know if I can use the same basic commands to move the robot as in the URX driver.

gavanderhoorn, 2020-03-03:
If by:

> if I can use the same basic commands to move the robot as in the URX driver

you mean: can I use commands such as "movel(..)" and "movej(..)", then no, that is not supported by this driver at this time.

Note that URX has a rather different way of interacting with a remote controller, compared to this driver. URX treats the controller more as a remote host exposing several services, while this driver models it as a controllable resource compatible with `ros_control`.

Future development may/will allow different use-cases with `ur_robot_driver`, but for now the two are rather different.

mdxtinkernick, 2020-03-11:
does that mean movel and movej are not supported over the /ur_hardware_interface/script_command topic - is there documentation of what can and can't be done through that interface now - I couldn't see it anywhere ?

I am trying to use code that previously worked with the old driver that sent a def ... with a series of movej commands inside - it is not currently working with the new driver and this morning I was trying to work out what needs to change to get it to work -does this mean it can't ?

fmauch, 2020-03-27:
The `/ur_hardware_interface/script_command` is only a relay of the actual socket connection to the robot, so yes, you can send any URScript code there. You just have to take care about filling your movej or movel commands by hand.

mat-l, 2020-04-09:
@fmauch and @mdxtinkernick: is there some kind of list with available commands that can be send via /ur_hardware_interface/script_command ?
I did not yet understand where and how to send a x, y, z position with a orientation a, b, c to the robot? Are there some examples and a list of commands somewhere?
Best regards

fmauch, 2020-04-14:
@mat-l Have a look ad the URScript manual that can be [downloaded at the UR website](https://www.universal-robots.com/download/?option=68442)

sunnymints, 2022-11-17:
> @fmauch and @mdxtinkernick: is there some kind of list with available commands that can be send via /ur_hardware_interface/script_command ?
> I did not yet understand where and how to send a x, y, z [...]

> The `/ur_hardware_interface/script_command` is only a relay of the actual socket connection to the robot, so yes, you can send any URScript code there. You just have to take care about filling your movej or movel commands by hand.

sorry, i can not execute'' rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'movel([0.5,0.5,0.5,3.14,-1.57,1.57])'" ''

fmauch, 2023-01-02:
@sunnymints The command you postet works for me. Did you consider the comment in the [topic documentation](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#script_command-std_msgsstring)?

On an e-Series robot the robot has to be in remote_control mode in order to accept commands. (Note that (at least at the moment) you should put the robot into remote_control mode **before** starting the ROS driver.)

closed 2023-05-18 by github-actions[bot]