UniversalRobots/Universal_Robots_ROS_Driver #129: How to get joints temperature

alfaDoG1993, 2020-03-03:
# Summary
What about getting the joints temperature?

At the moment I am working at the limit of ur3 capabilities and I would like to monitor the temperature of the joints and warn when overheating. In the polyscope panel, I see temperature data, but in no topic I found them. It seems to me that the `ur_modern_driver` was able to display this data.

# Versions
 - ROS Driver version: melodic
 - Affected Robot Software Version(s): 5.4.0.76181
 - Affected Robot Hardware Version(s): Ur3e
 - URCaps Software version(s): 1.0.1

# Impact
When the manipulator is operating in extreme conditions (the load is close to the maximum permissible), high heating has a bad effect on the inside of the robot (I think). Named the ability to track the temperature in growth, you can give the operator warnings about the need to interrupt and cool the manipulator.

## Expected Behavior
I think you can try to create a separate topic with temperature data for the hinges. And publish data only if a parameter such as "publish_ощштеы_temperature" is raised. But i think  the type of [temperature](http://docs.ros.org/api/sensor_msgs/html/msg/Temperature.html) message is not very suitable, since it is not an array. Maybe use [diagnostics](http://docs.ros.org/api/diagnostic_msgs/html/msg/DiagnosticStatus.html)?

## Actual Behavior
No joint temperature information available.

alfaDoG1993, 2020-03-04:
### Addition 
It would also be very useful to obtain current in the joints.

gavanderhoorn, 2020-03-04:
The current implementation does not publish this information.

That would be something that could be added (and I'm sure @fmauch would also refer to #35).

> But i think the type of [temperature](http://docs.ros.org/api/sensor_msgs/html/msg/Temperature.html) message is not very suitable, since it is not an array. Maybe use [diagnostics](http://docs.ros.org/api/diagnostic_msgs/html/msg/DiagnosticStatus.html)?

`sensor_msgs/Temperature` is actually a very good message for this, as its semantics perfectly match they type and form of data we're trying to publish here. Diagnostics messages are almost semanticless `key:value` pairs.

`sensor_msgs/Temperature` not being an array is not really a problem. We just need to publish more messages, one for each measurement. By giving them the same `stamp`, they can all be related to each other.

closed 2023-05-17 by github-actions[bot]