UniversalRobots/Universal_Robots_ROS_Driver #131: Publishing script commands doesn't yeild anything

Karthik-Ranga, 2020-03-04:
# Summary
The issue is related to #79  #77 
Not able to send any script command through rostopic 

Nothing happens after sending data through script_command

`rostopic pub /ur_hardware_interface/script_command std_gs/String""`

# Versions
 - ROS Driver version: ROS Melodic
 - UR+ product(s) installed: Robotiq Grippers
 - URCaps Software version(s): External Control 1.0.1

# Impact
Unable to send the script files

# Issue details
Sending below

`rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'set_digital_out(3,True)'" `

yields nothing. The external control program doesn't stop as said in #79 

## Project status at point of discovered
Found the problem when trying to send script commands to Robotiq gripper.

## Steps to Reproduce
After publishing the script_command,

Checked through
`rostopic echo ur_hardware_interface/script_command`

Received the information
`data: "set_digital_out(2,True)"`

## Expected Behavior
Expected the script file to execute the information

## Actual Behavior
The external control program did not stop and nothing happend

gavanderhoorn, 2020-03-04:
> The issue is related to #79 #77

Karthik-Ranga, 2020-03-04:
Agreed. it was also reported in #79 that the program was stopped once the script was sent. But in my case, nothing happened and no one has reported anything related to that so far. I guess it's 2 different problems but common root cause.

Karthik-Ranga, 2020-03-04:
Just FYI
Not sure if this would help. I got the below info when I checked it. 
```
rosnode info /ur_hardware_interface/script_command
--------------------------------------------------------------------------------
Node [/ur_hardware_interface/script_command]
Publications: None

Subscriptions: None

Services: None

cannot contact [/ur_hardware_interface/script_command]: unknown node
```
When I checked
`rosnode info /ur_hardware_interface`

under subscriptions I found,

```
Subscriptions: 
 * /ur_hardware_interface/script_command [unknown type]
```

fmauch, 2020-03-05:
`rosnode info /ur_hardware_interface/script_command`

The node is `/ur_hardware_interface` which is why a `rosnode info` on the actual topic doesn't yield anything. This is fine.

One important information is missing: Which robot model and polyscope version are you using? If you are on an e-Series robot you will have to put the robot into remote_control mode in order to make it accept remote script commands.

Karthik-Ranga, 2020-03-05:
> The node is `/ur_hardware_interface` which is why a `rosnode info` on the actual topic doesn't yield anything. This is fine.
> 
Okay understood.

> One important information is missing: Which robot model and polyscope version are you using? If you are on an e-Series robot you will have to put the robot into remote_control mode in order to make it accept remote script commands.

I am on e-series and the robot is in remote_control mode. But there is nothing happening on the robot side. 

But when I send the script file through a  socket(remote_control mode), the script gets executed.

DanielSchimpf, 2020-03-16:
I am also not able to use the script commands and for example set the I/O states with this command:
`rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'set_digital_out(3,True)'"`
I also use an e-version (UR 10e with Polyscope version 5.1). 
No matter if I am in manual, automatic or remote control mode or if the program with ExternalControl is running or not. The response in the log on the teach pendant to the before mentioned command is:
`RTMachine     CA210A0: Socket is read-only when the robot is in local (Teach pendant) control. `
So the robot seems to notice that it is receiving something... 

I am also able to change the I/O states through a socket with commands as such:
`echo "set_digital_out(0,True)" | netcat 192.168.56.1 30002`

fmauch, 2020-03-27:
Script code is indeed being sent to port 30002 which is hard coded [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/ur/ur_driver.cpp#L71).

Which is why this shouldn't make any difference to sending the code directly as done by @DanielSchimpf. The error `RTMachine CA210A0: Socket is read-only when the robot is in local (Teach pendant) control.` seems to indicate that the robot is indeed not in remote control mode...

To debug this further you have two levels of introspection. First of all, you could switch on DEBUG output for the driver node which will print a debug message when the script code has been sent to the socket successfully as defined [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/ur/ur_driver.cpp#L305).

To further debug we would have to have a look at the data actually being sent through the socket e.g. using [wireshark](https://wiki.wireshark.org/CaptureFilters).

jackie-wayland, 2020-09-28:
also not work here

GabrielGosden, 2020-11-04:
Hi,

I'll just leave a comment here if anyone has the same problem as @Karthik-Ranga .

According to [this forum post](https://forum.universal-robots.com/t/c210a0-socket-is-read-only-when-robot-is-in-local-teach-pendant-control/3815/2) the socket keeps it local/remote state from time of connect().

So to fix this i:

1. Enabled Remote control mode from TP.
2. Turned off robot. 
3. Turned on the robot.
4. Run UR_ROS_Driver

This ensures that the robot is in remote control mode from time of connect().

fmauch, 2020-11-06:
This is quite interesting. But would step 2 and 3 be really necessary? Restarting the driver should reestablish the socket connection, right?

GabrielGosden, 2020-11-06:
Hi,

Thanks for your comment @fmauch. I have researched this further and can now provide additional details. 

I have found that restart of the entire robot is not necessary. The important thing is ensuring connection is disconnected **before** entering remote mode from TP.

So:

1. Make sure physical connection is not established between robot and host computer (ethernet unplugged)
2. Enable Remote control mode from TP.
3. Establish physical connection between robot and host computer  (ethernet plugged in)
4. Run UR_ROS_Driver.
5. Send `rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'set_digital_out(3,True)'"`

> Restarting the driver should reestablish the socket connection, right?

So to answer your question: Restarting the driver does not reestablish the socket connection if inital connection was made while in manual mode.

fmauch, 2020-11-06:
Thanks for the clarification @GabrielGosden

akihikoy, 2021-02-18:
Hello,
I experienced the same error (`/ur_hardware_interface/script_command` does not work), and the solution by @GabrielGosden worked.  In my case, I started the ExternalControl urcap between 1 and 2:

1. Unplug Ethernet.
2. Run `ExternalControl` URCap on TP.
3. Enable remote control mode on TP.
4. Plug in Ethernet.
5. Run `UR_ROS_Driver`.
6. Send commands through `/ur_hardware_interface/script_command`.

Are there any updates about this issue?

closed 2023-05-15 by github-actions[bot]