UniversalRobots/Universal_Robots_ROS_Driver #133: Accepting socket connection failed

urielmtz, 2020-03-05:
# Summary

Hello,

I have a UR3 robot, which is controlled by a program running in a PC. They are connected using a wired connection and they are in the same network (robot: 192.168.56.10 and PC: 192.168.56.11), and the default script_sender_port 50002. On the robot side, I run the _External Control_, and on the PC, I run _roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=my_robot_ip_.

The communication works some sometimes, but then after some seconds I got the following message: _**Accepting socket connection failed. (errno: 22)**_. Then, the ur_robot_driver stops atuomatically and I have to stop the External Control, and run again the ur_robot_driver to restart the communication. Below you can see a picture of this issue.

[image]

Is there anything that maybe I am missing that causing this problem?
Any help will be appreciated.

Thanks

# Versions
 - ROS Driver version: 6f0f2c2
 - Affected Robot Software Version(s):
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed: UR3, PolyScope version 3.9
 - URCaps Software version(s): externalcontrol-1.0.urcap

gavanderhoorn, 2020-03-05:
Seeing that error message reminds me https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/112#issuecomment-592501651.

urielmtz, 2020-03-05:
Thanks for the quick reply and help.

I have initialised **addr_len**  to the size of **addr** and so far the communication has not stopped. Then, it seems that this has fixed my issue, but I will continue doing more tests.

Many thanks!

closed 2020-03-09 by urielmtz