UniversalRobots/Universal_Robots_ROS_Driver #134: Undefined parameters [kinematics_file] None None

DanielSchimpf, 2020-03-05:
# Summary

This may be a really simple problem, but I have not been able to figure it out. I have asked a similar question on ROS-answers, but I think that the question actually belongs here. I have been trying to follow these two tutorials, but with this new driver instead of the one used in the tutorials:
1. to create XACRO file representing a simple robot workcell
[](https://industrial-training-master.readthedocs.io/en/kinetic/_source/session3/Workcell-XACRO.html)
2. to build a moveit package
[](https://industrial-training-master.readthedocs.io/en/kinetic/_source/session3/Build-a-Moveit!-Package.html)

Going through both I am getting an error with : `Undefined parameters [kinematics_file] None None`

All I changed in the workcell.xacro tutorial is that I am using the description of the UR10e from the fmauch_universal_robot package instead of using the UR5 from the universal_robot package named in the tutorial. 

It seems like the kinematics file is not set, so I have tried to change the 32nd line in ur10e.urdf.xacro to name the robot calibration fail (.yalm) but it didn't solve the problem either. 

Why am I getting this error and what do I need to do to stop this error from occurring.

I am thankful for any advice!

Daniel

Below are the error outputs from 1. and 2.:

1.
```
dan@dan-Precision-7730:~/catkin_ws$ roslaunch grindcell_support urdf.launch ... logging to /home/dan/.ros/log/cfaa98cc-5f20-11ea-b042-98e74397a5e9/roslaunch-dan-Precision-7730-8241.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Undefined parameters [kinematics_file] None None
when instantiating macro: ur10e_robot (/home/dan/catkin_ws/src/fmauch_universal_robot/ur_e_description/urdf/ur10e.urdf.xacro)
in file: /home/dan/catkin_ws/src/grindcell_support/urdf/workcell.xacro
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/dan/catkin_ws/src/grindcell_support/urdf/workcell.xacro'] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro --inorder '$(find grindcell_support)/urdf/workcell.xacro'" name="robot_description"/>
The traceback for the exception was written to the log file
```

2.
```
dan@dan-Precision-7730:~/catkin_ws$ roslaunch moveit_setup_assistant setup_assistant.launch
... logging to /home/dan/.ros/log/9e1b01ac-5f20-11ea-b042-98e74397a5e9/roslaunch-dan-Precision-7730-8113.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dan-Precision-7730:35267/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

auto-starting new master
process[master]: started with pid [8123]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9e1b01ac-5f20-11ea-b042-98e74397a5e9
process[rosout-1]: started with pid [8136]
started core service [/rosout]
process[moveit_setup_assistant-2]: started with pid [8139]
[rospack] Error: no package given
[librospack]: error while executing command
Undefined parameters [kinematics_file] None None
when instantiating macro: ur10e_robot (/home/dan/catkin_ws/src/fmauch_universal_robot/ur_e_description/urdf/ur10e.urdf.xacro)
in file: /home/dan/catkin_ws/src/grindcell_support/urdf/workcell.xacro
^C[moveit_setup_assistant-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

yrchern, 2020-03-06:
I added the following line to the xacro file to address this issue:

<xacro:arg name="kinematics_config" default="your_ur_calibration_file.yaml"/>

DanielSchimpf, 2020-03-06:
So I canged the line:
  <xacro:arg name="kinematics_config" default="$(find ur_description)/config/ur10_default.yaml"/>
in ur10e_robot.urdf.xacro to 
<xacro:arg name="kinematics_config" default="$(find example_organization_ur_launch)/etc/ex-ur10-1_calibration.yaml"/>

it sadly does not change the error though... 
```
dan@dan-Precision-7730:~/industrial_ws$ roslaunch grindcell_support urdf.launch ... logging to /home/dan/.ros/log/67e32b08-5fd6-11ea-bd58-405bd8905c99/roslaunch-dan-Precision-7730-4576.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Undefined parameters [kinematics_file] None None
when instantiating macro: ur10e_robot (/home/dan/catkin_ws/src/fmauch_universal_robot/ur_e_description/urdf/ur10e.urdf.xacro)
in file: /home/dan/catkin_ws/src/grindcell_support/urdf/workcell.xacro
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/dan/catkin_ws/src/grindcell_support/urdf/workcell.xacro'] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro --inorder '$(find grindcell_support)/urdf/workcell.xacro'" name="robot_description"/>
The traceback for the exception was written to the log file
```

Or should I add the line to a different xacro file, which one?

Thanks!

DanielSchimpf, 2020-03-06:
Please correct me if I am wrong here, I am trying to work through this problem:
The workcell.xacro file from the tutorial calls ur10e.urdf.xacro.
with the line:
`  <xacro:include filename="$(find ur_e_description)/urdf/ur10e.urdf.xacro" />`
In the ur10e.urdf.xacro calls ur10e_robot.urdf.xacro with the lines:
```
 <xacro:macro name="ur10e_robot" params="prefix joint_limited
...
    kinematics_file">
```
 ur10e_robot.urdf.xacro then seems to set the kinematics_file in lines:
```
  <xacro:arg name="kinematics_config" default="$(find example_organization_ur_launch)/etc/ex-ur10-1_calibration.yaml"/>
  <xacro:ur10e_robot prefix="" joint_limited="false"
    kinematics_file="${load_yaml('$(arg kinematics_config)')}"
  />
```

This doesn't seem to transfer to the ur10e.urdf.xacro. 

How can I test if lines in  ur10e_robot.urdf.xacro are able to call the .yalm config file?

Thanks!

yrchern, 2020-03-07:
Add that line to workcell.xacro before loading ur10e.urdf.xacro.

DanielSchimpf, 2020-03-09:
The error stayed the same.
Is there maybe something else wrong in my workcell.xacro file? I will paste it below.

```
<?xml version="1.0" ?>
<robot name="grindcell" xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:arg name="kinematics_config" default="$(find example_organization_ur_launch)/etc/ex-ur10-1_calibration.yaml"/>
  <xacro:include filename="$(find ur_e_description)/urdf/ur10e.urdf.xacro" />
  <link name="world"/>

  <link name="table">
    <visual>
      <geometry>
        <box size="1.0 1.0 0.05"/>
      </geometry>
    </visual>
    <collision>
      <geometry>
        <box size="1.0 1.0 0.05"/>
      </geometry>
    </collision>
  </link>

  <link name="camera_frame"/>
  <xacro:ur10e_robot prefix="" joint_limited="true"/>

  <joint name="world_to_table" type="fixed">
    <parent link="world"/>
    <child link="table"/>
    <origin xyz="0 0 0.5" rpy="0 0 0"/>
  </joint>

  <joint name="world_to_camera" type="fixed">
    <parent link="world"/>
    <child link="camera_frame"/>
    <origin xyz="-0.25 -0.5 1.25" rpy="0 3.14159 0"/>
  </joint>

  <joint name="table_to_robot" type="fixed">
    <parent link="table"/>
    <child link="base_link"/>
    <origin xyz="0 0 0" rpy="0 0 0"/>
  </joint>
</robot>
```

Thanks!

yrchern, 2020-03-10:
Sorry, forgot to mention that you need to add the argument to the robot as well:

> <xacro:ur10e_robot prefix="" joint_limited="true" **kinematics_file="${load_yaml('$(arg kinematics_config)')}"**/>

DanielSchimpf, 2020-03-10:
That worked!!
I changed  
` <xacro:ur10e_robot prefix="" joint_limited="true"/> `
to 
`<xacro:ur10e_robot prefix="" joint_limited="true" kinematics_file="${load_yaml('$(arg kinematics_config)')}"/>`

Thank you very much yrchern!!! This can now be closed.

closed 2020-03-10 by DanielSchimpf

jwhendy, 2020-10-27:
Is there a way to debut which kinematics file is being used? I get the error that the kinematics file loaded does not match the robot, but have traced my bringup chain and _think_ that I am doing all that I see in the driver here.

In our moveit setup, we have:
```
<include file="$(find ur_robot_driver)/launch/ur10_bringup.launch">
  <arg name="robot_description_file" value="$(find custom_pkg)/launch/ur10_custom.launch"/>
</include>
```
`ur10_bringup.launch` calls `ur_common.launch`, which calls our custom `robot_description_file`, which contains:
```
<launch>
  <arg name="limited" default="false"/>
  <arg name="kinematics_config" default="$(find custom_pkg)/config/robot_calibration.yaml"/>
  <param name="robot_description"
        command="$(find xacro)/xacro --inorder '$(find custom_pkg)/urdf/custom_ur10.xacro' kinematics_config:=$(arg kinematics_config)"/>
</launch>
```

And then finally in `custom_ur10.xacro` we have:
```
<xacro:include filename="$(find ur_description)/urdf/ur10.urdf.xacro"/> 
<xacro:arg name="kinematics_config"
    default="$(find custom_pkg)/config/robot_calibration.yaml"/>
<xacro:ur10_robot prefix="${prefix}" joint_limited="false"
    kinematics_file="${load_yaml('$(arg kinematics_config)')}"/>
```

Does anything stand out as wrong here? I can open a new issue... this is just so similar I thought I could piggyback, as it's likely some oddity/typo I haven't seen despite pouring over these numerous times. It works, the error just bothers me.
```
The calibration parameters of the connected robot don't match the ones from the given kinematics config file. 
```

As an alternative, I'm wondering how I could verify easily which file it's loading?

fmauch, 2020-10-27:
The `kinematics_config` is also used [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/launch/ur_control.launch#L36) which is used for extracting the checksum which produces the error above. So, in order to get rid of this you'll also have to tell the driver about your custom `kinematics_config` file.

jwhendy, 2020-10-27:
@fmauch ahhh, that is supremely helpful. So we are passing this too "low" in the chain, correct?

Namely, we call `ur10_bringup.launch`, which calls both `ur_common.launch` (which knows about `kinematics_config` via our custom `robot_description`), but the default is not overridden when subsequently calling `ur_control.launch` below?

I'm not able to test on hardware until later today, but would I simply move the arg to our `moveit_planning_execution.launch` instead?
```
<include file="$(find ur_robot_driver)/launch/ur10_bringup.launch">
  <arg name="robot_description_file" value="$(find custom_pkg)/launch/ur10_custom.launch"/>
  <arg name="kinematics_config" default="$(find custom_pkg)/config/robot_calibration.yaml"/>
</include>
```

fmauch, 2020-10-27:
> I'm not able to test on hardware until later today, but would I simply move the arg to our `moveit_planning_execution.launch` instead?

Yes, this looks correct.

jwhendy, 2020-10-27:
Thanks @fmauch ! I can confirm that adding that line changed the values on the param server from defaults to the contents of my actual calibration file (and the associated hash). I wasn't able to test on hardware, but am sure this was the issue. Many thanks for the quick assistance.

xrihub, 2020-11-18:
Add this tag and all will be fine

<xacro:include filename="$(find ur_e_description)/urdf/ur10_robot.urdf.xacro"/>

jwhendy, 2020-11-18:
@xrihub I think that will still be insufficient. See my [post above](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/134#issuecomment-716939065), where we already include this `xacro`, and [my attempt to explain](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/134#issuecomment-717188397) where we were still missing this arg in the launch change thanks to insights from @fmauch . Let me know if that clarifies.