UniversalRobots/Universal_Robots_ROS_Driver #135: Could not get fresh data package from robot

mbaat, 2020-03-06:
When I connect to URSim (3.12) running UR10 with the ur_robot_driver I continuously get error messages:

[ INFO] [1583503125.382218862]: Connection to robot dropped, waiting for new connection.
[ INFO] [1583503125.399099714]: Robot ready to receive control commands.

I am able to communicate with the robot (via External Control) but the connection is constantly dropped and reestablished.

Command run:
`roslaunch ur_robot_driver ur10_bringup.launch robot_ip:=<robot-ip>`

URSim is running on a VM running Ubuntu 16.04.

gavanderhoorn, 2020-03-09:
Did you figure this out?

mbaat, 2020-03-09:
Hello

The erfor seems to en gone today. Not sure why but it has solved itself.

Venlig hilsen / Best regards

Mads Vainø Baatrup
Konsulent
Produktion og Innovation
+45 72 20 10 67<tel:+45%2072%2020%2010%2067>

Teknologisk Institut
Forskerparken 10F, 5230 Odense<x-apple-data-detectors://1>
www.teknologisk.dk<http://www.teknologisk.dk/>

mbaat, 2020-03-09:
My mistake. It seems to be back again.

fmauch, 2020-03-27:
Are you interacting with the simulation in any way? (e.g. manually driving the robot around)

If not, posting logs from the teach panel might be interesting. There, it should say why connection is being dropped.

casmarmik, 2020-07-02:
Hi,
I've been having the same problem. I am using URSim 5.8.2 running a UR3e. After a while the ur_robot_driver continually gives the error:

[ INFO] [1593691033.682445811]: Robot ready to receive control commands. 
[ERROR] [1593691033.749678600]: Could not get fresh data package from robot
[ERROR] [1593691033.849838099]: Could not get fresh data package from robot
[ INFO] [1593691033.879291149]: Connection to robot dropped, waiting for new connection.
[ INFO] [1593691033.883376555]: Robot ready to receive control commands.
[ERROR] [1593691033.949998866]: Could not get fresh data package from robot
[ERROR] [1593691034.050166200]: Could not get fresh data package from robot
[ INFO] [1593691034.080496004]: Connection to robot dropped, waiting for new connection.

I have saved the support file from the teach pendant and added it here: 
[robot_configuration20200623_13_48_07.zip](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/4864119/robot_configuration20200623_13_48_07.zip)

Hope this helps

stefanscherzinger, 2021-08-12:
@mbaat Is this still relevant with the most recent driver version?

closed 2023-05-14 by github-actions[bot]