UniversalRobots/Universal_Robots_ROS_Driver #137: Increasing publish rate of /joint_states

macsmitty, 2020-03-13:
Hi everyone, is there a way to increase the publish rate of the joint_states? I already changed &loop_hz in the controller.yaml for all controllers as well as for only the joint_state_controller but it does not work. I want to compare dynamics behavior in a mujoco simulation and the physical robot and therefore have to be able to read the joint angles with high frequency. 

I appreciate your help!

gavanderhoorn, 2020-03-13:
The controllers are "physically" limited to 125 and 500 Hz (e-Series), and the controller controls the control flow of the driver.

So I would say, "no", you cannot increase this rate, as it's already at the maximum.

macsmitty, 2020-03-15:
Thank you very much for the information.

closed 2020-03-15 by macsmitty