UniversalRobots/Universal_Robots_ROS_Driver #140: Why can't the driver use the extracted calibration info on startup?

carebare47, 2020-03-23:
# Summary

Currently, there is a two-step process for running an arm with calibration correction. First, you run `roslaunch ur_calibration calibration_correction.launch robot_ip:=<robot_ip> output_filename:=/home/user/calibration.yaml` to extract the calibration information. 

Then, to use an arm with calibration correction, you'd run `roslaunch ur_robot_launch ur10_bringup.launch robot_ip:=<robot_ip> kinematics_config:=/home/user/calibration.yaml`

From what I can tell, the driver then checks that the `calibration.yaml` file (generated in step 1) is for the correct arm by extracting the calibration information from the arm again, creating a checksum by concatenating these values, and comparing that checksum with the checksum in the `calibration.yaml` file. 

My question is, if the checksum that the driver creates to validate the calibration file is generated from freshly extracted calibration information, why can't this information be used to automatically apply calibration correction to the arm when it starts? Why the need for the two step process?

gavanderhoorn, 2020-03-23:
Because parameters are loaded on the parameter server *before* any nodes are started.

If you have consumers of your urdf and you'd like those consumers to use the "calibrated urdf", you cannot update it after it has been loaded onto the parameter server, as you cannot guarantee all consumers then see the same version.

fmauch, 2020-03-27:
Thanks @gavanderhoorn for this excellent explanation. Would you be ok with me copying this to the README's troubleshooting section?

gavanderhoorn, 2020-03-27:
Sure. I would reword it a little though. Reading back my own comment it's got too many alternatives and conditionals in there.

PR #179 (2020-05-20): Improve documentation
This MR aims at collecting a couple of documentation improvements.

Closes #140 

Closes #38  

Closes #51
closed 2020-05-20 by fmauch