UniversalRobots/Universal_Robots_ROS_Driver #142: catkin_make error

saalimperator, 2020-03-26:
# Summary

I am trying to install the latest ROS_Drivers. But while following the instructions in the building section, I encounterd the following error message after using catkin_make.

# Impact
Cant build ros-Drivers

# Issue details
CMake Error at fmauch_universal_robot/ur_bringup/CMakeLists.txt:10 (find_package):
  By not providing "Findroslaunch.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "roslaunch", but CMake did not find one.

  Could not find a package configuration file provided by "roslaunch" with
  any of the following names:

    roslaunchConfig.cmake
    roslaunch-config.cmake

  Add the installation prefix of "roslaunch" to CMAKE_PREFIX_PATH or set
  "roslaunch_DIR" to a directory containing one of the above files.  If
  "roslaunch" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/imperator/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/imperator/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

## Steps to Reproduce
Clean Ubuntu 18.04
New Installed ros-melodic
Instructions in the 'Building'-section followed

gavanderhoorn, 2020-03-26:
Could you show the output of:

```
rosdep check --from-paths /home/imperator/catkin_ws/src --ignore-src
```

saalimperator, 2020-03-26:
of course:
______________________________________________________
dpkg-query: no packages found matching ros-melodic-rviz
dpkg-query: no packages found matching ros-melodic-controller-manager
dpkg-query: no packages found matching ros-melodic-joint-state-controller
dpkg-query: no packages found matching ros-melodic-controller-manager-msgs
dpkg-query: no packages found matching ros-melodic-moveit-fake-controller-manager
dpkg-query: no packages found matching ros-melodic-moveit-core
dpkg-query: no packages found matching ros-melodic-actionlib
dpkg-query: no packages found matching socat
dpkg-query: no packages found matching ros-melodic-velocity-controllers
dpkg-query: no packages found matching ros-melodic-robot-state-publisher
dpkg-query: no packages found matching ros-melodic-joint-state-publisher
dpkg-query: no packages found matching ros-melodic-joint-trajectory-controller
dpkg-query: no packages found matching ros-melodic-tf
dpkg-query: no packages found matching ros-melodic-pluginlib
dpkg-query: no packages found matching ros-melodic-xacro
dpkg-query: no packages found matching ros-melodic-moveit-simple-controller-manager
dpkg-query: no packages found matching ros-melodic-moveit-ros-planning
dpkg-query: no packages found matching ros-melodic-rostopic
dpkg-query: no packages found matching ros-melodic-moveit-ros-move-group
dpkg-query: no packages found matching ros-melodic-dynamic-reconfigure
dpkg-query: no packages found matching ros-melodic-moveit-planners-ompl
dpkg-query: no packages found matching ros-melodic-roslaunch
dpkg-query: no packages found matching ros-melodic-urdf
dpkg-query: no packages found matching ros-melodic-force-torque-sensor-controller
dpkg-query: no packages found matching ros-melodic-moveit-kinematics
dpkg-query: no packages found matching ros-melodic-tf-conversions
dpkg-query: no packages found matching ros-melodic-tf2-ros
dpkg-query: no packages found matching ros-melodic-tf2-geometry-msgs
dpkg-query: no packages found matching ros-melodic-gazebo-ros-control
dpkg-query: no packages found matching ros-melodic-gazebo-ros
dpkg-query: no packages found matching ros-melodic-effort-controllers
dpkg-query: no packages found matching ros-melodic-rosunit
dpkg-query: no packages found matching ros-melodic-realtime-tools
dpkg-query: no packages found matching ros-melodic-moveit-ros-visualization
dpkg-query: no packages found matching ros-melodic-position-controllers
dpkg-query: no packages found matching ros-melodic-rospy
System dependencies have not been satisified:
apt	ros-melodic-rviz
apt	ros-melodic-controller-manager
apt	ros-melodic-joint-state-controller
apt	ros-melodic-controller-manager-msgs
apt	ros-melodic-moveit-fake-controller-manager
apt	ros-melodic-moveit-core
apt	ros-melodic-actionlib
apt	socat
apt	ros-melodic-velocity-controllers
apt	ros-melodic-robot-state-publisher
apt	ros-melodic-joint-state-publisher
apt	ros-melodic-joint-trajectory-controller
apt	ros-melodic-tf
apt	ros-melodic-pluginlib
apt	ros-melodic-xacro
apt	ros-melodic-moveit-simple-controller-manager
apt	ros-melodic-moveit-ros-planning
apt	ros-melodic-rostopic
apt	ros-melodic-moveit-ros-move-group
apt	ros-melodic-dynamic-reconfigure
apt	ros-melodic-moveit-planners-ompl
apt	ros-melodic-roslaunch
apt	ros-melodic-urdf
apt	ros-melodic-force-torque-sensor-controller
apt	ros-melodic-moveit-kinematics
apt	ros-melodic-tf2-ros
apt	ros-melodic-realtime-tools
apt	ros-melodic-gazebo-ros-control
apt	ros-melodic-gazebo-ros
apt	ros-melodic-effort-controllers
apt	ros-melodic-rosunit
apt	ros-melodic-tf2-geometry-msgs
apt	ros-melodic-rospy
apt	ros-melodic-moveit-ros-visualization
apt	ros-melodic-position-controllers
apt	ros-melodic-tf-conversions
ERROR[ur_robot_driver]: No definition of [boost] for OS version [bionic]
	rosdep key : boost
	OS name    : ubuntu
	OS version : bionic
	Data: alpine:
- boost-dev
arch:
- boost
cygwin:
- libboost-devel
- libboost1.40
debian:
  buster:
  - libboost-all-dev
  jessie:
  - libboost-all-dev
  squeeze:
  - libboost1.42-all-dev
  stretch:
  - libboost-all-dev
  wheezy:
  - libboost-all-dev
fedora:
- boost-devel
- boost-python2-devel
- boost-python3-devel
freebsd:
- py27-boost-libs
gentoo:
- dev-libs/boost[python]
macports:
- boost
openembedded:
- boost@openembedded-core
opensuse:
- boost-devel
rhel:
- boost-devel
slackware:
  slackpkg:
    packages:
    - boost
ubuntu:
  '*':
  - libboost-all-dev
  lucid:
  - libboost1.40-all-dev
  maverick:
  - libboost1.42-all-dev
  natty:
  - libboost1.42-all-dev
  oneiric:
  - libboost1.46-all-dev
  trusty_python3:
  - libboost-all-dev

ERROR[ur_kinematics]: No definition of [boost] for OS version [bionic]
	rosdep key : boost
	OS name    : ubuntu
	OS version : bionic
	Data: alpine:
- boost-dev
arch:
- boost
cygwin:
- libboost-devel
- libboost1.40
debian:
  buster:
  - libboost-all-dev
  jessie:
  - libboost-all-dev
  squeeze:
  - libboost1.42-all-dev
  stretch:
  - libboost-all-dev
  wheezy:
  - libboost-all-dev
fedora:
- boost-devel
- boost-python2-devel
- boost-python3-devel
freebsd:
- py27-boost-libs
gentoo:
- dev-libs/boost[python]
macports:
- boost
openembedded:
- boost@openembedded-core
opensuse:
- boost-devel
rhel:
- boost-devel
slackware:
  slackpkg:
    packages:
    - boost
ubuntu:
  '*':
  - libboost-all-dev
  lucid:
  - libboost1.40-all-dev
  maverick:
  - libboost1.42-all-dev
  natty:
  - libboost1.42-all-dev
  oneiric:
  - libboost1.46-all-dev
  trusty_python3:
  - libboost-all-dev

gavanderhoorn, 2020-03-26:
Just making sure. Could you run this command:

```
rosdep check --from-paths /home/imperator/catkin_ws/src --ignore-src --rosdistro=melodic
```

If that still returns such a long list, you'll want to run:

```
rosdep update
rosdep install --from-paths /home/imperator/catkin_ws/src --ignore-src -y
```

again.

It would seem `rosdep install` never actually ran successfully on your system.

Probably because of #143.

saalimperator, 2020-03-26:
okay I tried that, but the error still occurs. `rosdep install [...]` gives me following output:
_____________________________________________________
dpkg-query: no packages found matching ros-melodic-rviz
dpkg-query: no packages found matching ros-melodic-controller-manager
dpkg-query: no packages found matching ros-melodic-joint-state-controller
dpkg-query: no packages found matching ros-melodic-controller-manager-msgs
dpkg-query: no packages found matching ros-melodic-moveit-fake-controller-manager
dpkg-query: no packages found matching ros-melodic-moveit-core
dpkg-query: no packages found matching ros-melodic-actionlib
dpkg-query: no packages found matching socat
dpkg-query: no packages found matching ros-melodic-velocity-controllers
dpkg-query: no packages found matching ros-melodic-robot-state-publisher
dpkg-query: no packages found matching ros-melodic-joint-state-publisher
dpkg-query: no packages found matching ros-melodic-joint-trajectory-controller
dpkg-query: no packages found matching ros-melodic-tf
dpkg-query: no packages found matching ros-melodic-pluginlib
dpkg-query: no packages found matching ros-melodic-xacro
dpkg-query: no packages found matching ros-melodic-moveit-simple-controller-manager
dpkg-query: no packages found matching ros-melodic-moveit-ros-planning
dpkg-query: no packages found matching ros-melodic-rostopic
dpkg-query: no packages found matching ros-melodic-moveit-ros-move-group
dpkg-query: no packages found matching ros-melodic-dynamic-reconfigure
dpkg-query: no packages found matching ros-melodic-moveit-planners-ompl
dpkg-query: no packages found matching ros-melodic-roslaunch
dpkg-query: no packages found matching ros-melodic-urdf
dpkg-query: no packages found matching ros-melodic-force-torque-sensor-controller
dpkg-query: no packages found matching ros-melodic-moveit-kinematics
dpkg-query: no packages found matching ros-melodic-tf-conversions
dpkg-query: no packages found matching ros-melodic-tf2-ros
dpkg-query: no packages found matching ros-melodic-tf2-geometry-msgs
dpkg-query: no packages found matching ros-melodic-gazebo-ros-control
dpkg-query: no packages found matching ros-melodic-gazebo-ros
dpkg-query: no packages found matching ros-melodic-effort-controllers
dpkg-query: no packages found matching ros-melodic-rosunit
dpkg-query: no packages found matching ros-melodic-realtime-tools
dpkg-query: no packages found matching ros-melodic-moveit-ros-visualization
dpkg-query: no packages found matching ros-melodic-position-controllers
dpkg-query: no packages found matching ros-melodic-rospy
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_robot_driver: No definition of [boost] for OS version [bionic]
ur_kinematics: No definition of [boost] for OS version [bionic]

gavanderhoorn, 2020-03-26:
What's the output of:

```
rosdep resolve boost --rosdistro=melodic
```

and:

```
python /usr/lib/python2.7/dist-packages/rospkg/os_detect.py
```

saalimperator, 2020-03-26:
> What's the output of:
> 
> ```
> rosdep resolve boost --rosdistro=melodic
> ```
________________
ERROR: No definition of [boost] for OS version [bionic]

No definition of [boost] for OS version [bionic]
	rosdep key : boost
	OS name    : ubuntu
	OS version : bionic
	Data: alpine:
- boost-dev
arch:
- boost
cygwin:
- libboost-devel
- libboost1.40
debian:
  buster:
  - libboost-all-dev
  jessie:
  - libboost-all-dev
  squeeze:
  - libboost1.42-all-dev
  stretch:
  - libboost-all-dev
  wheezy:
  - libboost-all-dev
fedora:
- boost-devel
- boost-python2-devel
- boost-python3-devel
freebsd:
- py27-boost-libs
gentoo:
- dev-libs/boost[python]
macports:
- boost
openembedded:
- boost@openembedded-core
opensuse:
- boost-devel
rhel:
- boost-devel
slackware:
  slackpkg:
    packages:
    - boost
ubuntu:
  '*':
  - libboost-all-dev
  lucid:
  - libboost1.40-all-dev
  maverick:
  - libboost1.42-all-dev
  natty:
  - libboost1.42-all-dev
  oneiric:
  - libboost1.46-all-dev
  trusty_python3:
  - libboost-all-dev
________________
> and:
> 
> ```
> python /usr/lib/python2.7/dist-packages/rospkg/os_detect.py
> ```
______
OS Name:     ubuntu
OS Version:  18.04
OS Codename: bionic
__________________

gavanderhoorn, 2020-03-26:
Ok, so it would seem there is something not entirely ok with the way `rosdep` identifies your OS. Or at the very least it's unable to match the codename with the `rosdistro` db.

Please note that this is not a problem with the packages in this repository, but a general problem. As such, I would recommend you post an question on ROS Answers.

As to the dependency issues here, run this:

```
rosdep install --from-paths /home/imperator/catkin_ws/src --ignore-src -y -r --rosdistro=melodic
```

note that `-r` that was added, which makes `rosdep` ignore the fact it cannot resolve the `boost` key.

As `boost` is almost certainly already installed on your system (it's a dependency of ROS itself), I'm pretty sure you can start building your workspace after running the `rosdep install ..` command line above.

saalimperator, 2020-03-26:
You were right. Thank you!

closed 2020-03-26 by saalimperator

freesolo92, 2021-07-05:
Hey guys,

I have a similar problem with cmake if I try to build the driver with catkin_make_isolated (from the current repo). I followed the instructions in the readme for building all sources.

Output Log after "catkin_make_isolated":

CMakeFiles/ur_robot_driver_node.dir/src/hardware_interface.cpp.o: In function `ur_driver::HardwareInterface::init(ros::NodeHandle&, ros::NodeHandle&)':
/home/jetson/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:265: undefined reference to `urcl::UrDriver::UrDriver(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (bool)>, bool, std::unique_ptr<urcl::ToolCommSetup, std::default_delete<urcl::ToolCommSetup> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, int, double, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/ur_robot_driver_node.dir/build.make:193: recipe for target '/home/jetson/catkin_ws/devel_isolated/ur_robot_driver/lib/ur_robot_driver/ur_robot_driver_node' failed
make[2]: *** [/home/jetson/catkin_ws/devel_isolated/ur_robot_driver/lib/ur_robot_driver/ur_robot_driver_node] Error 1
CMakeFiles/Makefile2:498: recipe for target 'CMakeFiles/ur_robot_driver_node.dir/all' failed
make[1]: *** [CMakeFiles/ur_robot_driver_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'ur_robot_driver':
  Command '['/home/jetson/catkin_ws/devel_isolated/ur_kinematics/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/jetson/catkin_ws/build_isolated/ur_robot_driver && /home/jetson/catkin_ws/devel_isolated/ur_kinematics/env.sh make -j4 -l4

Command failed, exiting.

I actually don't know how to fix this issue to work with the driver. Maybe someone can give me a hint to solve the problem.
All ROS dependencys are just fine. I got no errors till "catkin_make_isolated". My OS is a Ubuntu 18.04 Bionic on a Jetson Nano.
ROS1 works alright. 

Greetings, Marcel

fmauch, 2021-07-06:
@freesolo92 that could be related to an installed version of the library being used for the driver. As you use `catkin_make_isolated` I assume that you also build the client library from source? In that case you could try to remove the binary installed version `sudo apt remove ros-melodic-ur-client-library`.