UniversalRobots/Universal_Robots_ROS_Driver #144: Switch Travis integration to the app?

gavanderhoorn, 2020-03-26:
Just noting this here as a TODO: Travis migrated from the old integration to the app.

I believe recent PRs don't get CI run any more as the old integration has been turned off/disabled.

fmauch, 2020-03-27:
Thanks for notifying!

gavanderhoorn, 2020-03-27:
Could also have been the recent problems Travis has reported: https://www.traviscistatus.com/ (search for "March 26").

fmauch, 2020-03-27:
As for #148 the build has succeeded on [travis.org](https://travis-ci.org/github/UniversalRobots/Universal_Robots_ROS_Driver/builds/667627387), but it does not show up on GH.

It might be a good opportunity to migrate towards GH actions to reduce the number of services involved...

fmauch, 2020-03-30:
Changed travis integration, so it is working again.

closed 2020-03-30 by fmauch