UniversalRobots/Universal_Robots_ROS_Driver #147: Can't start ur_robot_driver on the terminal

tonkei0361, 2020-03-26:
# Versions
 - Affected Robot Software Version(s): 5.3.0.64176
 - URCaps Software version(s): 1.0.1

# Issue details
I have a UR5e robot and have followed the instruction on this repository, including the installation of URCaps to include external control feature. The robot is connected to the host computer via Ethernet cable and I set the robot's IP address to the IP address of this Ethernet connection. However, when I do 

`$ roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=myipaddress`

the driver fails to start with these errors on the terminal:

[image]

What I don't understand is that after starting urdriver, the port following the IP address is 30002, not 50002. Could this be the issue? If so, how can I make this correct?

Thank you!

gavanderhoorn, 2020-03-29:
> The robot is connected to the host computer via Ethernet cable and I set the robot's IP address to the IP address of this Ethernet connection

The robot controller and your PC should each have their own IP address. You cannot give two devices on the same network the same IP address.

tonkei0361, 2020-03-30:
> The robot controller and your PC should each have their own IP address. You cannot give two devices on the same network the same IP address.

Thank you, I messed up the robot's network IP address and never use the correct IP address for parameter `robot_ip` on ROS terminal. The robot is now connected.

closed 2020-03-30 by tonkei0361