UniversalRobots/Universal_Robots_ROS_Driver #149: UR and URe series cannot be used alongside each other

felixvd, 2020-03-27:
# Summary

The macros `ur_transmission` and `ur_gazebo` are defined in both `ur_description` and `ur_e_description`, causing errors if both UR and URe robots are used in the same scene.

I solved this by adding the `_e` prefix to the files/macros in `ur_e_description`. I can file a PR if that's a solution that you want to use.

# Versions
 - ROS Driver version: Melodic
 - Affected Robot Software Version(s): All
 - Affected Robot Hardware Version(s): Old + e-series

# Impact
It affects scenes where multiple generations of UR robots are used.

# Issue details
The scene cannot be constructed.

## Use Case and Setup
E.g.: A lab bought a UR5 and later a UR5e, and now wants to use them in a dual-arm setup.

## Project status at point of discovered
I was setting up the URDF.

## Steps to Reproduce
Try to add a UR5 and UR5e to a URDF.

## Expected Behavior
The URDF and the xacro macros should load.

## Actual Behavior
The `ur_transmission` and `ur_gazebo` macros take different input parameters in each generation, but they use the same name. This prevents the URDF from being loaded.

## Workaround Suggestion
As described above, simply change the name of the macros.

gavanderhoorn, 2020-03-27:
`xacro` supports namespaces these days.

Could you try moving the `xacro:include` of the macros you mention some lines down, so they are inside the `xacro:macro` definition?

felixvd, 2020-03-29:
I just tried to move the includes into the robot macros but it didn't work on my machine.

fmauch, 2020-03-30:
Thanks for reporting this.

Technically, this is the wrong repo, as it is related to https://github.com/fmauch/universal_robot.

As the `_e_` packages get removed inside https://github.com/ros-industrial/universal_robot/pull/371, I will however not put any work into that currently.

You can try using the `calibration` branch of my fork which is built ontop of https://github.com/ros-industrial/universal_robot/pull/371

felixvd, 2020-04-02:
That looks clean! I couldn't try it yet, but it seems like it won't be an issue after that reorganization. I'll close this.

closed 2020-04-02 by felixvd