UniversalRobots/Universal_Robots_ROS_Driver #151: Consider lowering action server state reporting frequency

gavanderhoorn, 2020-03-30:
@fmauch wrote in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/138#issuecomment-605828287:
> I think the `state_publish_rate` is the same as the control loop rate only for legacy reasons as this was the same in the `ur_modern_driver` for CBx series and every 125 Hz was replaced by 500Hz and later by `*loop_hz` without putting too much thought into that...

We might want to consider lowering the `state_publish_rate` of the various actionlib based controllers.

As these are typically long(er) running actions with more discrete state changes, publishing their state (ie: those of the actions, not of the robot) at 500 Hz seems wasteful (as the state has probably not changed sufficiently to be of interest).

gavanderhoorn, 2020-07-07:
I can quickly fix this, but a bit of discussion about a rate would be good.

The default is 25 Hz I believe.

Would 50 Hz be acceptable here?

It would still be something users could change, would they so desire.

fmauch, 2020-07-07:
According to [the docs](http://wiki.ros.org/joint_trajectory_controller#Parameters) the default value is 50 Hz. I think, it would be fine to stick with that.

gavanderhoorn, 2020-07-07:
> According to [the docs](http://wiki.ros.org/joint_trajectory_controller#Parameters) the default value is 50 Hz. I think, it would be fine to stick with that.

It is: [ros-controls/ros_controllers/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h](https://github.com/ros-controls/ros_controllers/blob/1fd8d25707e289700743925c93644b4295b1c486/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L218).

I'll set it to that value.

gavanderhoorn, 2020-07-07:
See #221.

fmauch, 2020-07-07:
Closed via #221

closed 2020-07-07 by fmauch