UniversalRobots/Universal_Robots_ROS_Driver #153: UR5e: Function equivalent to 'Freedrive' on this driver

tonkei0361, 2020-04-02:
# Versions
 - ROS Driver version: Melodic
 - Affected Robot Hardware Version(s): UR5e
 - URCaps Software version(s):1.0.1

# Issue details
Does current drive have a controller that achieves same/similar function to the Freedrive mode in TP? If there is one, how to use that controller? `force_torque_sensor_controller` seems like the right controller to use but there is no topic/service/msg to communicate with this controller.

fmauch, 2020-04-03:
No, such a thing does not exist. We deliberately decided against adding a freedrive option as in our opinion that carries too much risk as the robot will be falling down if the tool setup is not done correctly.

The `force_torque_sensor_controller` is not an "active" controller, it rather reads force-torque sensor information.

That being said, you can use the scripting interface on `/ur_hardware_interface/script_command` and use the URScript functions `freedrive_mode()` and `end_freedrive_mode()` there. Be aware of the risks mentioned above, though. Deactivating freedrive is not as simple as letting go of a button, but you'll have to send another command.

P.S.: When using the script interface, make sure the robot is actually in remote control mode as otherwise commands sent to the scripting interface will be ignored by the robot.

closed 2020-04-03 by fmauch

fmauch, 2020-04-03:
Update: I just tried it out. It seems as if freedrive mode gets stopped as soon as the program execution stops (which does make sense, though). So, in order to really make use of that, you have to send a program with multiple lines with a command after the 'freedrive_mode()' command. e.g.

```
$ rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'def prog():

  freedrive_mode()

  sleep(10)

end'"
```

To make it stay active for an infinite time, you could use a `while(true)` loop instead of the `sleep()` command. Then, you would have to send any other command to overwrite the currently running script. (Note: I haven't tested this).

tonkei0361, 2020-04-03:
@fmauch Thank you so much for thorough answers! It's good to know that there is still a method to achieve freedrive mode using the remote control.

shuobh, 2022-03-09:
> 

I am testing this method but it doesn't seem to work on my end.
The way I did is first launch UR driver by:
`roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.2.2`
and then ran the remote script on UR such that rqt can control the robot.
However, nothing changes by running:
`rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'def prog():
  while(true)
    freedrive_mode()
end'"`
There are also no additional printout on the UR driver tab.
I also tried stopping the remote script on TP but had no effect as well.

fmauch, 2022-03-10:
As mentioned above, the robot has to be in `remote_control` mode in order to send scripts to the robot.

shuobh, 2022-03-10:
> As mentioned above, the robot has to be in `remote_control` mode in order to send scripts to the robot.

I've verified the TP is set to remote_control mode, here is a screen shot.
[image]
The driver is returning this failing log to send it through socket.
[image]
I am using ROS noetic and UR software version 5.11.1

fmauch, 2022-03-17:
This would need further investigation...

shuobh, 2022-03-18:
The issue was resolved. I think it might have something to do with TP setting, so factory reset and start from scratch fixed it.
However, to properly switch between freedrive mode and remote trajectory control, there are some tips to follow:
1. Stop the remote control program before enter freedrive mode via:
`rosservice call /ur_hardware_interface/dashboard/stop "{}"`
Note, if you do not stop, you can move the robot but then UR protective stops claiming deviate from path
2. Add a delay before publishing the enter freedrive script mentioned above, otherwise there might be a race condition.
3. To change back to remote trajectory control, run the stop program from 1., add some delay (I used 0.5s), then run play program, i.e., `rosservice call /ur_hardware_interface/dashboard/play "{}"`