UniversalRobots/Universal_Robots_ROS_Driver #154: Service raw_request always returns "could not understand: <request>" (when trying to set payload mass+CoG)

argargreg, 2020-04-03:
# Summary

ur_robot_driver does not advertise a service set_payload anymore (which existed on ur_driver and ur_modern_driver). In order to set the payload mass (which is mandatory in our case), raw URscript "set_payload(X)" and "set_payload_cog(Y)" have to be sent, and I assume this is done via the /ur_hardware_interface/dashboard/raw_request service. However, this service always returns "could not understand:", regardless of the request (tried: "set_payload(0)", 'print("hello")', "a=force()", "a = 1+1".

# Versions
 - ROS Driver version: Feb 7, 2020 
 - ROS version: melodic
 - Affected Robot Software Version(s): URsim 3.12.1.90940 UR10
 - Affected Robot Hardware: UR10
 - URCaps Software version(s): External Control 1.0.1

# Impact

The robot puts itself in safety when changing tools due to the change in payload mass and CoG. Without a way to send raw URscript commands, the development of our industrial station is stopped while this issue is not resolved (high priority for us).

## Use Case and Setup

- URSim UR10 launched
- ur_robot_driver/ur10_bringup.launch in one terminal
- rosservice call to raw_request OR rospy script in another terminal 

## Project status at point of discovered

Systematic issue easily reproduced.

## Steps to Reproduce

1. Start URSim UR10 with external control URcap on port 50002
2. In one terminal, run "roslaunch ur_robot_driver ur10_bringup.launch robot_ip:=127.0.0.1"
3. Hit "play" in URsim, the terminal says "robot ready to receive commands"
4. In another terminal, type "rosservice list"
Note that set_payload is absent.
5. Then type in the terminal opened in 4: rosservice call /ur_hardware_interface/dashboard/raw_request 'popup("hello")'
Note that the answer is "could not understand" and that the popup did not appear.
6. Try again with other URscript commands.
7. Try again with the following python script saved as "test_raw_request.py" in "ur_driver" to launch using rosrun ur_driver test_raw_request.py
```
#!/usr/bin/env python
from ur_dashboard_msgs.srv import RawRequest
import rospy

if __name__ == "__main__":
    rospy.init_node('RawRequest_service_server')
    s = rospy.ServiceProxy('/ur_hardware_interface/dashboard/raw_request', RawRequest)
    print s('popup("hello")')
```

## Expected Behavior

Actually executing the URscript command sent, for example showing a popup on the pendant.

## Actual Behavior

Service successfully called but failed during execution due to a so-called unknown command: "could not understand <command>"

## Workaround Suggestion

None. Help?

gavanderhoorn, 2020-04-03:
`raw_request` is afaik for Dashboard interactions / calls. It's not intended to forward URScript statements.

That would be `/ur_hardware_interface/script_command`.

argargreg, 2020-04-03:
Thanks for your answer. /ur_hardware_interface/script_command does not appear in rosservice list. Is that normal?

gavanderhoorn, 2020-04-03:
Yes. Because it's not a service, but a topic.

argargreg, 2020-04-03:
Ah, apologies ! Guess I'll close the issue then - thanks a lot for your help ! 

(P.S. If you know why set_payload() disappeared and if yes, how to get it back I'd very much appreciate as well!)

gavanderhoorn, 2020-04-03:
See #50.

argargreg, 2020-04-03:
Ah, I see - it still is not implemented. Well I'll run a script_command and call ur_hardware_interface/dashboard/play to resume external control (because I just tested that it pauses external control), that should do it. Case closed, thanks!

closed 2020-04-03 by argargreg

gavanderhoorn, 2020-04-03:
I haven't checked, but can't payload functions be invoked from secondary programs?

That's the way to run anything without motion without interrupting the external control program.

argargreg, 2020-04-03:
That would be it, but I don't know how to start secondary program from the ur_robot_driver interface... Do you?

Related to the initial issue, I've put together this code which works, for those who end up here looking for a same thing:
```
#!/usr/bin/env python
from std_srvs.srv import Trigger
from std_msgs.msg import String
from ur_dashboard_msgs.srv import IsProgramRunning
import rospy

PLAY_SERVICE_NAME = '/ur_hardware_interface/dashboard/play'
PROGRAMRUNNING_SERVICE_NAME = '/ur_hardware_interface/dashboard/program_running'
URSCRIPT_TOPIC_NAME = '/ur_hardware_interface/script_command'
TEST_COMMAND = 'set_payload(3., [0,0,.3])'
WAIT_DELAY = 0.1

class URscriptSender:
    def __init__(self):
        #resumeProgram becomes a function which calls the service when it is executed
        self.resumeProgram = rospy.ServiceProxy(PLAY_SERVICE_NAME, Trigger)

        #programRunning becomes a function which calls the service to know if a program is running on the robot (see below)
        self.programRunning = rospy.ServiceProxy(PROGRAMRUNNING_SERVICE_NAME, IsProgramRunning)

        #Creates a publisher to the script_command topic and 
        self.urscriptTopic = rospy.Publisher(URSCRIPT_TOPIC_NAME, String, queue_size=1)

    def sendCmd(self,cmd):
        #wait for a connection (publisher takes some time to establish a connection at the beginning)
        while self.urscriptTopic.get_num_connections() < 1:
            print("Waiting for publisher")
            rospy.sleep(WAIT_DELAY)

        #Then publish the command
        self.urscriptTopic.publish(TEST_COMMAND)

        #Publishing a script_command will halt the execution of the external control URcap, but we don't know when exactly so wait for the program to stop running...
        while(self.programRunning().program_running):
            print("Waiting for URscript command to be processed")
            rospy.sleep(WAIT_DELAY)

        #Then resume the program by hitting "play" on the pendant
        self.resumeProgram()

        while(not self.programRunning().program_running):
            print("Waiting for external control program to restart")
            rospy.sleep(WAIT_DELAY)

if __name__ == "__main__":
    rospy.init_node('test_set_payload') #Do not copy/paste if node already initialised wherever the code is put

    urscriptSender = URscriptSender()
    urscriptSender.sendCmd(TEST_COMMAND)
```

gavanderhoorn, 2020-04-03:
> That would be it, but I don't know how to start secondary program from the ur_robot_driver interface... Do you?

Well, it's pretty well documented on the UR support page (and in the scripting manuals). See [Secondary program](https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/secondary-program/) fi.