UniversalRobots/Universal_Robots_ROS_Driver #159: Model of internal controller

macsmitty, 2020-04-08:
Hi!
I'm trying to approximate the internal controller model of the Ur10 to use it in Simulation (approximate torques from position deviations). I need to approximate a similar model to achieve similar behaviour especially in contact richt scenarios. Are there any model approximations available that you know of?
I recorded some trajectories and could not yet make total sense of them, but I noticed something:

the target_q does not correspond to the control sequence I send to the controller. There seems to be some kind of smoothening happening. I assume that's the smoothening that is mentioned in the urscript documentation about servoj(). Does anyone know the algorithm that is used in this smoothening function? (see plot below)
image
[image]

Furthermore, any help regarding a methodology to identify the behaviour of the internal controller is highly appreciated.

fmauch, 2020-04-21:
I don't feel I can answer your question completely. My first question would be: Which interface did you use for commanding and for creating the plots?

fmauch, 2020-05-12:
Any updates on this?

closed 2023-04-13 by github-actions[bot]