UniversalRobots/Universal_Robots_ROS_Driver #162: How to control robot via rostopics?

mat-l, 2020-04-22:
Dear all,
I have successfully installed ROS and the Universal_Robots_ROS_Driver and I am able to control the robot via the graphical interface of "rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller".

Additionally I can set/unset the digital outputs via: rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'set_digital_out(2,True)'"

But I cannot send to robot to a defined position via the script command: rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'movel([0.5,0.5,0.5,3.14,-1.57,1.57]'".
How do I have to adjust the command above to move the robot in movej or movel mode to a defined point?

Additionally, how does a message look like to send the robot again in movel or movej to a point at "/scaled_pos_traj_controller/follow_joint_trajectory"

Best regards and thanks in advance for your help

gavanderhoorn, 2020-04-22:
> Additionally, how does a message look like to send the robot again in movel or movej to a point at "/scaled_pos_traj_controller/follow_joint_trajectory"

Please see this comment: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/122#issuecomment-591861319.

It would seem https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/122 discusses a similar question (or the `follow_joint_trajectory` part anyway).

fmauch, 2020-04-22:
Commanding robots using `movel` or `movej` is in my experience nothing usually done inside of ROS. This is why we have standardized interfaces such as `follow_joint_trajectory`.

If you want to do that nevertheless, you already figured out the right way to do it, only your script syntax is wrong. (This is actually one of the reasons this is not being used usually). You missed the closing bracket at the end.
```
rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'movel([0.5,0.5,0.5,3.14,-1.57,1.57])'"
```

You just won't receive any feedback or whatsoever from the robot. We do not recommend using this driver this way. See the comment @gavanderhoorn linked above, instead.

mat-l, 2020-04-22:
@fmauch thanks for your fast answer. Yes it was just a problem of my syntax - small mistake, but I didn't get anywhere a error message, therefore I didn't see the mistake.

For controlling a robot in industrial applications linear movement with defined speed and acceleration parameters are very important. If I control the robot via Universal Robots ROS Driver, will I be able to perform movel and movej commands as well?

And still I didn't get yet how to control the robot with your driver as mentioned in #122. Is there somewhere example code how this would like? Maybe some small client in Python where I can try to understand it?

closed 2020-04-22 by mat-l

gavanderhoorn, 2020-04-22:
> I didn't get anywhere a error message, therefore I didn't see the mistake.

That's actually a limitation of the controller, not of the driver. Scripts uploaded to `tcp://:30002` et al. don't get to return any output, so there's nothing for the driver to report back to you.

> For controlling a robot in industrial applications linear movement with defined speed and acceleration parameters are very important.

Somewhat pedantic, but: while I do agree with you, please don't generalise like this, or nuance your statements a bit. For applications you have in mind, this is important. I know of quite a few applications where linear motions are not necessarily required.

If you're mostly interested in using `movel()` and `movej()`, one could ask whether using ROS/`ur_robot_driver` is the appropriate interface for you.

A mode where the controller can 'request' the ROS side to become active (which is planned as a future extension to the driver), with the program on the TP in full control, would perhaps be better.

> And still I didn't get yet how to control the robot with your driver as mentioned in #122. Is there somewhere example code how this would like? Maybe some small client in Python where I can try to understand it?

You could see whether [this](https://github.com/ipa-hsd/universal_robot/blob/e09a05d69b3f1790a52b282d824347a86fcea0a8/ur_driver/test_move.py) helps.

You'll have to update [this line](https://github.com/ipa-hsd/universal_robot/blob/e09a05d69b3f1790a52b282d824347a86fcea0a8/ur_driver/test_move.py#L43) to make it use the correct Action server topic, and beware that this actually moves your robot if/when you run it.

But it should get the idea across.

Note that the Action interfaces are typically not used in this way, but to execute entire trajectories (ie: much more complex motions).

mat-l, 2020-04-23:
@gavanderhoorn thanks a lot for your comment, that worked great. I am able to control the robot with the Python script. :-)

One last question regarding the set up of the trajectories. How can I hand over a trajectory that is planned in Cartesian space instead of joint space/angles? Is there a package existing that calculates a linear movement with respect to velocity and acceleration that can be used?

fmauch, 2020-04-23:
Cartesian interfaces are until now not very common inside of ROS. Actually, we are currently working on that, we've just released a first draft design for a Cartesian interface: https://fzi-robot-interface-proposal.readthedocs.io/

If you feel like commenting or contributing to this, please do so in the corresponding github repository: https://github.com/fzi-forschungszentrum-informatik/fzi_robot_interface_proposal

anton-brinster, 2020-06-17:
@fmauch My goal is also to control the UR3 using `follow_joint_trajectory` to move the robot to given positions [#122 - for trajectories]. I was following the instruction but I am not familar with actionlib (actionclient). So I stuck at the point of how to set the action server exactly.

Do I just need to modify and run this .py script?
[https://github.com/ipa-hsd/universal_robot/blob/e09a05d69b3f1790a52b282d824347a86fcea0a8/ur_driver/test_move.py](url)

This one could not help me: [http://wiki.ros.org/pr2_controllers/Tutorials/Moving%20the%20arm%20using%20the%20Joint%20Trajectory%20Action](url)

@mat-l I would also be grateful for your advice because you finally get a solution for the issue.