UniversalRobots/Universal_Robots_ROS_Driver #163: Forward script execution failures back to the user

fmauch, 2020-04-23:
The robot's primary interface actually reports errors in executed script code via a `ROBOT_MESSAGE` of type `ROBOT_MESSAGE_TYPE_RUNTIME_EXCEPTION`.

We can catch those and at least print a warning. This way false script code sent to the script interface will not just silently dissolve in the nirvana.

However, from my understanding we cannot know at this point whether this error message actually is triggered by script sent through our driver, which is why I would only print a warning.

gavanderhoorn, 2020-04-23:
Is the primary interface intended to be used for submitting user scripts?

It was my understanding that it's mostly used by Polyscope.

fmauch, 2020-04-23:
Well, it is what we use for script commands anyway. (Technically, we use the secondary interface)

The guys at UR suggested to take a look into that solution, so I wanted to write it down here the least.

gavanderhoorn, 2020-04-23:
O wait. I'm confusing two things. `/ignore`.