UniversalRobots/Universal_Robots_ROS_Driver #165: Can't accept new commands. Controller is not running.

TFLQW, 2020-04-29:
# Summary

Connect the ROS with URSIM using ros_control.
First, I have install the URCaps in the ursim. 
Then, I have execute the command 'roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=127.0.0.1' then I start the program in the ursim by plugin the external control, in the  terminal I have got the message:
` INFO] [1588149911.373263600]: Robot requested program
[ INFO] [1588149911.373427040]: Sent program to robot
[ INFO] [1588149911.476540546]: Robot ready to receive control commands.
`
#Issue
First, in a new terminal I execute 'rostopic echo /joint_states' I could get the joint position in URsim.This part indicated that I have got connect with the robot in URsim.
But the problem is that : When I try to start a now node to publish a message to the 'pos_traj_controller/command' , the error in the terminal show that: 
**[ERROR] [1588150073.314638437]: Can't accept new commands. Controller is not running.**
The rqt_graph is like that:
[image]

The whole terminal message like that
`PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_traj_controller/action_monitor_rate: 10
 * /pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/goal_time: 0.6
 * /pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_traj_controller/state_publish_rate: 125
 * /pos_traj_controller/stop_trajectory_duration: 0.5
 * /pos_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /scaled_pos_traj_controller/action_monitor_rate: 10
 * /scaled_pos_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_traj_controller/state_publish_rate: 125
 * /scaled_pos_traj_controller/stop_trajectory_duration: 0.5
[... 198 lines cut ...]
[INFO] [1588150007.085224]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1588150007.086801]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1588150007.087655]: Loading controller: joint_state_controller
[INFO] [1588150007.088889]: Loading controller: pos_traj_controller
[INFO] [1588150007.099369]: Loading controller: scaled_pos_traj_controller
[ INFO] [1588150007.153501060]: Robot mode is now RUNNING
[ INFO] [1588150007.153599012]: Robot's safety mode is now NORMAL
[INFO] [1588150007.157839]: Loading controller: joint_group_vel_controller
[INFO] [1588150007.215984]: Loading controller: speed_scaling_state_controller
[INFO] [1588150007.233155]: Controller Spawner: Loaded controllers: pos_traj_controller, joint_group_vel_controller
[INFO] [1588150007.241366]: Loading controller: force_torque_sensor_controller
[INFO] [1588150007.258197]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1588150007.266858]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[ INFO] [1588150051.483819132]: Robot requested program
[ INFO] [1588150051.483916623]: Sent program to robot
[ INFO] [1588150051.516312704]: Robot ready to receive control commands.
[ERROR] [1588150073.314638437]: Can't accept new commands. Controller is not running.`

# Versions
 - ROS Driver version: kinetic
 - Affected Robot Software Version(s): ursim-3.12.1.90940
 - Affected Robot Hardware Version(s):
 - Robot Serial Number: No
 - UR+ product(s) installed: No
 - URCaps Software version(s): External Control

TFLQW, 2020-04-29:
I have found the error, because the pos_traj_controllers is not started, so I have change to scaled_pos_traj_controllers ,it's work.

TFLQW, 2020-05-01:
Actually, the robot in ursim can move by this solution, but in terminal, the error is still alive,'[ERROR] [1588150073.314638437]: Can't accept new commands. Controller is not running' , the robot in ursim is unstable when tracking a circle trajectory in end-effector space. 
  I guess some error maybe occur in the topic publish rate, but I fail to get the right hz to solve this problem, could anyone give me some advice?
The following is the publish rate of '/scaled_pos_traj_controller/comand':
rostopic hz /scaled_pos_traj_controller/command 
subscribed to [/scaled_pos_traj_controller/command]
average rate: 39.701
	min: 0.009s max: 0.046s std dev: 0.01133s window: 36
average rate: 41.183
	min: 0.009s max: 0.059s std dev: 0.01169s window: 79
average rate: 40.075
	min: 0.006s max: 0.112s std dev: 0.01596s window: 117
average rate: 42.361
	min: 0.004s max: 0.112s std dev: 0.01483s window: 166
average rate: 45.836
	min: 0.002s max: 0.112s std dev: 0.01378s window: 225
average rate: 45.249
	min: 0.002s max: 0.112s std dev: 0.01399s window: 268
average rate: 44.509
	min: 0.002s max: 0.112s std dev: 0.01451s window: 307
average rate: 45.359
	min: 0.002s max: 0.112s std dev: 0.01435s window: 359
average rate: 45.708
	min: 0.002s max: 0.115s std dev: 0.01536s window: 408
average rate: 47.252
	min: 0.002s max: 0.115s std dev: 0.01505s window: 469
average rate: 46.887
	min: 0.002s max: 0.115s std dev: 0.01516s window: 511
average rate: 47.862
	min: 0.002s max: 0.115s std dev: 0.01501s window: 571
average rate: 46.900
	min: 0.002s max: 0.115s std dev: 0.01512s window: 605
^Caverage rate: 47.450
	min: 0.002s max: 0.115s std dev: 0.01507s window: 632

gavanderhoorn, 2020-05-01:
It would help if you could provide some more information.

The "controller not running" message is expected if you don't start the External Control program on the TP. Have you started it? The [quickstart](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#quick-start) documentation describes how to do this, and the [Installing a URCap on a CB3 robot](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md) gives more detailed steps.

TFLQW, 2020-05-01:
@gavanderhoorn 
Hi~, Thanks for your reply. I have confirmed that  I have started the External Control program on TP, and I have also recieve that 
'[ INFO] [1588150051.483819132]: Robot requested program
[ INFO] [1588150051.483916623]: Sent program to robot
[ INFO] [1588150051.516312704]: Robot ready to receive control commands.'
Maybe I will try to use 'JointGroupPositionController', but I am still suspect the topic publish rate not matching.

TFLQW, 2020-05-01:
@gavanderhoorn 
Hi~, I have just verified the solution as you suggest to change from `JointTrajectoryController` to `JointGroupPositionController`, the error disapears, but the robot in ursim is more unstable and it's hard to track a circle trajectory. Could help me to find the problem?
The robot is like that:
[image]

TFLQW, 2020-05-02:
@gavanderhoorn
Hi~  I have a good news that I have solved this problem by using  velocity_controllers/JointGroupVelocityController, maybe velocity control is suitable for this kind of tracking problem.

closed 2020-05-02 by TFLQW

fmauch, 2020-05-04:
I agree that velocity control is more suitable for this kind of situation.

To add to an earlier post of @gavanderhoorn: The message "Can't accept new commands. Controller is not running" can have two different sources
 
 1. The actual ROS controller is not in a running state, you are trying to use a loaded, but unstarted controller.
 2. The external control mode is not active on the robot as @gavanderhoorn suggested

Technically both boil down to the first version, as controllers will get temporarily stopped when the external control is not active on the robot.