UniversalRobots/Universal_Robots_ROS_Driver #168: Control Robotiq 2F-85 gripper attached to UR5e

tonkei0361, 2020-05-01:
# Summary

I'm not sure how to control 2F-85 gripper attached to UR5e. The gripper is directly connected to the robot's tip, not via USB to PC. Is it through tool_communication? I've set up tool_communication by following the documentation and the driver runs successfully. However, I'm not sure what to do next so that I can command the gripper to do things like open and close.

Thank you!

# Versions
 - Affected Robot Software Version(s): 5.3.1
 - Affected Robot Hardware Version(s): UR5e
 - URCaps Software version(s):1.0.1

fmauch, 2020-05-04:
This highly depends on how you would usually connect to the gripper.

Imagine to get the robot our of the way and you connect the gripper directly to your ROS pc. If you use an RS485 interface connection to communicate with the gripper, you will usually do something like this:

```
roslaunch gripper_driver gripper_node device:=/dev/ttyUSB0
```

where `/dev/ttyUSB0` is the usb device of your serial converter.

You can do the same using the robot's interface by activating `tool_communication` and configuring it to the right parameters (namely `tool_voltage` and baud rate).

Then, when starting the driver, you will get another tty device (defaults to `/tmp/ttyUR` that you can hand to the driver:

```
roslaunch gripper_driver gripper_node device:=/tmp/ttyUR
```

In case of the robotiq 2f gripper, I think (unverified / untested) this boils down to
```
rosrun robotiq_2f_gripper_control nodes/Robotiq2FGripperRtuNode.py /tmp/ttyUR
```

I hope, this helps. If you need to take any additional steps, please let me know here, maybe we can create a tutorial together that can be used for other people, as well.

tonkei0361, 2020-05-04:
@fmauch Thank you for your response.

I'm using the gripper that is directly attached to the robot. However, there's no launch file called `gripper_driver` or package `robotiq_2f_gripper_control`. Did I miss anything during the installation process? I just followed the instructions in Universal_Robots_ROS_Driver repo step-by-step and didn't do anything else.

gavanderhoorn, 2020-05-04:
The names @fmauch used in his comment were examples. They're not meant to be copy-pasteable instructions.

fmauch, 2020-05-04:
That was just meant as an example...

As you are asking how to get a gripper from another manufacturer running with this driver, please do not expect that people here do know how to start the specific gripper.

The last code sample I wrote could work, but I don't have any chance to test it.

Please note that this driver only supports forwarding the serial interface to the Ros machine. Everything from there is the user's responsibility.

tonkei0361, 2020-05-04:
@fmauch @gavanderhoorn  I see, thanks for clarifications.

In this case, do I need to make my own gripper driver that works with current UR ROS driver? I'm more than happy to contribute to the driver.

fmauch, 2020-05-04:
> In this case, do I need to make my own gripper driver that works with current UR ROS driver? I'm more than happy to contribute to the driver.

Well, I would start with the existing one over at https://github.com/ros-industrial/robotiq

fmauch, 2020-05-12:
Anything to be added to this from a driver's perspective? Otherwise I would go ahead and close this.

tonkei0361, 2020-05-12:
@fmauch Nothing to add for now. I will close it, thank you!

closed 2020-05-12 by tonkei0361

felixvd, 2020-08-04:
Just for the record, I added a driver for Robotiq grippers attached to UR robots to [this package](https://github.com/crigroup/robotiq), which we used for CB-series. It works for the e-series, but restarting the `ROS_external_control` script is somewhat awkward.

rrapi, 2022-05-06:
Hi everyone!

I am doing my master thesis with this robot and gripper, and I am trying to connect and control the same gripper through my PC but still did not succeed. I tried until now even with gripper directly attached to robot and also attached to the ROS PC through USB but still not able to control it.
I uninstalled also the Robotiq_Gripper URCap but the light is always red because of the connection never established with my PC.

I tried the steps that @fmauch told in last messages, with tool communication and gripper driver but still nothing and this is the error that I get:
> pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[rtu baud[115200]]

Could you please help me to make it work? The gripper is very essential in my thesis project and I would be very grateful if I achieve to control it.

Best regards,
Franci