UniversalRobots/Universal_Robots_ROS_Driver #169: operate ur5 with dynamically changing gravity vector

fmessmer, 2020-05-04:
We are trying to operate an UR5 as part of an extended manipulator, i.e. it's mounted as part of a kinematic chain with additional movable joints prior to the UR5.
Those joints change the orientation of the UR5's base_link and thus have an impact on the gravity vector of the UR's internall controller.
When moving the joints of the kinematic chain e.g. tilting the UR5's base link, the safety stop of the UR kicks and stops the movement.

Questions:
- Is there a way to update the gravity vector of the UR controller from ROS? Preferably streaming via ROS topic...
- Is there a way to disable the safety controller from getting activated?

fmauch, 2020-05-04:
You can update the gravity vector using URScript, which you can send to `/ur_hardware_interface/script_command`

```
rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'set_gravity([9.81, 0, 0])'"
```

Probably you can also wrap this call into a [secondary program](https://www.universal-robots.com/articles/ur-articles/secondary-program/), so it won't interrupt other scripts (e.g. running External Control).

For your second question I cannot help you, I'm afraid.

gavanderhoorn, 2020-05-04:
https://static1.squarespace.com/static/51df34b1e4b08840dcfd2841/t/5e5e5b4b5406da1b3944503e/1583242082440/04_Anders_Beck_Universal+Robots+-+Towards+Industrial-grade+robot+control+in+ROS_12-12-2019.pdf

fmauch, 2020-05-04:
Sorry, I think I wasn't reading this thoroughly enough. As it is something that might continuously change I agree that the URScript topic approach isn't ideal as each time you send script to that interface, a new script gets compiled on the controller.

But I'm also not sure whether dynamic path offset helps here.

One approach could be to sneak it to the communication between ROS and the controller and then add a thread to the program's URScript that will read those values and call `set_gravity` there. This way, it won't have to get recompiled each time...

gavanderhoorn, 2020-05-04:
> But I'm also not sure whether dynamic path offset helps here.

I haven't used it myself, so this was just a guess.

If it really only compensates for motions of the base, then it would most likely not be useful, no.

If it is a bit more generic and can be use for updating the gravity vector while moving, it may be useful.

benmaidel, 2020-06-03:
> One approach could be to sneak it to the communication between ROS and the controller and then add a thread to the program's URScript that will read those values and call `set_gravity` there. This way, it won't have to get recompiled each time...

For a proof of concept this sounds promising to me. Unfortunately I'm totally unexperienced in UR development (urcap). I could not even find the source code for the external_control urcap. Isn't the code on github? It would be great if you could give us some hints where to start if we like to implement this.

fmessmer, 2020-06-09:
I'm going to pick up on this:

We are currently experimenting with an extension of the driver that subscribes to a `geometry_msgs/Vector3` message and feeding the content into `internals.set_gravity(d)` (see https://s3-eu-west-1.amazonaws.com/ur-support-site/32554/scriptManual-3.5.4.pdf)

This approach seems to work while the robot is mounted upright (e.g on a table) - i.e. we publish a `geometry_msgs/Vector3` of `(0, 0, 9.81)` and the robot can be moved around with ROS without protective stops
Thus, continuously calling/updating `internals.set_gravity` does not stop movements or abort control.

However, in our case, the robot is mounted in a wall-like start configuration - e.g. 90° rotation about x-axis of (internal) UR base link.
When starting the robot in that mount configuration, the TeachPanel asks us to adjust the "Mounting" of  the UR - where you can select either predefined configs (table-upright, wall-sideways, roof-hanging) or adjust it in 45° angle steps manually.

We also have a node that calculates the gravity vector (0, 0, g) with respect of coordinate systems - i.e. use tf to retreive respective rotation matrix and multiply (0, 0, g) with it - the result is published on the `geometry_msgs/Vector3` topic.

However, we do not know with respect to which (internal) coordinate system the gravity vector should be expressed and whether the configuration of the mounting on the TeachPanel changes those reference coordinate systems.

So, atm we are not yet talking about dynamic system, but rather about setting gravity vector for a non-upright mounting.

Does anybody have information on gravity vector reference coordinate systems and/or knows more about the effect of the mounting settings (on the TeachPanel) on those reference coordinate systems? e.g. technical drawings?

gavanderhoorn, 2020-06-09:
I would really recommend you contact UR. They would be the ones to give you an authoritative answer.

fmessmer, 2020-06-09:
We'll try to...
Still: Are people from UR not following this repo? It's the official driver now...

gavanderhoorn, 2020-06-09:
Well, according to [Support > Articles > UR articles > Universal Robots ROS Driver](https://www.universal-robots.com/articles/ur-articles/universal-robots-ros-driver/):

> NOTICE
> 
> Universal Robots Technical Support **does not** provide support in the usage of the ROS driver.
If you encounter issues with the ROS driver, such as bugs or similar, please report these under GitHub's issue reporting templates. 
Should you require support using the driver, please consult the ROS-category in the [UR+ Developer Community](https://www.universal-robots.com/plus/).

fmauch, 2020-06-09:
I also think, that going to the UR plus developer forum might be a good place to go in detail about that.

gavanderhoorn, 2020-06-09:
It's really annoying registration is needed though to be able to *read* the posts there.

@fmessmer: if you do post there, could you please post a comment here with a link?

cjsavoia, 2020-06-09:
@gavanderhoorn I work in the UR+ team. We are working on opening up the forums for general public read access, although it will still require a log in to post. More to come in the next couple of months. BTW, I'm still learning ROS, but have been following since `ur_modern_driver` and love the discussion that goes on here. Sorry to highjack the thread...

benmaidel, 2020-06-10:
> @fmessmer: if you do post there, could you please post a comment here with a link?

I've started a new thread on the UR plus forum.
https://forum.universal-robots.com/t/dynamically-changing-gravity-vector/7633

gavanderhoorn, 2020-06-10:
> We are working on opening up the forums for general public read access, although it will still require a log in to post. 

That's good to hear.

Is there any ETA for this? "a couple of months" is a rather long time.

fmessmer, 2020-06-13:
I’m reporting back the results form our latest tests:

as stated in https://forum.universal-robots.com/t/dynamically-changing-gravity-vector/7633/9, we were able to verify the coordinate system to be used (:point_right: `base`) and how to send the dynamic gravity vector
in the end, we were able to dynamically update the gravity vector based on the orientation of the reference coordinate system without interrupting ongoing motion, without the protective stop kicking in and without any other gravity related fault situation

while it works for our proof-of-concept use-case we found various reasons for this to be possibly dangerous and unsafe as it is highly sensitive to the initial config of the mounting on the panel and what is being sent as the current gravity vector

in the current state, we don’t consider the driver extension ready for the open public
also, the part that calculates the current dynamic gravity vector is specific to our use-case and thus closed-source

gavanderhoorn, 2020-06-13:
>  the part that calculates the current dynamic gravity vector is specific to our use-case and thus closed-source

I guess you have the same gravity vectors as everyone else, no?

benmaidel, 2020-06-16:
The changes for sneaking in the gravity vector to the ROS - UR communication can be found here: https://github.com/benmaidel/Universal_Robots_ROS_Driver/tree/feature/gravity_vector

This is just a simple proof of concept implementation. If this feature is interesting and should be added I think a few things should be considered:
- send the gravity vector to the urscript only if it is wanted and only if received from the subscriber?
- add e.g. a header to the communication protocol to be able to define the amount of bytes / features that are send to the urscript?
- add rosparam to define default gravity vector?
- what should happen if updating the gravity vector interrupts (e.g. gravity publisher dies)?
- setting completely wrong gravity could lead to immediate fast motions or just the arm falling down (happened to me). Should this be handled or is it the risk of the integrator?
- ...

Hope this could be helpful for others

cjsavoia, 2020-06-23:
@gavanderhoorn @fmauch, the UR+ Forums can now be viewed without logging in, you can test in Incognito Mode https://forum.universal-robots.com/. You will still need to log in to post and to download the URCap SDK.

gavanderhoorn, 2020-06-24:
Thanks for the update @cjsavoia :+1:

closed 2023-05-13 by github-actions[bot]