UniversalRobots/Universal_Robots_ROS_Driver #170: diagnostics concept

fmessmer, 2020-05-04:
I can't find any code related to the [ROS diagnostic concept](http://wiki.ros.org/diagnostics)
Is there any ongoing effort/plan to support this feature?

I'd like to - at least - track the operation status of the UR and the driver, like:
 - robot operable/initialized
 - safety stop active

fmauch, 2020-05-12:
Since #141 we have an [industrial_robot_status_controller](http://wiki.ros.org/industrial_robot_status_controller) included.

To recover after a failure I suggest using the `set_mode` action As [described in the action](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_dashboard_msgs/action/SetMode.action#L10) the robot will continue moving. If you don't use a controller being aware of the robot not following the trajectory anymore (which is why we suggest using the `scaled_trajectory_controller`) this will lead to a fast motion. In that case the safe reset is to request a program stop (in the same action) as well. As stopping the program also stops running controllers it will be reset afterwards.

Note: The reason why this is implemented that way is because the robot is actually in a "paused" state, so resuming after an EM-Stop or protective stop can be done directly. I'm happy to put this out for discussion, though.

fmauch, 2020-05-20:
Would the `industrial_robot_status_controller` together with the changes from #170 implement what you had in mind? Otherwise we might merge #170 and this together to create something that covers both issues.

gavanderhoorn, 2020-05-20:
`industrial_msgs/RobotStatus` won't tell you what kind of stop is currently active though.

`in_error` will be `TRUE` whenever the robot is in an error mode, but as it's not an e-stop, the message does not have a way to encode what put it there (or at least: the `error_code` field is not set to anything right now).

`error_code` could be used for this, but I would only do it if UR has defined numerical errors for the various kinds of stops (I believe there are) *and* those are communicated to the driver via RTDE or the primary/secondary interface (with #171).