UniversalRobots/Universal_Robots_ROS_Driver #172: How to send basic position control commands using ROS

Sumilsood, 2020-05-05:
Hi,

I am using Robot driver to control a UR10e robot. I have followed the steps as mentioned and everything is working well. However in the Quick start section it says to use use the action server on
' /scaled_pos_traj_controller/follow_joint_trajectory '. I am having trouble understating how to use this action server. If you can send a simple script to do basic position control of the robot then that will be really helpful.

gavanderhoorn, 2020-05-05:
Please see whether https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/122#issuecomment-591861319 and #162 already help.

Sumilsood, 2020-05-08:
Hi Thank You for fast reply
So i went through this and I am able to control the robot through Python code. I had few more doubts. So if I am doing a closed loop control what is the frequency at which I can send joint angles to the robot. This means what's the sampling rate for this?
Secondly can you tell me how to get joint angles to my computer from the robot. I mean which topic are these joint angles published.

Thank You

tonkei0361, 2020-05-08:
@Sumilsood For getting joint angles, you can subscribe to `/joint_states` topic. I don't have an answer to your first question.

fmauch, 2020-05-11:
This driver doesn't (directly) offer an interface to do closed-loop control. The ' /scaled_pos_traj_controller/follow_joint_trajectory ' is a high-level trajectory controller where you send complete trajectories for execution.

Inside the driver we use [ROS-Control](http://wiki.ros.org/ros_control) to implement different controllers. See its documentation for details.

If you really want to do low-level closed loop control, you could (in theory) instantiate a `position_controllers/JointGroupPositionController` controller and send position commands directly to this. It's maximum frequency will be 500Hz on an e-Series robot. However, we do absolutely not recommend doing this, as closed loop control on a topic level is a bad idea due to additional levels of communication that potentially can (and will!) introduce delays in your command processing. Your robot's motions will most probably be jerky and potentially lead to a protective stop.

If you are not satisfied with the controllers available inside ROS control you can always write your own controllers and load them dynamically. However, this goes out of scope of this driver, so please don't open an issue on that. Rather ask on [answers.ros.org](answers.ros.org) if you have problems implementing this.

I hope, this clarifies things?

closed 2021-08-09 by stefanscherzinger