UniversalRobots/Universal_Robots_ROS_Driver #173: Change the order of published joint states

MinhDangTG, 2020-05-06:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

# Versions
 - ROS Driver version: 16.04 Kinetic
 - Affected Robot Software Version(s): 3.12
 - Affected Robot Hardware Version(s): CB3 UR3 
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):

Hello everyone, 

I am currently using joint_group_vel_controller to apply velocity controller. Is there any way that I can change the order of published joint_states in /joint_states topic? Since the order of joints follows alphabet, it creates some problems during inputting and configuring data. 

[image]

It should be liked this: 
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint

Another question is that can I publish acceleration of each joint when using velocity control? I used ur_modern_driver (which has accelerations in trajectory_msgs/JointTrajectory) before switching to ur_robot_driver package and using the joint_group_vel_controller (std_msgs/Float64MultiArray).

Thank you so much.

fmauch, 2020-05-06:
To this issue's topic:

The joint_state ordering is being done by the [joint_state_controller](https://github.com/ros-controls/ros_controllers/blob/melodic-devel/joint_state_controller/src/joint_state_controller.cpp) which in turn gets the joint names from the Hardware Interface, where the handles get registered [here](https://github.com/ros-controls/ros_control/blob/melodic-devel/hardware_interface/include/hardware_interface/internal/resource_manager.h#L100)

As handles are stored inside an `std::map`, the sorting happens there. If you would like to propose to change this, you'll have to do it there.

However, that being said, it might make sense to take care about that on your client's code side, as it is always a good idea not to rely on the ordering of some external input.

For your second question: This driver also offers a `control_msgs/FollowJointTrajectory` action where you can specify your setpoints' accelerations. Just use the `scaled_traj_controller` instead of the `joint_group_vel_controller`. But I'm not sure what you want to achieve, as a trajectory controller is quite different from a direct velocity controller...

gavanderhoorn, 2020-05-06:
> As handles are stored inside an `std::map`, the sorting happens there. If you would like to propose to change this, you'll have to do it there.

Please don't do this @MinhDangTG.

This is actually very sound advise and a best-practice:

> However, that being said, it might make sense to take care about that on your client's code side, as it is always a good idea not to rely on the ordering of some external input.

fmauch, 2020-05-12:
I'll go ahead and close this, as I think the initial question has been answered.

closed 2020-05-12 by fmauch