UniversalRobots/Universal_Robots_ROS_Driver #174: tf2 frames published by this driver

TheDelus, 2020-05-06:
Is there some kind of documentation on which TF frames are published by this driver? As suggested in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/127#issuecomment-605032696 I want to retrieve the TCP by making a TF lookup. I guess it hast to be one of the following transformations:

* world -> tool0_controller
* world -> tool0
* world -> ee_link 

One thing I´m wondering as well is that none of this 3 transformations correspond to the TCP shown on the teachpendent. 
I´m using an UR5e with Polyscope 5.6.0.
Every help is much appreciated.

gavanderhoorn, 2020-05-06:
None of the transforms you list as candidates will correspond to what the controller reports.

You'll have to ask for `base -> tool0_controller`.

TheDelus, 2020-05-06:
Thanks for the lightning fast and helpful response. The lookup `base -> tool0_controller` retrieves the wanted TCP that corresponds to the TCP in polyscope.

Further I can confirm that an exported robots calibration results in `base -> tool0_controller` and `base -> tool0` beeing (nearly) identical.

closed 2020-05-06 by TheDelus