UniversalRobots/Universal_Robots_ROS_Driver #175: Set the robot to force mode

tonkei0361, 2020-05-08:
# Summary

Hello, I wonder if there is a way to set the robot to force mode with this driver. The only way I know to set to force mode is by using TP but I need to do this using ROS. Thanks.

The reason I'm asking is that with force mode, I may be able to implement impedance control. I want to set the force parameter along specified axis as a variable that depends on another variable (e.g. robot's state). The robot is free to move along that axis but exerted force can vary.

# Versions
 - Affected Robot Hardware Version(s): UR5e

fmauch, 2020-05-11:
No, you can't.

You could, however, pack the "External Control" node on the TP inside a force mode container, but I have no idea how well that will work. If you want to do compliant control at the tool only (force mode won't give you impedance control for each joint either) I can recommend [FZI's cartesian_controllers](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers) (specifically the [compliance_controller](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/cartesian_compliance_controller)) that work with any robot with a force-torque sensor at the TCP.

tonkei0361, 2020-05-12:
@fmauch Thank you for your response!

Your suggestion is totally ok for my application. However, I'm not very familiar with using ROS control. This may be off topic but can you briefly talk about what I should do in order to get this controller running and able to control the robot? What I know so far is that I need to download the repo to my workspace, run catkin_make, and then load the controller. I've tried loading other controllers inside ROS Control in the past but `ok` always returns false. I'm not sure where to find the name of the controller and how to make a controller ready to be loaded.

fmauch, 2020-05-12:
See my colleague's repo I mentioned above and the [controller_manager](http://wiki.ros.org/controller_manager) wiki page for details. I'll close this issue if that's fine with you.

closed 2020-05-12 by fmauch

Csp0920, 2020-11-18:
> @fmauch谢谢您的回复！
> 
> 您的建议完全适合我的申请。但是，我对使用ROS控制不是很熟悉。这可能不在主题之列，但是您能否简要谈一谈我应该怎么做才能使此控制器运行并能够控制机器人？到目前为止，我所知道的是我需要将存储库下载到我的工作区，运行catkin_make，然后加载控制器。我过去曾尝试在ROS Control中加载其他控制器，但`ok`始终返回false。我不确定在哪里可以找到控制器的名称以及如何使控制器准备好加载。

hello.
I also want to use force mode. So if you can turn the robot into force mode with "pack the "External Control" node on the TP inside a force mode container"? Because I sent the robot back to test that I can't do this.