UniversalRobots/Universal_Robots_ROS_Driver #176: Uninitialized variable in robot_state_helper leads to invalid_argument

RobertWilbrandt, 2020-05-11:
# Summary

Using the ```set_mode``` action of ```robot_state_helper``` directly after driver startup (e.g. in the same launch file can lead to a ```robot_state_helper``` crash with the following error message:

```console
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Unknown robot mode: -128
```

This is most likely caused by an uninitialized variable.

I will probably create a PR to fix this, but only when i am able to test with a robot again.

# Versions

(I cannot really test this on master, as for some reason nobody gave me a robot arm to take home... but from looking at the code the error seems to still exist)

 - ROS Driver version: version 0.0.3, commit a926789adc0cefc616beb0fda15793cdcd65831b
 - Affected Robot Software Version(s):
 - Affected Robot Hardware Version(s): CB3 UR5
 - Robot Serial Number:
 - UR+ product(s) installed: -
 - URCaps Software version(s): -

# Impact

Not tragic, as this only happens when using a robot startup script (i.e. a script that handles robot startup from the launch file) and a workaround is straightforward.

# Issue details

This is probably caused by the uninitialized variables ```robot_mode_``` and ```safety_mode_``` in [robot_state_helper.cpp](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/ros/robot_state_helper.cpp). When the ```set_mode``` action is called with a target state of ```RUNNING``` before a ```/robot_mode``` or ```/safety_mode``` message is received, ```doTransition()``` is called, which goes to default for bot ```robot_mode_``` and ```safety_mode_``` and tries to call ```robotModeString()```, which throws the exception.

## Project status at point of discovered

Unsurprisingly, the issue appeared nondeterministically, but with about 50% likelihood.

## Steps to Reproduce

Start both the ur driver and a startup script in the same launch file. Make the startup script call the ```set_mode``` action immediately.

## Workaround Suggestion

Wait for messages on the ```/robot_mode``` and ```/safety_mode``` topics before calling the action.

gavanderhoorn, 2020-05-11:
> I cannot really test this on master, as for some reason nobody gave me a robot arm to take home.

`ursim`?

fmauch, 2020-05-11:
Thanks for reporting this. That should obviously not be the case. It Should probably be fairly easy to reproduce by only starting the robot_state_helper (without any robot connected) and calling the action.

RobertWilbrandt, 2020-05-25:
Thanks for the quick fix

closed 2020-05-25 by RobertWilbrandt