UniversalRobots/Universal_Robots_ROS_Driver #180: Cannot locate rosdep definition for [ur_msgs] [ur_e_description] [ur_description]

Keith-Wangg, 2020-05-17:
when i try this:
`rosdep check --from-paths /home/keith/catkin_ws/src --ignore-src`
then it tells me:
`All system dependencies have been satisfied
ERROR[ur_robot_driver]: Cannot locate rosdep definition for [ur_msgs]
	rosdep key : ur_msgs
	OS name    : ubuntu
	OS version : bionic
	Data:
<no data>
`
the same as [ur_e_description] [ur_description].

so i delete  [ur_msgs] [ur_e_description] [ur_description] at the package.xml of the ur_robot_driver as the picture shows.

[image]

now there is no error. But i am worried which may have influence on my further usage .

So if you see this issue, please tell me how can i resolve this problem correctly, thanks.

Keith-Wangg, 2020-05-17:
also ,when i use catkin_make ,it tells me :
`In file included from /home/keith/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/ros/hardware_interface.cpp:28:0:
/home/keith/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/ros/hardware_interface.h:43:10: fatal error: ur_msgs/IOStates.h: 没有那个文件或目录
 #include <ur_msgs/IOStates.h>
          ^~~~~~~~~~~~~~~~~~~~
`
it seems that i lost some files...

gavanderhoorn, 2020-05-17:
Did you follow the steps under [Building](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building)? Especially the `clone fork of the description to use the calibration feature` step?

Keith-Wangg, 2020-05-18:
OH！ i am so sorry! i had thought `clone fork of the description to use the calibration feature` was an optional item. now i follow the whole steps and finish 'catkin_make' successfully.
thank you very much!

closed 2020-05-18 by Keith-Wangg

fmauch, 2020-05-18:
Indeed, this is misleading...