UniversalRobots/Universal_Robots_ROS_Driver #185: Thread 1 "ur_robot_driver" received signal SIGSEGV, Segmentation fault.

gdufay, 2020-05-27:
# Summary

Launching
```
roslaunch ur_robot_driver ur10_bringup.launch robot_ip:=10.31.1.70 \
kinematics_config:="${HOME}/left_ur_calibration.yaml" debug:=true
```
display
```shell
Thread 1 "ur_robot_driver" received signal SIGSEGV, Segmentation fault.
ur_driver::rtde_interface::RTDEClient::negotiateProtocolVersion (this=this@entry=0x55555585c9f0)
    at /home/gaetan/ur_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/rtde/rtde_client.cpp:87
87	  if (!tmp_version->accepted_)
```
and then quit due to missing required process.

# Version
 - ROS Driver version: commit 8d8e909554323d1d54ca2b141782a7e0184569b7
 - Affected Robot Software Version(s): Polyscope 3.13

# Impact

Cannot control robot

# Issue details

## Use Case and Setup

UR10 CB3.1 Series with default Polyscope version 3.2.21545 upgrading with a new USB key with version 3.13 on it

## Actual Behavior

Segfault then quit due to missing required process

Moreover, launching
```
roslaunch ur_calibration calibration_correction.launch robot_ip:=10.31.1.70 \
target_filename:="${HOME}/left_ur_calibration.yaml"
```
works but says
```
[DEBUG]: Invalid robot package type recieved: 24
[DEBUG]: Invalid robot package type recieved: 23
[DEBUG]: Invalid robot package type recieved: 5
```
cannot say if this is linked or not

fmauch, 2020-05-27:
Thanks for reporting this! Does this happen every time? I haven't tried v3.13 myself, yet.

gdufay, 2020-05-27:
Every time I'm connected to the robot yes

fmauch, 2020-05-27:
I cannot reproduce this at least with a 3.13 URSim.

> [DEBUG]: Invalid robot package type recieved: 24
[DEBUG]: Invalid robot package type recieved: 23
[DEBUG]: Invalid robot package type recieved: 5

This is most likely not linked to this problem, but part of #171

fmauch, 2020-05-27:
@gdufay could you pleas try adding the following line inside `/home/gaetan/ur_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/rtde/rtde_client.cpp` directly after line `84`

I think, that part of the code is suboptimal, but I am very curious what kind of RTDE package is received there.

You will find it in the output, mine looks like this:
```
[ INFO] [1590594075.744725501]: Received RTDE answer: accepted: 1
```
```c++
ROS_INFO_STREAM("Received RTDE answer: " << package->toString());
```

gdufay, 2020-05-28:
Thanks for your answer !

The output is : 
```
[ INFO] [1590647445.174693364]: Received RTDE answer: message: SafetySetup has not been confirmed yet, therefore no data will be send yet
source: 
```

So my bad sorry, I didn't configure properly the robot when I upgrade it, I'll try to fix it and come back to you

fmauch, 2020-05-28:
I don't hink, it's your bad at all. You clearly indicated a problem in our driver that we should check for this.

For you, the workaround would be to confirm the safety setup.

For us that means we should provide a fix.

gdufay, 2020-05-28:
Trying Polyscope v3.10.1 everything works (not the same security required)

Thanks again !

PR #186 (2020-06-12): Retry reading RTDE interface when unexpected messages appear during s…
…tartup

At startup we make a couple of requests to the RTDE interface. If the interface
publishes messages by itself, a simple read() from the interface might grab another
message than the answer.

This change checks whether an answer to our requests was received and reports
a warning otherwise while retrying.

For example, this was raised inside #185 

Closes #185
closed 2020-06-12 by fmauch