UniversalRobots/Universal_Robots_ROS_Driver #187: Reading data Streams - Could not get fresh data package from robot

orat, 2020-05-28:
# Summary

While reading from the topics ur_hardware_interface/rtde_data/target_q and ur_hardware_interface/rtde_data/temperature sometimes connection is brakes and we get log messages as follow: 

`[INFO] [1589967979.606725]: Loading controller: joint_state_controller
[INFO] [1589967979.607874]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1589967979.611608]: Loading controller: pos_traj_controller
[INFO] [1589967979.618885]: Loading controller: scaled_pos_traj_controller
[INFO] [1589967979.672892]: Loading controller: joint_group_vel_controller
[INFO] [1589967979.720808]: Loading controller: speed_scaling_state_controller
[INFO] [1589967979.728871]: Controller Spawner: Loaded controllers: pos_traj_controller, joint_group_vel_controller
[INFO] [1589967979.732789]: Loading controller: force_torque_sensor_controller
[ INFO] [1589967979.737161355]: Robot's safety mode is now NORMAL
[ INFO] [1589967979.738849418]: Robot mode is now RUNNING
[INFO] [1589967979.742886]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1589967979.748990]: Started controllers: joint_state_controller, scaled_pos_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[ INFO] [1589992656.595420159]: Robot mode is now POWER_ON
[ INFO] [1589992656.909132755]: Robot mode is now POWER_OFF
[ INFO] [1589992657.524765450]: Robot's safety mode is now FAULT
[ERROR] [1589992659.718762803]: Could not get fresh data package from robot
[ERROR] [1589992659.818972283]: Could not get fresh data package from robot
[ERROR] [1589992659.919288408]: Could not get fresh data package from robot
[ERROR] [1589992660.019491602]: Could not get fresh data package from robot
[ERROR] [1589992660.119762006]: Could not get fresh data package from robot
[ WARN] [1589992660.204704932]: Failed to read from stream, reconnecting in 1 seconds...
[ERROR] [1589992660.219980952]: Could not get fresh data package from robot`

It looks like that the error "Could not get fresh data package from robot" always occures in combination with the error "Robot's safety mode is now FAULT".

*Introduction to the issue*

# Versions
 - ROS Driver version: forked at 19.4.2020
 - ROS Meldic, Ubuntu 18.4, no-realtime adjustments
 - UR5e CB3

# Impact
The arbitraray behavoir of this error blocks completly our measurements campaign. We need to collect temperatures and joint angles during a complete day.

# Issue details
The error occures also when nothing else is running in the ROS system. There is no much payload. the UR is connected with a switch directly connected on the PC running ROS.

## Project status at point of discovered
*When did you first observe the issue?*
 - *Before I started?* No
 - *In first couple of tries?* No
 - *In normal use?* Yes

We need to know in which conditions principally the error message is loged to understand more about the conditions.

## Steps to Reproduce
Simple listen to the topics: ur_hardware_interface/rtde_data/target_q or ur_hardware_interface/rtde_data/temperature

fmauch, 2020-05-28:
If the robot goes to a Safety faul there's a reason for that. What happened there?

orat, 2020-05-29:
I just had a situataion with the "Could not get fresh data packages from robot" without the"safety fault". But indeed also the "safety fault" was message is strange because it occures during the robot is still standing.
I got the tip to log into the ur via ssh to look for running threads. Any idea at which I should look in detail?

orat, 2020-11-09:
The error occures more and more often without any other error messages in combination, e.g. no "safety fault"-errors. We have no idea how to solve this problem. It is unclear if the problem has its orign on the ur-robot itself or inside the ros-driver. Any idea what can go wrong on the ur itself? Maybe we have to leave ROS and connect the RTDE interface of the ur by our own code.

orat, 2020-11-09:
Upgrade of the ur5e robot to version 5.9 does not change the behavoir. It looks that the problem typically occures if the dashboard-server is used to switch to an other robot program on the ur (play) or if there is used some XRPC-communication by other ROS programs to communicate with the UR robot.  There was a discussion about non-robustness of the communication iwith the dashboard-server. Maybe is is the same with the streaming data from RTDE. So if maybe if access of the next frame is delayed (maybe caused by the ur robot) than a timeout occures and there is no automatically reconnection. If the error messages comes in big time distances it looks that the connection does not brake. But if there are too many delayed answers than the communication brakes and no reconnection is done.

fmauch, 2020-11-12:
From my understanding the RTDE connection breaks, if sending RTDE inputs fails (As of writing this, I am not sure whether there is a number of allowed missed packages), so that's why the RTDE connection breaks down when there are too many of those errors.

However, the question is still why those errors occur in the first place.

I can currently think of those sources:
 - UR-Controller doesn't send data in time. I would not assume this happening as RTDE is running as part of the real-time control-loop on the robot. 
 - Transmission problems on the connection between the robot and the ROS PC. Typically those are introduced by using a shared network over a switch, WiFi, etc. One way to mitigate this would be to skip the switch and connect the robot to the control pc directly.
 - Network traffic could be delayed by the network stack on the ROS machine.
 - The control thread on the ROS machine does not get enough priority from the scheduler. This is why we suggest using a realtime-enabled system with this driver. Please try that as the first thing. 
 - Too much load on the ROS pc. This would be closely connected (if not the same) to the point above.

Please try using a realtime system and a direct network connection and see whether this still persists.

orat, 2020-11-16:
Very thanks for you detailed explainations. We will try to check each point (direct network connection is just tested - it makes no difference, realtime kernel we will try soon). 

One more detail we have found: The data we getfrom RTDE before it breaks is complete. That means it looks like that we do not loose any frame before the connection breaks. So we think that die UR sends correctly the data and the ROS-side reads correctly the data but not fast enough. At some point a timeout occures because ROS does not read the data in time. There is sometimes a specific warning that says that data is not read for 32seconds. If this warning occures we get no further error message but we miss the frames for these 32 seconds.

fmauch, 2020-11-16:
> So we think that die UR sends correctly the data and the ROS-side reads correctly the data but not fast enough.

If this happens, you should see [this error](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/include/ur_client_library/comm/pipeline.h#L412) (under the assumption the the underlying network stack transports the data to the corresponding socket in time).

stefanscherzinger, 2021-08-17:
@orat Is this still relevant?

orat, 2021-08-18:
I have looked for error messages of type "Pipeline producer overflowed!" as fmauch has written but I havnt seen one. So I think my previous explaination what I think what is reason for the error maybe is wrong. The problem is still there and blocks longer measurement sessions. 
We have also tried to switch to a more realtime kernel without success because of conflicts with our NVidea graphic cards. We have also tried to use simple Rasberry pi but also without luck because the ur-driver has dependencies to rviz due to used messages. On our pi we are not able to install ros with the complete graphical environment so rviz cauld not be installed. We also tried to get rid of the dependencies of the ur-driver from rviz without success.

stefanscherzinger, 2021-08-18:
Reconsidering your use case and the clash of dependencies, it might in fact be more advantageous to implement your *measurement application* around the standalone [ur_client_library](https://github.com/UniversalRobots/Universal_Robots_Client_Library#rtdeclient). It should be possible to setup a Raspberry Pi with realtime-patched kernel.

> We need to collect temperatures and joint angles during a complete day.

I'm not sure how high-frequent you require these data, but at least the temperature indicates that a *lower reading rate* might suffice. Although not merged yet, you could consider this [PR ](https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/85)on the library? Maybe that goes into your direction.

gavanderhoorn, 2021-08-18:
@orat wrote:
> We have also tried to use simple Rasberry pi but also without luck because the ur-driver has dependencies to rviz due to used messages

which messages are those?

stefanscherzinger, 2021-09-01:
@orat Any progress with your application?

juniorsundar, 2022-05-24:
@orat 

If you are still looking to install kernel with PREEMPT-RT compatibility w/ Nvidia drivers, try the following method. We were successful in doing so, although we had to use a much older kernel (5.4.78).
Although FYI we still face the 'Could not get fresh data package from robot' error.

> ## Setting up Linux Kernel with Real-Time Capabilities
> 
> ### Building and Installing Real-Time Kernel
> 
> ```bash
> ## Set up source lists and install required dependencies
> echo "deb-src http://archive.ubuntu.com/ubuntu focal main" | sudo tee -a /etc/apt/sources.list.d/focal.list
> sudo apt-get update
> sudo apt-get build-dep linux
> sudo apt-get install libncurses-dev flex bison openssl libssl-dev dkms libelf-dev libudev-dev libpci-dev libiberty-dev autoconf fakeroot dwarves
> 
> ## Create installation and build directories. Can be deleted after the kernel is installed and functional
> mkdir ~/kernel && cd ~/kernel
> 
> ## Installing and patching Linux Ubuntu 5.4 kernel with real-time patch
> ## Instructions copied from (https://docs.ros.org/en/foxy/Tutorials/Building-Realtime-rt_preempt-kernel-for-ROS-2.html)
> wget https://mirrors.edge.kernel.org/pub/linux/kernel/v5.x/linux-5.4.78.tar.gz
> tar -xzf linux-5.4.78.tar.gz
> wget http://cdn.kernel.org/pub/linux/kernel/projects/rt/5.4/older/patch-5.4.78-rt44.patch.gz
> gunzip patch-5.4.78-rt44.patch.gz
> cd linux-5.4.78/
> patch -p1 < ../patch-5.4.78-rt44.patch
> 
> ## Editing configs for build
> cp /boot/config${MOST RECENT KERNEL} .config
> yes '' | make oldconfig
> make menuconfig
> ```
> 
> In the terminal GUI, do:
> 
> ```
> # Enable CONFIG_PREEMPT_RT
>  -> General Setup
>   -> Preemption Model (Fully Preemptible Kernel (Real-Time))
>    (X) Fully Preemptible Kernel (Real-Time)
> 
> # Enable CONFIG_HIGH_RES_TIMERS
>  -> General setup
>   -> Timers subsystem
>    [*] High Resolution Timer Support
> 
> # Enable CONFIG_NO_HZ_FULL
>  -> General setup
>   -> Timers subsystem
>    -> Timer tick handling (Full dynticks system (tickless))
>     (X) Full dynticks system (tickless)
> 
> # Set CONFIG_HZ_1000 (note: this is no longer in the General Setup menu, go back twice)
>  -> Processor type and features
>   -> Timer frequency (1000 HZ)
>    (X) 1000 HZ
> 
> # Set CPU_FREQ_DEFAULT_GOV_PERFORMANCE [=y]
>  ->  Power management and ACPI options
>   -> CPU Frequency scaling
>    -> CPU Frequency scaling (CPU_FREQ [=y])
>     -> Default CPUFreq governor (<choice> [=y])
>      (X) performance
> ```
> 
> in .config remove the KEYS
> 
> ```
> CONFIG_SYSTEM_TRUSTED_KEYS="debian/canonical-certs.pem" > CONFIG_SYSTEM_TRUSTED_KEYS=""
> ```
> 
> run the make:
> 
> ```bash
> ## Making .deb packages will take a lot of time (~10-30 mins)
> make -j `nproc` deb-pkg
> 
> ## Check build there should be 4 .deb files
> ls ../*deb
> 
> ## Install .deb. This will also take some time
> sudo dpkg -i ../*.deb
> ```
> 
> 
> ### Preparing Real-Time Kernel for UR Integration
> 
> ```bash
> sudo groupadd realtime
> sudo usermod -aG realtime $(whoami)
> ```
> 
> Then, make sure /etc/security/limits.conf contains. If it does not, then open it as superuser and copy in the below lines into the file:
> 
> ```
> @realtime soft rtprio 99
> @realtime soft priority 99
> @realtime soft memlock 102400
> @realtime hard rtprio 99
> @realtime hard priority 99
> @realtime hard memlock 102400
> ```
> 
> Check for RT capabilities
> 
> ```bash
> ## Update grup and reboot
> sudo update-grub
> sudo reboot
> 
> ## Once returned to the rt kernel, check if PREEMPT-RT is functional
> uname -v | cut -d" " -f1-4
> ## Outpu should read >> #1 SMP PREEMPT_RT
> ```
> 
> ### Installing Real-Time Compatible Graphic Drivers
> 
> Download .run file for a compatible NVIDIA Driver (version 455) using following website:
> 
> ```
> https://www.nvidia.com/Download/driverResults.aspx/166177/en-us
> ```
> 
> Create/Edit following file:
> 
> ```bash
> sudo nano /etc/modprobe.d/blacklist-nouveau.conf
> ```
> 
> If the lines do not already exist, add the following lines:
> 
> ```
> blacklist nouveau
> options nouveau modeset=0
> ```
> 
> Update kernel initramfs
> 
> ```bash
> sudo update-initramfs -u
> sudo reboot
> ```
> 
> Upon reboot, login to the real-time kernel through Recovery Mode and access kernel though root. Then, navigate to the location where the .run file is downloaded and run the following command:
> 
> ```bash
> sudo IGNORE_PREEMPT_RT_PRESENCE=1 bash NVIDIA-Linux-x86_64-455.38.run
> ```
> 
> Move through installation with default option (i.e. select Yes for everything). Then exit root and boot normally.
> 
> Once this is done, check the installation of NVIDIA driver with command:
> 
> ```bash
> nvidia-smi
> ```
> 
> It should show something like:
> 
> ```bash
> +-----------------------------------------------------------------------------+
> | NVIDIA-SMI 455.38       Driver Version: 455.38       CUDA Version: 11.1     |
> |-------------------------------+----------------------+----------------------+
> | GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
> | Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
> |                               |                      |               MIG M. |
> |===============================+======================+======================|
> |   0  GeForce GTX 105...  Off  | 00000000:01:00.0  On |                  N/A |
> | 30%   35C    P0    N/A /  75W |    215MiB /  4031MiB |      1%      Default |
> |                               |                      |                  N/A |
> +-------------------------------+----------------------+----------------------+
> ```

orat, 2023-04-13:
The problem is still there from time to time, even if we use a RT kernel. We havent seen the error a long time go in combination with "safety fault"-errors.

fmauch, 2023-07-17:
keeping open as https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/187#issuecomment-1506330734 stated the problem still exists.