UniversalRobots/Universal_Robots_ROS_Driver #188: Detaching teach Pendants from CB3 arms - Status Change

tahir1069, 2020-05-31:
# Summary

I want to detach the teach pendant from CB3 UR arm. Everything in the robot and ROS driver setup works perfectly fine. I have followed the tutorial [here](https://www.universal-robots.com/articles/ur-articles/disable-and-enable-the-cb-series-teach-pendant/) to detach the teach pendent. But after changing the settings to detach the teach pendent, the system prompts to  save and restart. After the restart the arm goes to fault state and I can not use the ROS driver as the systems safety status stays in fault. 
In current driver version I know this safety change status is not possible. But is there anyway to do it or some other alternate to detach the teach pendent and get the ROS driver working.

fmauch, 2020-05-31:
I've never done that myself. Did you follow the instructions and setup a separate EM-Stop?

You could try using the branch from #171 to find out what the actual error message is, but I'm not sure whether that actually runs fine on CB3 currently, as this is still at a very early draft version.

tahir1069, 2020-05-31:
I followed all the instructions on the manual for detaching the teach pendant.

I think my issue was not clear.

I can see the robot status in ROS for the robot from 'Booting to Power On and in Fault state'. I can not use the robot with its own software apart from ROS driver. 

what I want to know is that is there any way from ROS like: Powering Off the robot, Powering On etc. just from ROS driver without doing anything from teach pendant or polyscope.

gavanderhoorn, 2020-05-31:
> what I want to know is that is there any way from ROS like: [..] Powering On etc. just from ROS driver

Seeing as "powering on" is a physical action which boots up the controller (ie: applies power): how would the driver do this?

fmauch, 2020-05-31:
As @gavanderhoorn mentioned, powering on requires a physical interaction with the hardware. See point 8 of the manual you linked.

For powering off, close safety popups, etc see the [dashboard interface](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#ur_robot_driver_node).

tahir1069, 2020-06-02:
@gavanderhoorn Sorry for being not so clear. 
Powering on mean to initialize the robot and release the brakes(as usually done in the teach pendent) like in this service _/ur_hardware_interface/dashboard/power_on_ and _/ur_hardware_interface/dashboard/brake_release._ I have followed the tutorials and have configured the switches properly. 

As from @fmauch's comment for [dashboard interface](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#ur_robot_driver_node), I was able to do what I want from the robot like initializing the robot, releasing the brakes etc.

When the teach pendent is connected. Auto program is enabled and everything works fine from intiazing the robot to releasing the brakes, loading the program and playing it.

Calling the respective rosservices:

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/power_on  "{}" 
success: True
message: "Powering on"

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/brake_release "{}" 
success: True
message: "Brake releasing"

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/load_program "filename: '/programs/external_control.urp'" 
answer: "Loading program: /programs/external_control.urp, /programs/default.installation"
success: True

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/play "{}" success: True
message: "Starting program"

All above services are confirmed in the terminal running ur_driver:

> [ INFO] [1591101240.407090863]: Robot mode is now POWER_OFF
[ INFO] [1591101245.079297225]: Robot mode is now POWER_ON
[ INFO] [1591101247.750833537]: Robot mode is now IDLE
[ INFO] [1591101257.614861371]: Robot mode is now RUNNING
[ INFO] [1591101298.415724315]: Robot requested program
[ INFO] [1591101298.415769007]: Sent program to robot
[ INFO] [1591101298.460960885]: Robot ready to receive control commands.

But when I detach the teach pendent and restart the system first it goes to fault state (as it should be because in hardware teach pendent is connected but the software says 'disconnected'). So I turn it off detach the teach pendent, power on the robot and the robot boots properly.

> [ INFO] [1591099175.271830766]: Robot mode is now BOOTING
[ INFO] [1591099185.382120689]: Robot mode is now POWER_OFF
[ INFO] [1591099187.023139801]: Connected: Universal Robots Dashboard Server
[ INFO] [1591099191.829232454]: Robot mode is now POWER_ON
[ INFO] [1591099194.517023182]: Robot mode is now IDLE
[ INFO] [1591099198.429189953]: Robot mode is now RUNNING

Now when I load the external control program on the robot. It does not load giving an unknown error.

> rosservice call /ur_hardware_interface/dashboard/load_program "filename: '/programs/external_control.urp'" 
answer: "Error while loading program: /programs/external_control.urp:unknown failure"
success: False

The same thing happens for the installation load service

>rosservice call /ur_hardware_interface/dashboard/load_installation "filename: '/programs/default.installation'" 
ERROR: service [/ur_hardware_interface/dashboard/load_installation] responded with an error: Did not receive answer from dashboard server in time. Disconnecting from dashboard server.(Configured timeout: 1 sec)

tahir1069, 2020-06-02:
I think the problem might be with the _Auto program start_. It is working when the teach pendent is there and while detaching (save and restart) the robot goes to fault states and disables the _Auto program start_. 

After removing the teach pendent and turning on the robot the system safety status goes to _NORMAL_ while _Auto program start_ does not.

But this is guess may be you guys can point out where might be the problem.

tahir1069, 2020-06-02:
A small update after some work around.

I created two installation files one with teach pendant and one without teach pendant.

Now when the robot robot boots after detaching the teach pendant it goes to fault state. Than I load the installation file without teach pendant. Service call gives an error and disconnects from _Universal Robots Dashboard Server_. 

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/load_installation "filename: '/programs/default_without_tp.installation'" 
ERROR: service [/ur_hardware_interface/dashboard/load_installation] responded with an error: Did not receive answer from dashboard server in time. Disconnecting from dashboard server.(Configured timeout: 1 sec)

I reconnect to _ Universal Robots Dashboard Server_ and than _brake_release_ does its job i.e. powering on and releasing brakes. 

> mbs@mybotshop-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/connect "{}" 
success: True
message: ''

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/brake_release "{}" 
success: True
message: "Brake releasing"

Now its the time to load and play the external control program. But it again gives the same error as like installation file and disconnects from the _ Universal Robots Dashboard Server_.

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/load_program "filename: '/programs/external_control.urp'" 
ERROR: service [/ur_hardware_interface/dashboard/load_program] responded with an error: Did not receive answer from dashboard server in time. Disconnecting from dashboard server.(Configured timeout: 1 sec)

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/play "{}" 
ERROR: service [/ur_hardware_interface/dashboard/play] responded with an error: Failed to send request to dashboard server. Are you connected to the Dashboard Server?

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/connect "{}" 
success: True
message: ''

> mbs@mbs-rosPC:~$ rosservice call /ur_hardware_interface/dashboard/play "{}" 
success: False
message: "Failed to execute: play"

fmauch, 2020-06-03:
That looks more like a problem with interfacing the robot than a driver's problem. As stated above, I have no experience with detaching the teach pendant.

The dashboard services are only wrappers around the dashboard interface (as described [here](https://www.universal-robots.com/articles/ur-articles/dashboard-server-cb-series-port-29999/)).

You might be more lucky posting your issues inside the UR+ developer forum.

Nevertheless it would definitely interesting to have this solved for other ROS users, as well. I'll try to dig around, as well...

tahir1069, 2020-06-03:
This issue is already there on the UR+ forum but unfortunately nothing there so far. 

I will look and be back in case of an update...

gavanderhoorn, 2020-06-03:
> This issue is already there on the UR+ forum but unfortunately nothing there so far.

In that case: *link* to places where you cross-post your questions.

Duplicated effort quickly turns into wasted effort otherwise.

EbbeFuglsang, 2020-06-03:
Have you tried with the Dashboard server command?
set operational mode <mode>,
where <mode> is
• manual
• automatic

tahir1069, 2020-06-04:
@EbbeFuglsang _set operational mode_ command is for E series robot only. As can be seen [here](https://s3-eu-west-1.amazonaws.com/ur-support-site/15690/DashCommand.pdf), it is supported from _SW 5.0.0_ and I have _SW 3.13_ for _CB3_

cjsavoia, 2020-06-26:
Also posted on the UR+ Forums, but I try to always use default.installation in headless setups, it loads automatically on boot. A Safety mismatch will throw an error.

closed 2023-05-12 by github-actions[bot]