UniversalRobots/Universal_Robots_ROS_Driver #189: Source code for urcap missing

ipa-lth, 2020-06-05:
I would like to build upon the urcap functionality of this driver and add some specialized features to my fork of the urcap.
Further, I would expect this urcap to be a good example for urcap builders.

Since the urcap is actually a jar file, I can probably extract it and have a look, but at least for maintance the urcap maven project should be available somewhere.

Kind regards!

gavanderhoorn, 2020-06-05:
Are you referring to [UniversalRobots/Universal_Robots_ExternalControl_URCap](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap)?

ipa-lth, 2020-06-05:
Yes, great.
Thanx

ipa-lth, 2020-06-12:
It seems like the content of  https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap is not the source code of [externalcontrol-1.0.1.urcap](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/externalcontrol-1.0.1.urcap).

[externalcontrol-1.0.1.urcap](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/externalcontrol-1.0.1.urcap) offers two InputFields on the InstallationTab; while [InstallationView](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap/blob/b834adf8e7a9761c661fa319d75ecbb3edb8d12f/src/main/java/com/fzi/externalcontrol/impl/ExternalControlInstallationNodeView.java#L51) defines three Fields with different texts.

Could you please update the URcaps source repo?

gavanderhoorn, 2020-06-12:
Wouldn't this be something to report on the `Universal_Robots_ExternalControl_URCap` repository?

ipa-lth, 2020-06-12:
True. I will copy the issue. But I would leave it here as well.
The repos should be in sync.

gavanderhoorn, 2020-06-12:
So technically the driver here is "just another user" of the ExternalControl URCap. The URCap is not ROS specific, and was specifically designed to not be.

But I see your point.

PR #580 (2022-11-08): Remove URCap installation files from driver and replace references
Link to the respective Github release pages instead.

This replaces #576 and #511 and therefore closes #189.
closed 2022-11-08 by RobertWilbrandt