UniversalRobots/Universal_Robots_ROS_Driver #191: Not able to control the UR10e and can't connect UR10e with PC

tt2cloud, 2020-06-10:
# Summary

Hello, brother,I recently encountered a problem.I am trying to connect PC with UR10e. Following the tutorials, I don't able to the UR10e.I downloaded (https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git src/Universal_Robots_ROS_Driver) and (https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot) two packages, when I was connecting the PC IP
# Versions
 - ROS Driver version:ubuntu16.04+kinetic
 - URCaps Software version(s):5.6.0
robot_ip:=192.18.1.60   PC_ip:=192.168.1.61

### termial1:roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=192.168.1.60

auto-starting new master

process[master]: started with pid [16046]

ROS_MASTER_URI=http://localhost:11311

setting /run_id to f9f4466a-aaf6-11ea-87df-c8d9d21823f2

process[rosout-1]: started with pid [16063]

started core service [/rosout]

process[robot_state_publisher-2]: started with pid [16080]

process[ur_hardware_interface-3]: started with pid [16081]

process[ros_control_controller_spawner-4]: started with pid [16082]

[ INFO] [1591778833.244461464]: Main thread: SCHED_FIFO OK

[ INFO] [1591778833.244515999]: Main thread priority is 99

process[ros_control_stopped_spawner-5]: started with pid [16097]

[ INFO] [1591778833.254436740]: Initializing dashboard client

[ INFO] [1591778833.259312031]: Connected: Universal Robots Dashboard Server

process[controller_stopper-6]: started with pid [16133]

process[ur_hardware_interface/ur_robot_state_helper-7]: started with pid [16151]

[ INFO] [1591778833.279400014]: Initializing urdriver

[ INFO] [1591778833.279905846]: Checking if calibration data matches connected robot.

[ INFO] [1591778833.280442042]: Producer thread: SCHED_FIFO OK

[ INFO] [1591778833.280469870]: Thread priority is 99

[ INFO] [1591778833.286314783]: Waiting for controller manager service to come up on /controller_manager/switch_controller

[ INFO] [1591778833.287298633]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...

[ERROR] [1591778833.342191180]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [TODO Link to documentation] for details.

[INFO] [1591778833.634791]: Controller Spawner: Waiting for service controller_manager/load_controller

[INFO] [1591778833.691968]: Controller Spawner: Waiting for service controller_manager/load_controller

[ INFO] [1591778834.345545407]: Producer thread: SCHED_FIFO OK

[ INFO] [1591778834.345570416]: Thread priority is 99

[ INFO] [1591778834.389224845]: Setting up RTDE communication with frequency 500.000000

[FATAL] [1591778835.405016530]: Variable 'speed_slider_mask' is currently controlled by another RTDE client. The input recipe can't be used as configured

[ERROR] [1591778835.405094844]: Could not correctly initialize robot. Exiting

[... 151 lines cut ...]

/rosdistro: kinetic
/rosversion: 1.12.14
/rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018/manipulator/kinematics_solver: kdl_kinematics_pl...
/rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018/manipulator/kinematics_solver_attempts: 3
/rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018/manipulator/kinematics_solver_search_resolution: 0.005
/rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018/manipulator/kinematics_solver_timeout: 0.005
NODES
/
rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018 (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[rviz_yun_HP_EliteDesk_800_G4_TWR_16141_4429582541512616018-1]: started with pid [16294]
[ INFO] [1591778834.186943878]: rviz version 1.12.17
[ INFO] [1591778834.187022497]: compiled against Qt version 5.5.1
[ INFO] [1591778834.187047235]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1591778834.483021439]: Stereo is NOT SUPPORTED
[ INFO] [1591778834.483084141]: OpenGl version: 3 (GLSL 1.3).
[ERROR] [1591778835.330058916]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
[ INFO] [1591778835.330242282]: Loading robot model 'ur10e'...
[ INFO] [1591778835.330292133]: No root/virtual joint specified in SRDF. Assuming fixed joint

**The TP show don't connect to the remote PC establish.
Could you help me ,this question has really been bothering me, thanks advance!@fmauch@gavanderhoorn@Csp0920**

fmauch, 2020-06-11:
The error message you encounter says that there is another RTDE client controlling the speed slider.

Can you confirm that there is no other URcap installed on the robot and there is no other clie nt connected over RTDE?

gavanderhoorn, 2020-06-11:
Could this be EtherNet/IP again perhaps? Or Profinet?

fmauch, 2020-06-12:
> Could this be EtherNet/IP again perhaps? Or Profinet?

I would not rule out that possibility. @tt2cloud Is any fieldbus enabled (Installation -> Fieldbus)?

pradeepr-roboticist, 2020-06-13:
> The error message you encounter says that there is another RTDE client controlling the speed slider.
> 
> Can you confirm that there is no other URcap installed on the robot and there is no other clie nt connected over RTDE?

I am having the same issue. I got the same error message regarding the RTDE and 'speed_slider_mask'. I do not have any other program running that uses RTDE.
I traced it down to the "rtde_input_receipe.txt". Just for debugging, I removed all of the text in receipe.txt; still did not workout. But, I guess I hit a roadblock at that point.

I desperately need this driver to work. I am not allowed to use the older "ur_modern_driver" due to certain restrictions at my workplace and the fact that it does not support speed commands. And, I even resorted to talking directly via the RTDE client API but found no luck in making it work. Any suggestions would be welcome.

If it helps, here are some settings I have:
Hardware: UR5 CB3
Polyscope version: 3.7
OS: Ubuntu 18.04
ROS: Melodic

EtherNet/IP is enabled
PROFINET: Disabled
MODBUS: Disabled

Robot IP: 192.168.1.3
Desktop IP: 192.168.1.2

I do not appear to have the "Fieldbus" option.

Here are the error messages I received when I tried ur5_bringup.launch :
============================================================
[ INFO] [1592078115.878362080]: Initializing urdriver
[ INFO] [1592078115.878673261]: Checking if calibration data matches connected robot.
[ WARN] [1592078115.878962471]: No realtime capabilities found. Consider using a realtime system for better performance
[ INFO] [1592078115.883056940]: Waiting for controller manager service to come up on /controller_manager/switch_controller
[ INFO] [1592078115.884002996]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
[INFO] [1592078116.152466]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1592078116.157702]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1592078116.205561079]: Calibration checked successfully.
[ WARN] [1592078116.907728617]: No realtime capabilities found. Consider using a realtime system for better performance
[ INFO] [1592078117.003019754]: Negotiated RTDE protocol version to 2.
[ INFO] [1592078117.003538817]: Setting up RTDE communication with frequency 125.000000
[FATAL] [1592078118.005495009]: Variable 'speed_slider_mask' is currently controlled by another RTDE client. The input recipe can't be used as configured
[ERROR] [1592078118.005662044]: Could not correctly initialize robot. Exiting
================================================================================REQUIRED process [ur_hardware_interface-3] has died!
process has died [pid 8366, exit code 1, cmd /home/pradeepr/Desktop/test_ws/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/pradeepr/.ros/log/cc69506c-adaf-11ea-a7d7-1866da1cb998/ur_hardware_interface-3.log].
log file: /home/pradeepr/.ros/log/cc69506c-adaf-11ea-a7d7-1866da1cb998/ur_hardware_interface-3*.log
Initiating shutdown!
================================================================================
[ur_hardware_interface/ur_robot_state_helper-7] killing on exit
[controller_stopper-6] killing on exit
[ros_control_stopped_spawner-5] killing on exit
[ros_control_controller_spawner-4] killing on exit
[ur_hardware_interface-3] killing on exit
[robot_state_publisher-2] killing on exit
[WARN] [1592078118.361305]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1592078118.877379]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_stopper-6] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
=========================================================================

Thanks for your help.

fmauch, 2020-06-14:
> EtherNet/IP is enabled
PROFINET: Disabled
MODBUS: Disabled

>...

>I do not appear to have the "Fieldbus" option.

@pradeepr-roboticist  You seem to indeed have a Fieldbus enabled, namely EtherNet/IP. On a CB3 polyscope the fieldbusses are not under a separate subentry, but are listed inside the Installation menu. Also see the according [troubleshooting entry](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#when-starting-the-program-on-the-tp-i-get-a-c207a0-error).

Could you try to disable the EtherNet/IP fieldbus and then try again, please?

tt2cloud, 2020-06-15:
> The error message you encounter says that there is another RTDE client controlling the speed slider.
> 
> Can you confirm that there is no other URcap installed on the robot and there is no other clie nt connected over RTDE?

Year that's right .
Now I can successfully connect my pc with ur10e.Thanks for your help

tt2cloud, 2020-06-15:
> @pradeepr-roboticist You seem to indeed have a Fieldbus enabled, namely EtherNet/IP. On a CB3 polyscope the fieldbusses are not under a separate subentry, but are listed inside the Installation [...]

Thanks you help, I can successfully connect my pc with ur10e

pradeepr-roboticist, 2020-06-15:
> @pradeepr-roboticist You seem to indeed have a Fieldbus enabled, namely EtherNet/IP. On a CB3 polyscope the fieldbusses are not under a separate subentry, but are listed inside the Installation [...]

Wow! Thank you so much for your time! I am now able to connect.

gavanderhoorn, 2020-06-15:
@fmauch: would we already have an idea what causes the fieldbus setup / daemon to interfere here?

fmauch, 2020-06-15:
I wanted to investigate this today.

tt2cloud, 2020-06-15:
> > Could this be EtherNet/IP again perhaps? Or Profinet?
> 
> I would not rule out that possibility. @tt2cloud Is any fieldbus enabled (Installation -> Fieldbus)?

This problem has been sloved,thank you !
But,, I met another difficulty.Could you help me [(https://github.com/ros-controls/ros_control/issues/465)]

fmauch, 2020-06-15:
I confirm that as soon as EtherNet/IP is enabled, all RTDE inputs (at least the ones used by the recipe) are marked `IN_USE`

```
[ INFO] [1592224509.216983662]: speed_slider_mask confirmed as datatype: IN_USE
[ INFO] [1592224509.217059164]: speed_slider_fraction confirmed as datatype: IN_USE
[ INFO] [1592224509.217087356]: standard_digital_output_mask confirmed as datatype: IN_USE
[ INFO] [1592224509.217110027]: standard_digital_output confirmed as datatype: IN_USE
[ INFO] [1592224509.217139018]: configurable_digital_output_mask confirmed as datatype: IN_USE
[ INFO] [1592224509.217160624]: configurable_digital_output confirmed as datatype: IN_USE
[ INFO] [1592224509.217180648]: tool_digital_output_mask confirmed as datatype: IN_USE
[ INFO] [1592224509.217211312]: tool_digital_output confirmed as datatype: IN_USE
[ INFO] [1592224509.217232307]: standard_analog_output_mask confirmed as datatype: IN_USE
[ INFO] [1592224509.217260886]: standard_analog_output_type confirmed as datatype: IN_USE
[ INFO] [1592224509.217281989]: standard_analog_output_0 confirmed as datatype: IN_USE
[ INFO] [1592224509.217307054]: standard_analog_output_1 confirmed as datatype: IN_USE
```

gavanderhoorn, 2020-06-15:
Hm, weird.

Is this expected behaviour/documented?

fmauch, 2020-06-15:
I've searched around but couldn't find anything in the documentation. I'll come back to this when I know more.

tt2cloud, 2020-06-17:
> > Could this be EtherNet/IP again perhaps? Or Profinet?
> 
> I would not rule out that possibility. @tt2cloud Is any fieldbus enabled (Installation -> Fieldbus)?

Brother,could you help me solve the problems, I have no idea.
When I can moved the model under the rviz, but the real Hardware can't move
(https://github.com/ros-controls/ros_control/issues/465)   
Thanks in advance!

fmauch, 2020-06-18:
Closing, as one part is duplicate of #195 and second part is handled in #200

closed 2020-06-18 by fmauch