UniversalRobots/Universal_Robots_ROS_Driver #198: Add support for ros_control joint limits?

gavanderhoorn, 2020-06-16:
# Summary

The current implementation of the `hardware_interface` does not seem to register any [joint limit interfaces](https://github.com/ros-controls/ros_control/wiki/joint_limits_interface).

It's unclear whether this is by design or an oversight.

Earlier issue in `ur_modern_driver`: https://github.com/ros-industrial/ur_modern_driver/issues/249.

# Versions

All versions.

# Impact

Currently nothing for me personally. But a user expecting to be able to set more conservative limits on their robot motions by setting the appropriate values on the ROS parameter server will not be able to affect motion execution, as the driver doesn't load nor enforce them.

# Issue details
As the robot controller itself already adheres to the relevant safety standards, not having limits enforced in the driver is not really a safety risk (and even if it were: `ros_control` is not safety-rated).

At this point it affects user experience more than anything I believe.

## Steps to Reproduce

 - update the `.launch` file to load joint limits on the parameter server
 - start the driver
 - try to command motion clearly violating a specific (ie: more conservative) limit

## Expected Behavior
As `ros_control` implements saturation mostly, saturated velocity profiles (fi).

## Actual Behavior
No limiting.

## Workaround Suggestion
Reducing the slider on the TP, but that does not allow individual limits to be set, nor does it support any units. It's just a percentage of maximum performance.

Alternative: configuring reduced limits on the UR controller, but this is invisible to the ROS side.

gavanderhoorn, 2020-06-16:
I noticed this as I was working on aligning the joint limits files in the new `ur_description` (like [this one](https://github.com/ros-industrial/universal_robot/blob/8ca7b2ec7d48ef8b66dedf637d8d84f6a4013c68/ur_description/config/ur10/joints_limits_parameters.yaml)) with the `ros_control`/MoveIt style.

Doing that would allow reuse of those limits files, instead of needing to maintain a few separate ones.

fmauch, 2020-06-16:
Yes, this was never prioritized enough to tackle this. I fully agree

gavanderhoorn, 2020-06-16:
Related: https://github.com/ros-industrial/universal_robot/issues/510.

ipa-hsd, 2020-07-07:
@fmauch could you please assign this to me?

ipa-hsd, 2020-07-07:
[HardwareInterface](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/include/ur_robot_driver/ros/hardware_interface.h) class doesn't use `urdf::Model` object at the moment. Is that left out on purpose? I can read the limits only from the `joint_limits.yaml` and ignore the ones from `URDF` file altogether if that's the case. In any case, the values in yaml file will override the ones in `URDF`.

ipa-hsd, 2020-07-07:
[WIP] https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/223