UniversalRobots/Universal_Robots_ROS_Driver #199: Meaning of HMC_MESSAGE ?

oPiZiL, 2020-06-16:
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/fad8c0b34f291280cb03bde791b71ac7a3ba1349/ur_robot_driver/include/ur_robot_driver/primary/package_header.h#L51

fmauch, 2020-06-16:
I honestly don't know, this enum is built from the interface description that is linked here ("MessageSources" tab): https://www.universal-robots.com/articles/ur-articles/remote-control-via-tcpip/

oPiZiL, 2020-06-17:
I didn't find `HMC_MESSAGE` over there

fmauch, 2020-06-18:
At the bottom of the page, there's an excel sheet for download. I was referring to that.

oPiZiL, 2020-06-19:
Thanks. I found it  
MESSAGE_TYPE_DISCONNECT | -1 | X | X
-- | -- | -- | --
MESSAGE_TYPE_ROBOT_STATE | 16 | X | X
MESSAGE_TYPE_ROBOT_MESSAGE | 20 | X | X
MESSAGE_TYPE_HMC_MESSAGE | 22 | X | X
MESSAGE_TYPE_MODBUS_INFO_MESSAGE | 5 | X | X
MESSAGE_TYPE_SAFETY_SETUP_BROADCAST_MESSAGE | 23 | X | X
MESSAGE_TYPE_SAFETY_COMPLIANCE_TOLERANCES_MESSAGE | 24 | X | X
MESSAGE_TYPE_PROGRAM_STATE_MESSAGE | 25 | X | X

oPiZiL, 2020-06-19:
```
hmc 11 pathj([0.0, -2.1791969934615665, 0.3724006666666678, -1.358395660128231, 9.799131557042256E-31, -8.881784197001252E-16], r=0.005, [-0.2980006666666908, -2.5503976601282576, 0.3724006666666666, -1.9895963267948957, 4.3681398819133424E-17, 0.0], 0.05)
```   
Do you know pathj urscript function?

fmauch, 2020-06-21:
I think, asking question's about UR's interfaces isn't necessarily the right place here. I'll close this, as the original question was answered.

closed 2020-06-21 by fmauch