UniversalRobots/Universal_Robots_ROS_Driver #200: UR10e can connect the pc,but doesn't move it

tt2cloud, 2020-06-17:
# Summary
Hi, Dear all, I found a problem,the UR10e can connect the PC,I can see the gesture of ur10e under rviz, but when I use moveit to move it, I find that the real Hardware cannot move, and the error is as follows：

*Introduction to the issue*

# Versions
ROS Driver version:ubuntu16.04+kinetic
Affected Robot Software Version(s): 5.6.0
Affected Robot Hardware Version(s): UR10e
URCaps Software version(s):external control 1.0.1

terminal 1:roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=192.168.1.10
SUMMARY
PARAMETERS

/controller_stopper/consistent_controllers: ['joint_state_con...
/force_torque_sensor_controller/publish_rate: 500
/force_torque_sensor_controller/type: force_torque_sens...
/hardware_control_loop/loop_hz: 500
/hardware_interface/joints: ['shoulder_pan_jo...
/joint_group_vel_controller/joints: ['shoulder_pan_jo...
/joint_group_vel_controller/type: velocity_controll...
/joint_state_controller/publish_rate: 500
/joint_state_controller/type: joint_state_contr...
/pos_traj_controller/action_monitor_rate: 10
/pos_traj_controller/constraints/elbow_joint/goal: 0.1
/pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
/pos_traj_controller/constraints/goal_time: 0.6
/pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
/pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
/pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
/pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
/pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
/pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
/pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
/pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
/pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
/pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
/pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
/pos_traj_controller/joints: ['shoulder_pan_jo...
/pos_traj_controller/state_publish_rate: 500
/pos_traj_controller/stop_trajectory_duration: 0.5
/pos_traj_controller/type: position_controll...
/robot_description: <?xml version="1....
/robot_status_controller/handle_name: industrial_robot_...
/robot_status_controller/publish_rate: 10
/robot_status_controller/type: industrial_robot_...
/rosdistro: kinetic
/rosversion: 1.12.14
/scaled_pos_traj_controller/action_monitor_rate: 10
/scaled_pos_traj_controller/constraints/elbow_joint/goal: 0.1
/scaled_pos_traj_controller/constraints/elbow_joint/trajectory: 0.2
/scaled_pos_traj_controller/constraints/goal_time: 0.6
/scaled_pos_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
/scaled_pos_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
/scaled_pos_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
/scaled_pos_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
/scaled_pos_traj_controller/constraints/stopped_velocity_tolerance: 0.05
/scaled_pos_traj_controller/constraints/wrist_1_joint/goal: 0.1
/scaled_pos_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
/scaled_pos_traj_controller/constraints/wrist_2_joint/goal: 0.1
/scaled_pos_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
/scaled_pos_traj_controller/constraints/wrist_3_joint/goal: 0.1
/scaled_pos_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
/scaled_pos_traj_controller/joints: ['shoulder_pan_jo...
/scaled_pos_traj_controller/state_publish_rate: 500
/scaled_pos_traj_controller/stop_trajectory_duration: 0.5
/scaled_pos_traj_controller/type: position_controll...
/scaled_vel_traj_controller/action_monitor_rate: 10
/scaled_vel_traj_controller/constraints/elbow_joint/goal: 0.1
[... 373 lines cut ...]
[ INFO] [1592186901.386982485]: Replanning: yes
[ INFO] [1592186902.782810630]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186903.865067708]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186905.108830083]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186907.609091666]: Looking around: yes
[ INFO] [1592186914.485051143]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186915.356293093]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186915.577693663]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186917.639029757]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186929.123774766]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186930.265397395]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186931.238410735]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186939.893111610]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186941.204564882]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186968.658662198]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186969.758597597]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186970.864241100]: ABORTED: Solution found but controller failed during execution
[ INFO] [1592186972.708550601]: ABORTED: Solution found but controller failed during execution
[image]
terminal 4:rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller
[image]
I tested the script file today `rosrun ur_driver test_move.py` ,but I found the real Hardware can't move.
[image]

Could you help me solved the pronlems, Thank you advance!!@gavanderhoorn @fmauch

gavanderhoorn, 2020-06-17:
First: 请别再说兄弟了

Also: don't use the `ur_driver` package.

tt2cloud, 2020-06-17:
> First: 请别再说兄弟了
> 
> Also: don't use the `ur_driver` package.

Hi, I am sorry for my behavior（对不起，我为我的行为感到抱歉）
您好，如果您能看懂中文，我就能很方便跟您交流了，感谢你之前一直回我信息，我目前也一直在调试UR10e，PC现在能连上UR10e，但是我在规划路径的时候，实体UR10e并不能动，如果需要修改程序，我改如何修改代码呢？您能帮我么，感谢！
If I don't use the ur_driver package，What driver package do I need to use？  Thanks in advance！

gavanderhoorn, 2020-06-17:
I don't actually speak Chinese. That was Google translate.

> If I don't use the ur_driver package，What driver package do I need to use

you're already using a driver, namely `ur_robot_driver`.

What you shouldn't be using is the `test_move.py` script.

Instead, use `rqt_joint_trajectory_controller`.

You'll need to select the correct controller in the drop down on the right though.

Also: don't start MoveIt to test the driver. It's not needed. Please first verify that the driver works by moving the robot with the `rqt_joint_trajectory_controller`.

tt2cloud, 2020-06-17:
> I don't actually speak Chinese. That was Google translate.
>
>
> you're already using a driver, namely `ur_robot_driver`.
>
> What you shouldn't be using is the `test_move.py` script.
>
> Instead, use [...]

OK, I will try it tomorrow. Thank you!

tt2cloud, 2020-06-20:
> OK, I will try it tomorrow. Thank you!

Dear gavanderhoorn,I can already control the real Hardware UR10e through `rqt_joint_trajectory_controller `commands，thanks!
But ,,I still can't control the real Hardware UR10e through rivz interface,Could you help me? thanks in advance!

tt2cloud, 2020-06-21:
Dear gavanderhoorn,I want to use the Moveit API to realize the motion plan trajectory through the real Hardware UR10e. How to change the code? Or if you have information, please give me reference. thanks in advance!

fmauch, 2020-06-21:
I think, https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/55#issuecomment-562215033 could be a place to start.

tt2cloud, 2020-06-22:
> I think, [#55 (comment)](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/55#issuecomment-562215033) could be a place to start.
Thanks!! I published a new question above

chentongtong121, 2020-12-08:
hi  have you solved this problem already？I have the same problem，
“
Waiting for /follow_joint_trajectory to come up
Action client not connected :/follow_joint_trajectory 
”

stefanscherzinger, 2021-09-01:
@chentongtong121 @tt2cloud Driver and library changed significantly. Is this issue still relevant?

closed 2023-04-12 by github-actions[bot]